AdnanHodzic/displaylink-debian #1: Post installation guide is needed

yishinli, 2015-12-02:
Thanks for your displylink-debian script. The installation runs smoothly on my Debian Jessie laptop. However, the usb-LCD monitor still not work after executing the script.

My laptop is Dell Vostro 3350, which is with Intel+Radeon display output. Here's my post installation steps for the usb-LCD monitor:
* add "udl" to /etc/modules, to load udl kernel module for DisplayLink
* edit /etc/modprobe.d/fbdev-blacklist.conf
```
blacklist udlfb
blacklist radeon
```
* Refer to [arhcLinux/DisplayLink](https://wiki.archlinux.org/index.php/DisplayLink) for xrandr usage
```
xrandr --setprovideroutputsource 1 0 # add DisplayLink device, DVI-1-0
xrandr --output DVI-1-0 --auto --right-of LVDS1
```

kopax, 2016-01-13:
Hi, I have executed and installed the script as recommended on http://www.displaylink.org/forum/showthread.php?p=79352#post79352

I still don't know how to configure my display.

My Laptop is a X1 Carbon Lenovo on debian jessie.

+1 for adding a post installation guide

LauraMgne, 2016-01-28:
Hi !
First of all, thank you. I really appreciate your efforts to make a Debian version of the script.
As far as I'm concerned, the installation went well on my Dell Studio 1535, running Debian Jessie. 
Sadly, the USB-monitor is still not recognized. 

I didn't perform any other modification during the process. 
If you could kindly add a post installation guide, it would be really awesome.

AdnanHodzic, 2016-01-29:
Yes, sorry for delayed reply I've been very busy last couple of weeks.

This weekend, I'll post the post installation guide and also make some improvements to the script itself. Stay tuned!

LauraMgne, 2016-01-29:
Fantastic ! Thanks a lot.

AdnanHodzic, 2016-01-31:
I've made a [Post Install Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md), hopefully this will resolve some of the ambiguity. If it doesn't let me know and I'll extend it.

In latest script version, I've also added support for Ubuntu >= 15.04 and <= 16.04, so if you know anybody who has same problem on Ubuntu please point them here.

closed 2016-01-31 by AdnanHodzic

LauraMgne, 2016-03-20:
Hi Adnan !
Just to give you a heads-up. 
I have reinstalled the script and followed the post-install guide --> Everything works perfectly now.
Thanks man !

kopax, 2016-03-27:
I still can't use my MB168B+ with the highest resolution using your driver. 
I know it's a different issue but people are here, does anyone know how I could add my resolution to the list ?

AdnanHodzic, 2016-04-03:
@LauraMgne great to hear it worked as intended :)

@kopax you simply add "auto" or add your specific resolution to the xrandr line after "--mode"

kopax, 2016-04-05:
@AdnanHodzic Thanks for your answer.
I have added the "auto" and the resolution is stuck to `1366x768`.
How do I add the xrandr specific resolution of 1920x1080

AdnanHodzic, 2016-04-09:
Setting a custom resolution is a whole new topic from what this issue was originally for. Also what I see above is that you only set "1920x1080" resolution for for DVI-1-0

If you referred to "[Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#automated-configuration".

and look at alias I created for two:
```
# two
alias two="xrandr --setprovideroutputsource 1 0 && xrandr --setprovideroutputsource 2 0 && xrandr --output VIRTUAL1 --off --output DVI-1-0 --primary --auto --pos 0x0 --rotate normal --output DP1 --off --output HDMI2 --off --output HDMI1 --off --output eDP1 --off --output DVI-2-1 --auto --pos 1680x0 --rotate normal"
```

Although resolution on my laptop is 1366x768, once connected to two external displays it'll go to 1920x1080. That's what I meant by "auto". 

Also as suggested in [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#screen-layout-configuration), I'd suggest you install "arandr" and click around it until you get hang of xrandr. 

What's also cool about arandr is, once you configure the layout you want, you can save it as a shell script which you can later use without need for arandr.

Since this is specific problem to your hardware, and is not related to displaylink-debian tool itself, I really can't help you any further.

Google is your friend, and good luck.