AdnanHodzic/displaylink-debian #3: Service wouldn't start

sechanbask, 2016-02-20:
Hi,

Thank you for your mail... But, I have another problem, the service was not found 

```
cl@tosh:~$ sudo systemctl status displaylink.service
[sudo] password for cl: 
● displaylink.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)
cl@tosh:~$ sudo systemctl start displaylink.service
Failed to start displaylink.service: Unit displaylink.service failed to load: No such file or directory.
cl@tosh:~$ su -
# systemctl start displaylink.service
Failed to start displaylink.service: Unit displaylink.service failed to load: No such file or directory.
```

I am running it on debian jessie x64, and toshiba dynadock is connected on USB2. can you help me please ?

AdnanHodzic, 2016-04-03:
It seems like you have a faulty install. If you still have the binaries installed using displaylink-debian, just run the script again and select uninstall.

Otherwise please run "sudo displaylink-installer uninstall" and then try installing it again.

mjwillson, 2016-04-12:
I had this issue too on Ubuntu 16.04 beta 2. Uninstalling and installing again seemed to fix it, not sure why though!

olineham, 2016-04-28:
I'm have this problem on 16.04 final.  Have tried uninstall/reinstall and reboot a few times. It seems evdi won't load:
```
# modprobe evdi
modprobe: ERROR: could not insert 'evdi': Required key not available

# modinfo evdi
filename:       /lib/modules/4.4.0-21-generic/updates/dkms/evdi.ko
license:        GPL
description:    Extensible Virtual Display Interface
author:         DisplayLink (UK) Ltd.
srcversion:     05C1EF99A3B86CF7EF5C327
depends:        drm,drm_kms_helper
vermagic:       4.4.0-21-generic SMP mod_unload modversions 
parm:           initial_loglevel:Initial log level (int)
```

closed 2016-05-08 by AdnanHodzic