AdnanHodzic/displaylink-debian #4: Linux Mint support

granteagon, 2016-03-07:
Is Linux Mint support possible?  What would need to happen in the code to make it work?  I'm willing to take a crack at in on my machine and do a pull-request if it works.

Thanks

AdnanHodzic, 2016-04-03:
Only reason why Mint is not support is that it's not using systemd. If they make a switch to systemd in near future I'd be willing to extend this tool to Mint as well.

Until then, if you feel like give it a try, please feel free :)

AdnanHodzic, 2016-05-08:
@granteagon could you please tell me which version of Mint are using?

danielmanesku, 2016-07-27:
I manged to install the driver on Mint 18 Cinnamon. Downloaded if from official [website](http://www.displaylink.com/downloads/ubuntu), `chmod +x displaylink-driver-1.1.62.run && ./displaylink-driver-1.1.62.run`. After it was done `sudo systemctl start displaylink`, but I am not sure if this was needed, if it did anything at all.
That's it, it works now :)

AdnanHodzic, 2016-09-19:
In latest commit I've added support for Mint Linux (sarah, rosa, petra, olivia).

Please pull the latest version of debian-displaylink, and let me know if you still have any problems.

closed 2016-09-19 by AdnanHodzic