AdnanHodzic/displaylink-debian #5: Does not seem to work with Ubuntu 15.10 and 4.2.0-34

unfulvio, 2016-03-31:
I run in the same issue as installing the drivers downloaded from Displaylink

at the end of the installation it signals an error:

```
Job for displaylink.service failed because the control process exited with error code. See "systemctl status displaylink.service" and "journalctl -xe" for details.
```

if I run systemctl to check I have:

```
displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/displaylink.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2016-03-31 15:07:40 CEST; 4s ago
  Process: 24920 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

Mar 31 15:07:40 moshimoshi systemd[1]: Failed to start DisplayLink Manager Service.
Mar 31 15:07:40 moshimoshi systemd[1]: displaylink.service: Unit entered failed state.
Mar 31 15:07:40 moshimoshi systemd[1]: displaylink.service: Failed with result 'exit-code'.
```

basically it's the same issue I got here without using your script:

https://github.com/DisplayLink/evdi/issues/15

I suspect the newer driver does not contain the same code snippet you're attempting to patch: `detect_distro()` function etc.

do you happen to have an older copy of the drive? would you be so kind to handle a patched version of the drive you're currently using? I'd really appreciate that!

thanks

AdnanHodzic, 2016-04-03:
I suggest you fully uninstall the driver (sudo displaylink-installer uninstall), after doing so please make sure there are no running modules related to displaylink. And then try installing it again.

unfulvio, 2016-04-03:
hey there, thanks for looking up

no I had to uninstall the intel graphics driver -- apparently 1.4.0 installs a patched version of libdrm used by displaylink - I didn't even had to use your script by that point... I had Ubuntu 15.10 and Kernel 4.20 drivers installed finely

on the other hand now without the Intel Graphics Driver I have an issue with Chrome and had to run the browser with disabled gpu acceleration -- I'd imagine that power management is also less efficient

will look forward if there's any improvement with 16.04

thanks!

closed 2016-04-03 by unfulvio