AdnanHodzic/displaylink-debian #9: Crash in 16.05 on Kernel 4.4.02-generic

asevans48, 2016-05-11:
I was able to get your drivers running for kernel 4.4.02-generic Ubuntu. However, I always get into an infinite loop, race condition, starvation due to bad priority donation, or something on one of my screens. I will see if I can recreate the issue and get a core dump later. Perhaps a driver installed here is incompatible.

AdnanHodzic, 2016-05-11:
I tested it 16.04 with 4.4 kernel when it first came out, and everything worked as expected.

Could you please try uninstalling the driver and rebooting, then installing it again and reboot again. Then if the problem persists, include logs and whatever else you think might help in resolving your problem.

AdnanHodzic, 2016-05-23:
With of [the latest updates](https://github.com/AdnanHodzic/displaylink-debian/commit/7eb3e401792301400dc9ba17cc6db0697ed88fe5l), support for 1.1.62 driver is added along with full Ubuntu 16.04 support.

closed 2016-05-23 by AdnanHodzic