AdnanHodzic/displaylink-debian #12: Installing D3000 dock on debian

pprw, 2016-05-30:
Hi,

Thanks you for porting displaylink to debian.

I just got a D3000 USB dock station and try to make it work with my Dell Latitude 6430U running Debian unstable (xfce).

Before anything, when I plug this USB dock to my laptop, nothing is working (neither display nor simple usb ports).

I installed displaylink with your file: `./displaylink-debian.sh`. The installation works well, without any error but, after reboot, I cannot see any improvement (when I plug a usb stick to my under power D3000 dock plug into my laptop, nothing happen...).

Displaylink seems to work however:
```
$ systemctl status displaylink.service
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/displaylink.service; enabled; vendor preset: e
   Active: active (running) since lun. 2016-05-30 23:27:34 CEST; 6min ago
  Process: 1239 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1257 (DisplayLinkMana)
   CGroup: /system.slice/displaylink.service
           └─1257 /usr/lib/displaylink/DisplayLinkManager
```

but dmesg shows lot of errors related to the dock:

```
[  302.310386] usb 3-2: new full-speed USB device number 17 using xhci_hcd
[  302.422461] usb 3-2: device descriptor read/64, error -71
[  302.638442] usb 3-2: device descriptor read/64, error -71
[  302.854430] usb 3-2: new full-speed USB device number 18 using xhci_hcd
[  302.966461] usb 3-2: device descriptor read/64, error -71
[  303.182430] usb 3-2: device descriptor read/64, error -71
[  303.398444] usb 3-2: new full-speed USB device number 19 using xhci_hcd
[  303.398661] usb 3-2: Device not responding to setup address.
[  303.602571] usb 3-2: Device not responding to setup address.
[  303.806378] usb 3-2: device not accepting address 19, error -71
[  303.918364] usb 3-2: new full-speed USB device number 20 using xhci_hcd
[  303.918532] usb 3-2: Device not responding to setup address.
[  304.122604] usb 3-2: Device not responding to setup address.
[  304.326425] usb 3-2: device not accepting address 20, error -71
[  304.326526] usb usb3-port2: unable to enumerate USB device

```

Do you have any ideas that could help me to install this dock (DELL D3000) to my dell Latitude 6430U using your displaylink port?

AdnanHodzic, 2016-05-31:
Hotplug doesn't work on Debian. Please follow [post-install guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)

pprw, 2016-05-31:
Thanks for your quick answer.

The post installation guide focuses on displays. I don't want to use the D3000 dock to connect my laptop to external screens but only to connect it to ethernet, USB keyboard and charge my laptop through it.

The USB/Ethernet ports don't seems to work out of the box. After installing DisplayLink ( and `systemctl status displaylink.service` indicates DisplayLink is working), these ports are still unavailable.

pprw, 2016-06-06:
The problem was related to a buggy D3000 docking station. The new one works fine. You can close this bug report.

closed 2016-06-06 by pprw