AdnanHodzic/displaylink-debian #13: Some trouble with DL with Thinkpad T440s + debian 8.4 + UltraDock (3 outputscreens)

laubosslink, 2016-06-01:
Hi,

# Context

I currently try the BVU3500H to connect the 3rd screen with my laptop.

Currently I have two external screens, connect on DisplayPort through the Thinkpad Ultra Dock. To do that I firstly apply the following pach : https://cweiske.de/tagebuch/thinkpad-ultradock-screens.htm 

So I recompile the kernel to have Linux 3.17 with MST and it works perfectly on my debian 8.4 (with KDE).

Then I installed the script from this repo, and I've needed to play with the option of ```xrandr``` to setup the third screen.

**It works, but a problem still persist.**

[image]

# Setup

- Run the script from this repository to install dependencies 
- Setup the screen as the extremity of one external screen (not between laptop and external screen, cause it doesn't work). See below:
```
xrandr --setprovideroutputsource 1 0
xrandr --output DVI-1-0 --auto --right-of DP4
```
**Note**:  I was not able to do the equivalent with the desktop manager of KDE (Display Configuration - System Settings)

# Problem

On the last step of the setup, I've a segmentation fault error, but the screen still working. I attached the file from KDE.
[kded4-20160527-194743.kcrash.txt](https://github.com/AdnanHodzic/displaylink-debian/files/292464/kded4-20160527-194743.kcrash.txt)

laubosslink, 2016-06-01:
Ticket debian open : https://bugs.kde.org/show_bug.cgi?id=363766

AdnanHodzic, 2016-06-12:
Yea, I would categorize this as a KDE bug. Thanks for your feedback!

closed 2016-07-06 by AdnanHodzic