AdnanHodzic/displaylink-debian #14: Green screen after install

CrsOne, 2016-06-06:
Hi, 

First, sorry for my english.

Second, 

I realize step by step, during installation and post installation, but, my second monitor only show green screen.

`uname -a
Linux Carlos-Deb 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt25-2 (2016-04-08) x86_64 GNU/Linux
`

`
systemctl status displaylink.service
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/displaylink.service; enabled)
   Active: active (running) since dom 2016-06-05 22:40:05 CLT; 9min ago
  Process: 592 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 600 (DisplayLinkMana)
   CGroup: /system.slice/displaylink.service
           └─600 /usr/lib/displaylink/DisplayLinkManager

`

Help me please.

AdnanHodzic, 2016-06-12:
I guess your screen setup is different and why xrandr fails.

Did you try setting up your screen with "arandr" or "gnome-control-center display"? If yes are results any different, did you manage to get any output then green "screen" on your second monitor?

AdnanHodzic, 2016-09-19:
Please update debian-displaylink tool to latest version, then run remove. After reboot, run install again which will now install latest (1.2.58) version of DisplayLink Linux driver. 

Please tell me know if you're still experiencing same problem. Also could you please tell me which version of xserver-xorg-video-intel package are you using?

CrsOne, 2016-09-21:
Adnan,

thank you very much for you support, it work perfectly.

CrsOne, 2016-09-21:
Sorry,

After reboot, come back the green screen... Now reinstaling for test again...

AdnanHodzic, 2016-10-20:
Could you please, pull the latest version of driver, I made bunch of updates and it also has latest driver version (1.2.65).

And then repeat same thing I suggested first time:

1. git pull to latest version
2. run uninstall.
3. reboot
4. run install again 

That should resolve your problem, in case it doesn't please let me know/re-open the issue.

closed 2016-10-20 by AdnanHodzic