AdnanHodzic/displaylink-debian #15: Service won't start on Ubuntu 16.04, kernel 4.4.0-22

lloydwatkin, 2016-06-06:
Hi,

Thanks for this repo, however I can't seem to get the service running on the mentioned version of ubuntu + kernel 4.4.0-22

```bash
lloyd@xps13:~$ sudo systemctl status displaylink.service
[sudo] password for lloyd: 
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/displaylink.service; enabled; vendor pres
   Active: activating (auto-restart) (Result: exit-code) since Mon 2016-06-06 09
  Process: 3104 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

Jun 06 09:08:58 xps13 systemd[1]: Failed to start DisplayLink Manager Service.
Jun 06 09:08:58 xps13 systemd[1]: displaylink.service: Unit entered failed state
Jun 06 09:08:58 xps13 systemd[1]: displaylink.service: Failed with result 'exit-
lloyd@xps13:~$ uname -a
Linux xps13 4.4.0-22-generic #40-Ubuntu SMP Thu May 12 22:03:46 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

Any suggestions about what I can do? If you need any more info please let me know.

lloydwatkin, 2016-06-06:
Ok, tracked this down to UFEI. After disabling I managed to get the service up. Now to sort out the other issues with my dock! Thanks.

closed 2016-06-06 by lloydwatkin

AdnanHodzic, 2016-07-06:
Updated [Troubleshooting](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting) section of [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) with this information.

fdbozzo, 2016-12-18:
Hi, got a question: And what if you can't disable UEFI becuase you have one or more OSs installed this way? Whay the software do not work with UEFI enabled? Can't be fixed? Thanks.-

finetype, 2017-02-18:
Could you please include a little more information here? I've been following a maze of instructions--from vague "it just works" in amazon reviews, to hunting down other reviews that merely mention displaylink (what? I have an asus display, why displaylink? whatever...), to downloading and following those directions, which just silently don't seem to have the same last two lines of output which are missing (the dlm service won't start?) guessing around and trying to start it (just "exit", nothing useful in dmesg, no recognition in the gui tool in ubuntu), to hunting down old info, to realizing this release came out yesterday, back to amazon to look through reviews again, and then comments to reviews, wherein I found a link to this service, which I run... and again, nothing happens... go through guide, find a post-install guide, mention of checking service, which isn't running, go to bottom, link to this issue, where someone mentions yet again the same 'service won't start' issue, and they say they stopped UEFI...

ran the command mentioned here ([mokutil](http://askubuntu.com/questions/761096/secure-boot-issue-win10-ubuntu16-04)), about to restart and see if this is the magic bullet I guess, though only understand in the vaguest possible way what in the world is going on.

AdnanHodzic, 2017-02-18:
@fdbozzo UEFI was enforced by Microsoft, as it was believed to be modern replacement for the aging BIOS system and would help ensure boot-time malware couldn’t be injected into a system.

However, since this was developed with Windows in mind, on LInux it brought nothing but problems. In particular problems with boot, and certain kernel modules not being loaded properly.  

Furthermore, because on Linux some hardware doesn't work correctly under UEFI mode. Video cards are usually most common source of the problems and DisplayLink using USB graphics technology might be enough of a reason why you simply want it disable if you want it to work.

@finetype thanks for suggesting [mokutil](http://askubuntu.com/questions/761096/secure-boot-issue-win10-ubuntu16-04).  

I just pushed an update, in which displaylink-debian will install latest driver version (1.3.52). Which I believe would've fixed your problem.

finetype, 2017-02-18:
Awesome, thanks. Appreciate this repo. :)

<http://askubuntu.com/questions/761096/secure-boot-issue-win10-ubuntu16-04>.

simon-hester, 2017-08-08:
Hi 

I'm running Jessie 8.6 and getting the same error when I check the service status.

hesters@sphlaptop:~$ systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2017-08-09 00:29:58 BST; 409ms ago
  Process: 1692 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 1689 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.3.54 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 1692 (code=exited, status=1/FAILURE)

Looks like i;m running a newer release than indicated above so hoped the UEIF issue would not be an issue. I did try to turn this off but this breaks my dual boot so not an option. 

When i run the command to start the service it does not error so seems to work. 

Output from dmegs seem to suggest the device is connecting but i just get a blank screen witht he service failure above. 

 820.139645] usb 2-1: new SuperSpeed USB device number 3 using xhci_hcd
[  820.156203] usb 2-1: New USB device found, idVendor=17e9, idProduct=ff0b
[  820.156207] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  820.156208] usb 2-1: Product: ASUSTEK MB169B+
[  820.156210] usb 2-1: Manufacturer: DisplayLink
[  820.156211] usb 2-1: SerialNumber: H5LMTF014598

Are you able to provide any help suggestions on what to try next would love to get this working. 

Thanks in advance

Simon