AdnanHodzic/displaylink-debian #16: Edge of Screen scrolls to next display

thsteinmetz, 2016-07-05:
Thanks for this useful script. With it I was able to get up and running but I hit one issue: when I move my mouse cursor to the edge of my second display to go onto the DisplayLink powered third display, it scrolls my second display (as if I am scrolling horizontally) so as my cursor moves onto the third display, the display of my primary and secondary monitors has been shifted over. Have you ever experienced anything like this? 

I'm running Gnome and have an Nvidia card powering two 1440p displays and a 1080p display connected to the DisplayLink. 

I've played around with xrandr and arandr but nothing seems to help. Thoughts?

AdnanHodzic, 2016-07-06:
I never experienced anything similar, and sorry but this has nothing to do with displaylink-debian.

I believe this is either related to nvidia or displaylink driver. Try finding more info on: http://displaylink.org/forum/

closed 2016-07-06 by AdnanHodzic

thsteinmetz, 2016-07-06:
Yeah, I didn't think it did relate, was sort of just hoping you had seen something like it. Thanks for your hard work!

AdnanHodzic, 2016-07-07:
Most welcome!

Also make sure to [check out closed issues](https://github.com/AdnanHodzic/displaylink-debian/issues?q=is%3Aissue+is%3Aclosed), as some of them might help figuring out what exactly the problem is.