AdnanHodzic/displaylink-debian #17: Crashes on Kernel 4.6

jasonodoom, 2016-07-16:
The log is not very helpful but it does detect the external monitor. 

Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 8192 x 8192
eDP1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 294mm x 165mm
   1920x1080     59.93*+
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1280x1024     60.02  
   1440x900      59.89  
   1280x960      60.00  
   1360x768      59.80    59.96  
   1152x864      60.00  
   1024x768      60.00  
   800x600       60.32    56.25  
   640x480       59.94  
DP1 disconnected (normal left inverted right x axis y axis)
HDMI1 disconnected (normal left inverted right x axis y axis)
DVI-1-0 connected
   1680x1050     59.95 +
   1280x1024     75.02    60.02  
   1152x864      75.00  
   1024x768      75.03    60.00  
   800x600       75.00    60.32    56.25  
   848x480       60.00  
   640x480       75.00    59.94  
   720x400       70.08  
  1680x1050 (0x42) 146.250MHz
        h: width  1680 start 1784 end 1960 total 2240 skew    0 clock  65.29KHz
        v: height 1050 start 1053 end 1059 total 1089           clock  59.95Hz
  1280x1024 (0x44) 108.000MHz
        h: width  1280 start 1328 end 1440 total 1688 skew    0 clock  63.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  60.02Hz
  1024x768 (0x47) 65.000MHz
        h: width  1024 start 1048 end 1184 total 1344 skew    0 clock  48.36KHz
        v: height  768 start  771 end  777 total  806           clock  60.00Hz
  800x600 (0x49) 40.000MHz
        h: width   800 start  840 end  968 total 1056 skew    0 clock  37.88KHz
        v: height  600 start  601 end  605 total  628           clock  60.32Hz
  800x600 (0x4a) 36.000MHz
        h: width   800 start  824 end  896 total 1024 skew    0 clock  35.16KHz
        v: height  600 start  601 end  603 total  625           clock  56.25Hz
  640x480 (0x4d) 25.175MHz
        h: width   640 start  656 end  752 total  800 skew    0 clock  31.47KHz
        v: height  480 start  490 end  492 total  525           clock  59.94Hz

AdnanHodzic, 2016-07-29:
I'm not sure I understand what exactly is the problem? What do you mean by random crashes, what do you for this to happen?

Also, could you attach output of: /var/log/displaylink/DisplayLinkManager.log instead.

jasonodoom, 2016-07-29:
I have attached the log. It might be of no use though.

 /var/log/displaylink/DisplayLinkManager.log
nL4TC2qWBc4VYcjAc441hXfsHETKVByQG+rnZVuBFbG87Hj72F2WjIGL0pRzZdN00AGITLtEN9ah47MBUgbvgiaT+vH0iHqaN9AAAB4KdLJKWbu89zo2EfAwQpdozvKY1f7/6u214g5rCRRJaloe/V8bbjdS+fioCO6w8ILjpRNQFhoLS2LPsa6fqYvpNAg2tI/e0v1PP4x5fvktFZUDNQ==
nL4TC2qWBc4VYcjAc441hc/vcrmOye1Q8OiyhxkAYvkMi58BJPpuvPQERnrqEH+qOnXJ4wdLKEf8qmkWlCqUvBjhR0XctJzk6iFew5M9MhRy1lLawd/EHxQuplOSwIN+bwYpOtuSqP7U4gh/RcEq8h7XAWfA28mXUkZYJ+BvEAvb2enPDVoNebpIbMSXTCXZCGljXVlb7e9zXj9Ku9mJBQ==
qhdM5DcbWZiR02whBfZ5X9W8rA0oosqLodRtSbOm+068kXmIyWEC7ECuPPEcv13wFxzyg4dUEEZcSiYHMaLN+rMhksSiCti3Y6tv83U+mBZnWiQQ1fCohWFc0m2vJx/IpMdNQtzgfdfJwAhtJmpYOP9kBxzHaFVZQ8uHZ36tchbLTKubPgRpVHbe8sszv+zE5rBAognSD9N/czVRRw2Jdo9+QyCpjoXCjbskZTNKDz8=
nL4TC2qWBc4VYcjAc441hYyZFK77+W4QRC33p9rYfUePFIUnFHtYbmtu7amh4RRA0OorPt/28ft9qJhM+bKamSFwHgfn04Q+el0JW9IRICbwCk12QYz74oRk0zxlVZybTRggT1rzye6PFr+hydFyzq/mLF1kBsVVvRxtjb5Im50KlveIXxoPgNLJ4RApGScLw+AUawkwHFKkwcUJe/BpWQ==
nL4TC2qWBc4VYcjAc441hYRm35AJ8M/0k904KlLXF9/WLgkGt7GSYhzNOQGQTIf+ra836f2l5Q5+d6fBl4THu2SpyRuqOPprfAaTubES3HS16WQNTDoH/D9o6wrVeiflqUxcSCYgsYGzTr+sZBtQlIoPqRsq9qyNpWwLx0/7Fu5Ii17tbfQ2L0RzFsb6ogRyMUFmwntUz6yP/0O71DlybQ==
nL4TC2qWBc4VYcjAc441hXXf/35kX4mxCiLqicJ2Uu8TwF9wZfNfZL94VzlGYxazu+8dtfrwMMzIkQW+Lvw3/HFqHHsZWHkhIXec3A2AqC/ij6PFMe2v0gq4qAnNEeHGV6Mn5Nxs/UMNf9VXa5d4TccvkCl22V7wh8gscqKHi7TY+COLB1Ay2W+Yb5ZjGnOPsDHzVymvdLTU/obAMzYbcQ==
qhdM5DcbWZiR02whBfZ5X0hCaIV5zTIQiT64eP44YR7Y69jxQbhNMQ+75ePUj1yp1ipnMNQDwHVFS4w78joxBLpsOCbxIGqzjFeC6bCpK28FM4FEB3a/Oh7uyKWGdTbSSHkbsn2HxzbhPvVoGPtuNd8mcbuYG3yjB5td287h25JJUByLnpPOWuXqFHN8xmaIfAOGwl6ZW/3sFuKXwKdP/09okgUus7LoWo90G3nRqM4=
RAKMYFB/LPbsEiqnq+5D1LvcLuuGX64+y0MncKlgHd6wgxG3sakPIt3xUjih09yx354ijSaDiGkLsqqBGMek5YIXObnspof5qlguatKDs1VeoKGhNqbHK1gp/JPGS0K2b0B42H9+LG1zhKDPctZX/w7tRy+1eRC1hNe1ZkvpYCiUuCZZNMebx8WQ4OcZSMbHWPFSF76Ysdx4Z/44cGTM7fmfzVLEihparYKmcU9nIEdthOW5u/Bwf8A6qATUFA2qYkNsvIOEQV6ZOk7+9sg/cTIMhNAOj16UQRSx5Zlpd90=
RAKMYFB/LPbsEiqnq+5D1JEEpiyT/EV1+Cf3PmG6v5qCBRLs3hzA+gpRt5WEwNLSQxuKnUtweOYDSfggFgRZVGOLJNYdf0O19iYqaMxWlc93lvNJFCXmqksSzORmsek7jpDuB4Wf2KZFYzyquieQGF2spMQByHC29yEBpc+H1f+IYX/VnpTevUKYQlfF8+awakYFew47LJAK1/oFLavUmVwqipIckpeS2SIihcjh+wNw5BoYgjGywK86t0GWvS23
Mjmgwyo87ZbH+xb6Qoh4RmYBUvIxAI+2D9z/PbkjUXZNSQDf9ZpR/isyhItoxuz78IIRWtpakyiPDRQ0aqyLejWH/oz2la3E4OCslnE3QOYHBleqKEMkmTZortmLBQSzoUsPb+O2oo9Vmt/Gdv5wKLvD3y4AbXbL7nBWz7dbrOGvafRsXv996Xy2xHrs8uRXfo7eLuDrgDVRq5PSPtxrB9GlIK9oMB3A1tbPD1ySzbkVVaOKUtISmCXFCF18hEh8

jasonodoom, 2016-07-29:
It crashes as in when I execute _xrandr --output eDP1 --auto --left-of DVI-1-0 --output DVI-1-0 --auto_ I am logged out of the session.

AdnanHodzic, 2016-07-30:
Maybe you're trying to use wrong output, i.e instead of eDP1 it's eDP-1, could you please run to see what you're capable of:

`sudo lshw -C display; lsb_release -a; uname -a; sudo dmidecode -t 1; xrandr`

jasonodoom, 2016-08-01:
Here is the output:

  _*-display               
       description: VGA compatible controller
       product: Broadwell-U Integrated Graphics
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 09
       width: 64 bits
       clock: 33MHz
       capabilities: msi pm vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       resources: irq:47 memory:f6000000-f6ffffff memory:e0000000-efffffff ioport:f000(size=64) memory:c0000-dffff
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.5 (jessie)
Release:	8.5
Codename:	jessie
Linux fear 4.6.0-0.bpo.1-amd64 #1 SMP Debian 4.6.3-1~bpo8+1 (2016-07-13) x86_64 GNU/Linux
# dmidecode 2.12
SMBIOS 2.8 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Dell Inc.
	Product Name: XPS 13 9343
	Version: Not Specified
	Serial Number: JVTDD72
	UUID: 4C4C4544-0056-5410-8044-CAC04F443732
	Wake-up Type: Power Switch
	SKU Number: 0665
	Family: Not Specified

Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 8192 x 8192
eDP1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 294mm x 165mm
   1920x1080     59.93*+
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1280x1024     60.02  
   1440x900      59.89  
   1280x960      60.00  
   1360x768      59.80    59.96  
   1152x864      60.00  
   1024x768      60.00  
   800x600       60.32    56.25  
   640x480       59.94  
DP1 disconnected (normal left inverted right x axis y axis)
HDMI1 disconnected (normal left inverted right x axis y axis)_

AdnanHodzic, 2016-08-02:
What would happen if you replace your second "DVI-1-0" output of xrandr with "DVI-2-1"?

I'm also using 4.6 kernel (4.6.0-1-amd64) on Debian Stretch, so I would advise you to also upgrade xserver-xorg from to Stretch version. I think installation went fine, and that old xserver version is to blame or it's usage of xrandr with your setup.

Before making changes to your system, please refer to [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md) first and install "arandr" and play around with your setup to see what works for you.

Once you configure it there the way you like it, you can even export it as bash script.

jasonodoom, 2016-08-03:
That display doesn't exist. It only detects the display as root so that's an issue. I am going to figure out what happened. I followed the post install guide exactly as well as the install guide which only required root once. I will reinstall.

jasonodoom, 2016-08-03:
After reinstalling I have the same issue. The display is detected but when using it I am logged out of my session. I won't upgrade xserver-xorg to Stretch as I would like to keep a stable system.

jasonodoom, 2016-08-03:
Here is detailed information from xrandr:

eDP1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 294mm x 165mm
   1920x1080     59.93*+
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1280x1024     60.02  
   1440x900      59.89  
   1280x960      60.00  
   1360x768      59.80    59.96  
   1152x864      60.00  
   1024x768      60.00  
   800x600       60.32    56.25  
   640x480       59.94  
DP1 disconnected (normal left inverted right x axis y axis)
HDMI1 disconnected (normal left inverted right x axis y axis)
DVI-1-0 connected
   1366x768      59.79 +
   1360x768      60.02  
   1024x768      75.03    70.07    60.00  
   800x600       72.19    75.00    60.32    56.25  
   848x480       60.00  
   640x480       75.00    72.81    59.94  
   720x400       70.08  
  1024x768 (0x4a) 65.000MHz
        h: width  1024 start 1048 end 1184 total 1344 skew    0 clock  48.36KHz
        v: height  768 start  771 end  777 total  806           clock  60.00Hz
  800x600 (0x4d) 40.000MHz
        h: width   800 start  840 end  968 total 1056 skew    0 clock  37.88KHz
        v: height  600 start  601 end  605 total  628           clock  60.32Hz
  800x600 (0x4e) 36.000MHz
        h: width   800 start  824 end  896 total 1024 skew    0 clock  35.16KHz
        v: height  600 start  601 end  603 total  625           clock  56.25Hz
  640x480 (0x52) 25.175MHz
        h: width   640 start  656 end  752 total  800 skew    0 clock  31.47KHz
        v: height  480 start  490 end  492 total  525           clock  59.94Hz

I am connecting my laptop to one external monitor.

JustinDevelops, 2016-08-23:
I am actually getting the same issues. I installed just fine, my system does detect my two new displays even see's what monitor brand and model but when I try and activate either of the secondary monitors desktop/gui crashes and I am presented with a new login screen. My log also looks similar to OP, but if needed I can provide it and/or the output of the commands above. I am running Jessie stable all up to date on a Lenovo x250. What else can I provide? This is actually my first time participating in a thread in Git so not sure what all is needed for input.

jasonodoom, 2016-08-29:
What's interesting is that only as root does arandr detect the external display. I followed the instructions exactly.

jasonodoom, 2016-08-31:
Installing as root allows me to detect displays via arandr as a normal user.

jasonodoom, 2016-08-31:
Here's some information from dmesg:

_[   16.857732] evdi: [D] add_store:195 Increasing device count to 1
[   16.858047] evdi: [D] evdi_crtc_init:304 drm_crtc_init: 0
[   16.858147] evdi: [W] evdi_painter_crtc_state_notify:362 Painter does not exist!
[   16.858150] evdi: [D] evdi_detect:72 Painter is disconnected
[   16.858153] evdi evdi.0: No connectors reported connected with modes
[   16.858155] [drm] Cannot find any crtc or sizes - going 1024x768
[   16.859118] evdi evdi.0: fb1: evdidrmfb frame buffer device
[   16.859123] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   16.859124] [drm] No driver support for vblank timestamp query.
[   16.859126] [drm] evdi: evdi_stats_init
[   16.859131] [drm] Initialized evdi 1.1.61 20160510 on minor 1
[   16.861939] evdi: [W] evdi_painter_disconnect:462 (dev=0) An unknown connection to ffff8800c34c5a00 tries to close us
[   16.861942] evdi: [W] evdi_painter_disconnect:463  - ignoring
[   16.862273] evdi: [W] evdi_painter_disconnect:462 (dev=0) An unknown connection to ffff8802121f7000 tries to close us
[   16.862274] evdi: [W] evdi_painter_disconnect:463  - ignoring
[   16.862307] ------------[ cut here ]------------
[   16.862314] WARNING: CPU: 3 PID: 1566 at /usr/src/linux-headers-4.6.0-0.bpo.1-common/include/drm/drm_gem.h:217 evdi_dumb_create+0xd7/0xf0 [evdi]
[   16.862315] Modules linked in: pci_stub(E) vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) evdi(OE) binfmt_misc(E) vboxdrv(OE) bnep(E) ipt_REJECT(E) nf_reject_ipv4(E) xt_tcpudp(E) nf_conntrack_ipv4(E) nf_defrag_ipv4(E) xt_conntrack(E) nf_conntrack(E) iptable_filter(E) ip_tables(E) x_tables(E) nfsd(E) auth_rpcgss(E) nfs_acl(E) nfs(E) lockd(E) grace(E) fscache(E) sunrpc(E) cdc_mbim(E) cdc_wdm(E) snd_usb_audio(E) cdc_ncm(E) snd_usbmidi_lib(E) snd_rawmidi(E) usbnet(E) snd_seq_device(E) mii(E) uvcvideo(E) videobuf2_vmalloc(E) videobuf2_memops(E) videobuf2_v4l2(E) videobuf2_core(E) videodev(E) media(E) btusb(E) btrtl(E) btbcm(E) btintel(E) bluetooth(E) arc4(E) iwlmvm(E) mac80211(E) iTCO_wdt(E) intel_rapl(E) dcdbas(E) iTCO_vendor_support(E) hid_multitouch(E) x86_pkg_temp_thermal(E) intel_powerclamp(E) coretemp(E)
[   16.862342]  iwlwifi(E) kvm_intel(E) rtsx_pci_ms(E) cfg80211(E) kvm(E) memstick(E) irqbypass(E) snd_hda_codec_hdmi(E) joydev(E) snd_hda_codec_realtek(E) rfkill(E) snd_hda_codec_generic(E) serio_raw(E) pcspkr(E) elan_i2c(E) snd_soc_rt286(E) snd_soc_rl6347a(E) snd_soc_ssm4567(E) snd_soc_core(E) snd_compress(E) snd_hda_intel(E) snd_hda_codec(E) snd_hda_core(E) snd_hwdep(E) snd_pcm(E) i2c_i801(E) intel_pch_thermal(E) mei_me(E) snd_timer(E) snd(E) lpc_ich(E) mei(E) soundcore(E) shpchp(E) battery(E) soc_button_array(E) int3403_thermal(E) snd_soc_sst_acpi(E) snd_soc_sst_match(E) dw_dmac(E) i2c_designware_platform(E) dw_dmac_core(E) i2c_designware_core(E) processor_thermal_device(E) tpm_crb(E) int3402_thermal(E) int340x_thermal_zone(E) tpm_tis(E) intel_soc_dts_iosf(E) int3400_thermal(E) tpm(E) acpi_thermal_rel(E)
[   16.862364]  acpi_als(E) kfifo_buf(E) acpi_pad(E) industrialio(E) evdev(E) ac(E) processor(E) fuse(E) ecryptfs(E) cbc(E) encrypted_keys(E) parport_pc(E) ppdev(E) lp(E) parport(E) autofs4(E) ext4(E) crc16(E) jbd2(E) mbcache(E) algif_skcipher(E) af_alg(E) hid_generic(E) dm_crypt(E) dm_mod(E) hid_logitech_hidpp(E) hid_logitech_dj(E) usbhid(E) sg(E) sd_mod(E) crct10dif_pclmul(E) crc32_pclmul(E) crc32c_intel(E) ghash_clmulni_intel(E) jitterentropy_rng(E) hmac(E) rtsx_pci_sdmmc(E) drbg(E) ansi_cprng(E) aesni_intel(E) aes_x86_64(E) lrw(E) gf128mul(E) glue_helper(E) ablk_helper(E) cryptd(E) psmouse(E) ahci(E) libahci(E) libata(E) scsi_mod(E) ehci_pci(E) xhci_pci(E) xhci_hcd(E) ehci_hcd(E) i915(E) rtsx_pci(E) i2c_algo_bit(E) mfd_core(E) usbcore(E) usb_common(E) drm_kms_helper(E) fan(E) thermal(E) drm(E) wmi(E)
[   16.862391]  video(E) i2c_hid(E) hid(E) sdhci_acpi(E) sdhci(E) mmc_core(E) fjes(E) button(E)
[   16.862397] CPU: 3 PID: 1566 Comm: Xorg Tainted: G     U     OE   4.6.0-0.bpo.1-amd64 #1 Debian 4.6.4-1~bpo8+1
[   16.862398] Hardware name: Dell Inc. XPS 13 9343/0TM99H, BIOS A07 11/11/2015
[   16.862399]  0000000000000286 00000000196b0119 ffffffff813124c5 0000000000000000
[   16.862401]  0000000000000000 ffffffff8107af94 0000000000000000 ffff88021353fdb8
[   16.862403]  ffff880212321500 ffff88021217e400 ffffffffc00cb570 00000000000000b2
[   16.862405] Call Trace:
[   16.862410]  [<ffffffff813124c5>] ? dump_stack+0x5c/0x77
[   16.862413]  [<ffffffff8107af94>] ? __warn+0xc4/0xe0
[   16.862417]  [<ffffffffc0dfc7e7>] ? evdi_dumb_create+0xd7/0xf0 [evdi]
[   16.862427]  [<ffffffffc00a9786>] ? drm_ioctl+0x126/0x490 [drm]
[   16.862438]  [<ffffffffc00b9180>] ? drm_mode_page_flip_ioctl+0x2f0/0x2f0 [drm]
[   16.862442]  [<ffffffff812053a9>] ? do_vfs_ioctl+0x99/0x5d0
[   16.862445]  [<ffffffff8111a56e>] ? __audit_syscall_entry+0xbe/0x110
[   16.862447]  [<ffffffff810038a9>] ? syscall_trace_enter_phase1+0x139/0x150
[   16.862450]  [<ffffffff81205956>] ? SyS_ioctl+0x76/0x90
[   16.862451]  [<ffffffff81003cc4>] ? do_syscall_64+0x84/0x190
[   16.862454]  [<ffffffff815c85e5>] ? entry_SYSCALL64_slow_path+0x25/0x25
[   16.862456] ---[ end trace dd883323d2fb1260 ]---
[   16.862516] evdi: [D] evdi_detect:72 Painter is disconnected
[   16.862518] evdi: [D] evdi_detect:72 Painter is disconnected
[   16.862545] evdi: [D] evdi_detect:72 Painter is disconnected
[   16.862546] evdi: [D] evdi_detect:72 Painter is disconnected
[   16.862702] ------------[ cut here ]------------
[... cut ...] c_intel(E) ghash_clmulni_intel(E) jitterentropy_rng(E) hmac(E) rtsx_pci_sdmmc(E) drbg(E) ansi_cprng(E) aesni_intel(E) aes_x86_64(E) lrw(E) gf128mul(E) glue_helper(E) ablk_helper(E) cryptd(E) psmouse(E) ahci(E) libahci(E) libata(E) scsi_mod(E) ehci_pci(E) xhci_pci(E) xhci_hcd(E) ehci_hcd(E) i915(E) rtsx_pci(E) i2c_algo_bit(E) mfd_core(E) usbcore(E) usb_common(E) drm_kms_helper(E) fan(E) thermal(E) drm(E) wmi(E)
[   16.862781]  video(E) i2c_hid(E) hid(E) sdhci_acpi(E) sdhci(E) mmc_core(E) fjes(E)_

jasonodoom, 2016-09-04:
Compiled Kernel 4.4. I was able to install the DisplayLink driver but I experienced the same issue.

jasonodoom, 2016-09-14:
I did a reinstall last night. The issue (I believe @AdnanHodzic mentioned) was the use of an older  xserver-xorg-video-intel version in Jessie. @JustinDevelops I would not risk the stability of your system by upgrading this package because you will run into issues. The way to do it is (if you can spare the time) on a new install immediately grab the package from backports so as to avoid any dependency issues. Nevertheless, I can now connect my external display while running  stable Debian Jessie.

AdnanHodzic, 2016-09-18:
I still don't believe that upgrading "xserver-xorg" can cause that much instability. Especially since you can always downgrade.

Even upgrading whole system to "Stretch" shouldn't be much of a problem, especially since it's going to freeze in couple of months. 

Either way, glad you managed to resolve your problem.

closed 2016-09-18 by AdnanHodzic

jasonodoom, 2016-09-18:
.@AdnanHodzic Upgrading to Stretch is what forced me to do the reinstall. Dependency hell.

AdnanHodzic, 2016-09-18:
@jasonodoom to install packages from Testing/Unstable/Experimental, while on Stable without dependency hell. 

You should've followed my apt pinning guide, described on: ["Taking control over Debian and its package repositories"](http://foolcontrol.org/?p=1708)