AdnanHodzic/displaylink-debian #18: Displaylink Service Failed on Ubuntu 16.04

MrFreak1712, 2016-07-28:
Ubuntu 16.04 4.4.0-31-generic

when i try to Install it i get following error:

Job for displaylink.service failed because the control process exited with error code. See "systemctl status displaylink.service" and "journalctl -xe" for details.

`systemctl status displaylink.service`
`● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/displaylink.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Don 2016-07-28 22:24:17 CEST; 1s ago
  Process: 15613 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

Jul 28 22:24:17 MrFreakLinux systemd[1]: Failed to start DisplayLink Manager Service.
Jul 28 22:24:17 MrFreakLinux systemd[1]: displaylink.service: Unit entered failed state.
Jul 28 22:24:17 MrFreakLinux systemd[1]: displaylink.service: Failed with result 'exit-code'.`

AdnanHodzic, 2016-07-29:
I believe this same problem was resolved in: https://github.com/AdnanHodzic/displaylink-debian/issues/15

Could you make sure your EUFI is disabled?

closed 2016-08-01 by MrFreak1712

FlavioOliveiraSilva, 2017-01-27:
Hi ! How can I disable the EUFI?

AdnanHodzic, 2017-02-18:
Please see [following issue for more info](https://github.com/AdnanHodzic/displaylink-debian/issues/15#issuecomment-280824050)