AdnanHodzic/displaylink-debian #22: Build error with kernel 4.8-rc3

ashic, 2016-08-26:
The build seems to fail with 4.8-rc3 kernel running on ubuntu 16.04. The last bits of logs running ./displaylink-debian.sh are:

```
Archive:  DisplayLink_Ubuntu_1.1.62.zip
  inflating: 1.1.62/displaylink-driver-1.1.62.run  
  inflating: 1.1.62/LICENSE          
Creating directory displaylink-driver-1.1.62
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.1.62  100%  

Installing ... 

DisplayLink Linux Software 1.1.62 install script called: install
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
ERROR (code 3): Failed to build evdi/1.1.62. Consult /var/lib/dkms/evdi/1.1.62/build/make.log for details..

Install complete, please reboot to apply the changes
```

And the make.log:

```
cat /var/lib/dkms/evdi/1.1.62/build/make.log
DKMS make.log for evdi-1.1.62 for kernel 4.8.0-040800rc3-generic (x86_64)
Fri 26 Aug 06:29:08 BST 2016
make KBUILD_VERBOSE=1 SUBDIRS=/var/lib/dkms/evdi/1.1.62/build SRCROOT=/var/lib/dkms/evdi/1.1.62/build CONFIG_MODULE_SIG= -C /lib/modules/4.8.0-040800rc3-generic/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.8.0-040800rc3-generic'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (	\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /var/lib/dkms/evdi/1.1.62/build/.tmp_versions ; rm -f /var/lib/dkms/evdi/1.1.62/build/.tmp_versions/*
make -f ./scripts/Makefile.build obj=/var/lib/dkms/evdi/1.1.62/build
  gcc -Wp,-MD,/var/lib/dkms/evdi/1.1.62/build/.evdi_drv.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/4.9/include -I./arch/x86/include -I./arch/x86/include/generated/uapi -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -fno-pie -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-pie -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -maccumulate-outgoing-args -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -O2 --param=allow-store-data-races=0 -Wframe-larger-than=2048 -fstack-protector-strong -Wno-unused-but-set-variable -fno-var-tracking-assignments -pg -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -DCC_HAVE_ASM_GOTO -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_drv"'  -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/1.1.62/build/.tmp_evdi_drv.o /var/lib/dkms/evdi/1.1.62/build/evdi_drv.c
  if [ "-pg" = "-pg" ]; then if [ /var/lib/dkms/evdi/1.1.62/build/evdi_drv.o != "scripts/mod/empty.o" ]; then ./scripts/recordmcount  "/var/lib/dkms/evdi/1.1.62/build/evdi_drv.o"; fi; fi;
  gcc -Wp,-MD,/var/lib/dkms/evdi/1.1.62/build/.evdi_modeset.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/4.9/include -I./arch/x86/include -I./arch/x86/include/generated/uapi -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -fno-pie -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-pie -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -maccumulate-outgoing-args -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -O2 --param=allow-store-data-races=0 -Wframe-larger-than=2048 -fstack-protector-strong -Wno-unused-but-set-variable -fno-var-tracking-assignments -pg -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -DCC_HAVE_ASM_GOTO -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_modeset"'  -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/1.1.62/build/.tmp_evdi_modeset.o /var/lib/dkms/evdi/1.1.62/build/evdi_modeset.c
/var/lib/dkms/evdi/1.1.62/build/evdi_modeset.c: In function ‘evdi_sched_page_flip’:
/var/lib/dkms/evdi/1.1.62/build/evdi_modeset.c:155:3: error: implicit declaration of function ‘drm_send_vblank_event’ [-Werror=implicit-function-declaration]
   drm_send_vblank_event(dev, 0, event);
   ^
/var/lib/dkms/evdi/1.1.62/build/evdi_modeset.c: In function ‘evdi_modeset_init’:
/var/lib/dkms/evdi/1.1.62/build/evdi_modeset.c:370:2: error: implicit declaration of function ‘drm_dev_set_unique’ [-Werror=implicit-function-declaration]
  drm_dev_set_unique(dev, dev_name(dev->dev));
  ^
cc1: some warnings being treated as errors
scripts/Makefile.build:289: recipe for target '/var/lib/dkms/evdi/1.1.62/build/evdi_modeset.o' failed
make[2]: *** [/var/lib/dkms/evdi/1.1.62/build/evdi_modeset.o] Error 1
Makefile:1477: recipe for target '_module_/var/lib/dkms/evdi/1.1.62/build' failed
make[1]: *** [_module_/var/lib/dkms/evdi/1.1.62/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.8.0-040800rc3-generic'
Makefile:21: recipe for target 'all' failed
make: *** [all] Error 2
```

AdnanHodzic, 2016-09-11:
Do you have kernel headers installed?

jasonodoom, 2016-09-11:
Yes I have Kernel headers installed.

AdnanHodzic, 2016-09-11:
Could you please report this problem with [evdi driver,](https://github.com/DisplayLink/evdi/issues) as I know there were/are driver problems with kernel 4.7/4.8, like [this one](https://github.com/DisplayLink/evdi/issues/38).

I think this problem has nothing to do with the installer, but the driver itself. Thanks and please keep me updated.

jasonodoom, 2016-09-11:
Sorry I received this via email but was not subscribed.... But in relation to issue 17 I did install Kernel headers. https://github.com/AdnanHodzic/displaylink-debian/issues/17

AdnanHodzic, 2016-09-19:
Please update debian-displaylink tool to latest version, then run remove. After reboot, run install again which will now install latest (1.2.58) version of DisplayLink Linux driver. 

Please tell me know if you're still experiencing same problem.

ashic, 2016-10-03:
I've now managed to install it, and it seems to recognise the monitor (in settings, I see two monitors). But the external monitor is blank, and if I do detect monitor, the internal one shows both as monitor 1 and 2. Any ideas?

NotBR, 2016-10-04:
Seeing similar here, installed latest DisplayLink drivers 1.2.58 both manually and via displaylink-debian and get the same issue so not an issue as such with displaylink-debian.
This is running on Ubuntu 16.10

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu Yakkety Yak (development branch)
Release:	16.10
Codename:	yakkety
```

I see the following log in dmesg when I plug in to the dock;

```dmesg
jamesr@prune:~/Downloads/displaylink-debian$ dmesg
[  616.999983] usb 1-5: new high-speed USB device number 12 using xhci_hcd
[  617.141713] usb 1-5: New USB device found, idVendor=2230, idProduct=0006
[  617.141716] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  617.141717] usb 1-5: Product: USB2.0 Hub             
[  617.141718] usb 1-5: Manufacturer: VIA Labs, Inc.         
[  617.142660] hub 1-5:1.0: USB hub found
[  617.143593] hub 1-5:1.0: 4 ports detected
[  617.260136] usb 2-5: new SuperSpeed USB device number 8 using xhci_hcd
[  617.282724] usb 2-5: New USB device found, idVendor=2230, idProduct=0006
[  617.282727] usb 2-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  617.282728] usb 2-5: Product: USB3.0 Hub             
[  617.282729] usb 2-5: Manufacturer: VIA Labs, Inc.         
[  617.284771] hub 2-5:1.0: USB hub found
[  617.284995] hub 2-5:1.0: 4 ports detected
[  617.451966] usb 1-5.1: new high-speed USB device number 13 using xhci_hcd
[  617.565965] usb 1-5.1: New USB device found, idVendor=2230, idProduct=0006
[  617.565967] usb 1-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  617.565969] usb 1-5.1: Product: USB2.0 Hub             
[  617.565970] usb 1-5.1: Manufacturer: VIA Labs, Inc.         
[  617.567036] hub 1-5.1:1.0: USB hub found
[  617.567961] hub 1-5.1:1.0: 4 ports detected
[  617.672181] usb 2-5.1: new SuperSpeed USB device number 9 using xhci_hcd
[  617.694874] usb 2-5.1: New USB device found, idVendor=2230, idProduct=0006
[  617.694877] usb 2-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  617.694878] usb 2-5.1: Product: USB3.0 Hub             
[  617.694879] usb 2-5.1: Manufacturer: VIA Labs, Inc.         
[  617.697680] hub 2-5.1:1.0: USB hub found
[  617.697911] hub 2-5.1:1.0: 4 ports detected
[  618.088106] usb 2-5.1.1: new SuperSpeed USB device number 10 using xhci_hcd
[  618.108888] usb 2-5.1.1: New USB device found, idVendor=17e9, idProduct=4323
[  618.108890] usb 2-5.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  618.108892] usb 2-5.1.1: Product: Plugable UD-ULTCDL
[  618.108893] usb 2-5.1.1: Manufacturer: DisplayLink
[  618.108894] usb 2-5.1.1: SerialNumber: 60439512
[  618.118649] usb 2-5.1.1: Warning! Unlikely big volume range (=511), cval->res is probably wrong.
[  618.118651] usb 2-5.1.1: [16] FU [Plugable Audio Playback Volume] ch = 6, val = -8176/0/16
[  618.119241] usb 2-5.1.1: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
[  618.119242] usb 2-5.1.1: [13] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
[  618.140840] cdc_ncm 2-5.1.1:1.5: MAC-Address: 8c:ae:4c:f6:02:2b
[  618.140846] cdc_ncm 2-5.1.1:1.5: setting rx_max = 16384
[  618.140998] cdc_ncm 2-5.1.1:1.5: setting tx_max = 16384
[  618.141194] cdc_ncm 2-5.1.1:1.5 usb0: register 'cdc_ncm' at usb-0000:00:14.0-5.1.1, CDC NCM, 8c:ae:4c:f6:02:2b
[  618.824429] cdc_ncm 2-5.1.1:1.5 enx8cae4cf6022b: renamed from usb0
[  618.857398] IPv6: ADDRCONF(NETDEV_UP): enx8cae4cf6022b: link is not ready
[  618.857467] IPv6: ADDRCONF(NETDEV_UP): enx8cae4cf6022b: link is not ready
[  618.923929] usb 1-5.3: new low-speed USB device number 14 using xhci_hcd
[  619.078694] usb 1-5.3: No LPM exit latency info found, disabling LPM.
[  619.083015] usb 1-5.3: New USB device found, idVendor=0451, idProduct=ace1
[  619.083017] usb 1-5.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  619.083018] usb 1-5.3: Product: TPS65982
[  619.083019] usb 1-5.3: Manufacturer: Texas Instruments
[  619.083020] usb 1-5.3: SerialNumber: 01.00.00
[  619.907819] evdi: [W] evdi_painter_disconnect:478 (dev=-1) An unknown connection to ffff97de9d971e00 tries to close us
[  619.907822] evdi: [W] evdi_painter_disconnect:479  - ignoring
[... 76 lines cut ...]
[  620.503623] Code: 1f 03 00 00 49 c7 c0 86 36 4a bf 48 c7 c2 30 0b 48 bf 48 c7 c6 4c 8e 49 bf 4d 89 e1 48 89 d9 48 c7 c7 10 03 4a bf e8 03 05 f7 ff <0f> 0b 4c 8b 75 b8 48 8b 5d d0 45 89 fd 4c 8b 65 c8 4c 89 e6 48 
[  620.505625] RIP  [<ffffffffbea2e421>] __check_object_size+0x111/0x49b
[  620.507548]  RSP <ffff97de32afbdb0>
[  620.509576] ---[ end trace de2b17021b40d47f ]---
[  620.620041] evdi: [D] evdi_detect:69 (dev=2) Painter is connected
[  620.620045] evdi: [D] evdi_painter_get_edid_copy:192 (dev=2) 00 ff ff
[  620.620372] evdi: [D] evdi_detect:69 (dev=1) Painter is connected
[  620.620374] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[  620.940016] evdi: [D] evdi_detect:69 (dev=2) Painter is connected
[  620.940020] evdi: [D] evdi_painter_get_edid_copy:192 (dev=2) 00 ff ff
[  620.940154] evdi: [D] evdi_detect:69 (dev=1) Painter is connected
[  620.940156] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[  620.991856] evdi: [D] evdi_painter_mode_changed_notify:391 (dev=2) Notifying mode changed: 1920x1080@60; bpp 32; 
[  620.991859] evdi: [D] evdi_painter_mode_changed_notify:392 pixel format 875713112
[  620.991862] evdi: [D] evdi_painter_crtc_state_notify:374 (dev=2) Notifying crtc state: 0
[... cut ...]
[  620.997564] evdi: [D] evdi_painter_mode_changed_notify:392 pixel format 875713112
[  620.997567] evdi: [D] evdi_painter_crtc_state_notify:374 (dev=1) Notifying crtc state: 0
[  620.997568] evdi: [D] evdi_painter_dpms_notify:361 (dev=1) Notifying dpms mode: 0
[  621.748026] evdi: [D] evdi_detect:69 (dev=2) Painter is connected
[  621.748030] evdi: [D] evdi_painter_get_edid_copy:192 (dev=2) 00 ff ff
[  621.748146] evdi: [D] evdi_detect:69 (dev=1) Painter is connected
[  621.748149] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
```

troglobit, 2016-10-18:
The latest PR to bring us up to v1.2.65 should fix all issues with Linux 4.8.  Works here, albeit quite shaky, on Ubuntu 16.10 (which comes with 4.8 by default).

AdnanHodzic, 2016-10-20:
Yes, I've made bunch of updates to the tool, it's also running 1.2.65 driver now.

Could you please pull the latest version and report if you're still facing the same problem?

closed 2016-10-20 by AdnanHodzic