AdnanHodzic/displaylink-debian #23: Unit displaylink.service not found.

leonnoel, 2016-09-26:
Hi,

I am trying to get a MB168B usb monitor to work with a t410 Thinkpad. I've tried installing and reinstalling, but still wind up with the following error: 

``
:~$ systemctl status displaylink.service
● displaylink.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)
:~$ systemctl start displaylink.service
Failed to start displaylink.service: Unit displaylink.service not found.
`` 

The only other odd thing I've noticed (installer worked fine) was when trying to uninstall I get the following error:

``
Removing evdi module 
rmmod: ERROR: Module evdi is in use
``

The uninstall, however, does complete,

Any ideas? Many thanks!

TinkerFu, 2016-10-03:
I'm having the same issue.  Running Bunsenlabs Hydrogen ( debian jessie ).

cilynx, 2016-10-15:
Upstream, the service name has been changed from `displaylink.service` to `dlm.service`.  I've submitted the PR you see above to address this in the Post Install Guide.

With respect to the header dependency, upstream is looking for Kconfig (which Ubuntu kernel header packages contain, but Debian kernel header packages do not) in your module build dir:

`# Required kernel version`
`KVER=$(uname -r)`
`KVER_MIN="3.14"`
`version_lt "$KVER" "$KVER_MIN" && missing_requirement "Kernel version $KVER is too old. At least $KVER_MIN is required"`
`# Linux headers`
`[ ! -f "/lib/modules/$KVER/build/Kconfig" ] && missing_requirement "Linux headers for running kernel, $KVER"`

You'll notice that `/lib/modules/$KVER/build` is actually a symlink to `/usr/src/linux-headers-$KVER`:
`lrwxrwxrwx  1 root root     36 Sep 19 14:09 build -> /usr/src/linux-headers-4.7.0-1-amd64`

It turns out that `Kconfig` itself is never actually used by the displaylink / EVDI build process, so you can get away with just `touch`ing Kconfig in your current header directory and everything should build properly after that.

It would be possible within the script here to modify upstream before it runs to check for one of the header files that it actually needs as opposed to looking for `Kconfig`, but I'll leave it up to @AdnanHodzic if he wants to go that far.

closed 2016-10-17 by AdnanHodzic

AdnanHodzic, 2016-10-20:
I just wanted to inform you  that I just pushed bunch of updates, one of them is ["Fixed Debian/Kali Linux install bug (failure to identify kernel headers)"](https://github.com/AdnanHodzic/displaylink-debian/commit/e1225f208e30e2aa1ace75969be6c81e479e14df)

Which adequately fixes problem mentioned here.

flipside2000, 2016-12-29:
Hi AdnanHodzic, I am however still receiving "Unit dlm.service could not be found". This is on a fresh Kali install. Performing the steps as per cilynx' steps above resolves this however. Just a heads up

AdnanHodzic, 2017-01-15:
Hi @flipside2000, I just pushed changes which take care of this problem, so no need to do @cilynx steps manually anymore.a

gitowiec, 2017-02-24:
I have the same issue here on Ubuntu 16.04.2 with Dell 7559 laptop and D3100 Displaylink adapter

PR #25 (2016-10-17): Upstream service name change
Upstream has changed the name of the systemd service from `displaylink` to `dlm`.  The tests explained here work fine if you substitute `dlm.service` for `displaylink.service`.  You can see the new file name in the `add_systemd_service()` function in `1.2.58/displaylink-driver-1.2.58/displaylink-installer.sh` starting on line 112.

rcw@antec:~/Projects/displaylink-debian/1.2.58/displaylink-driver-1.2.58$ systemctl status displaylink.service
Unit displaylink.service could not be found.
rcw@antec:~/Projects/displaylink-debian/1.2.58/displaylink-driver-1.2.58$ systemc [... cut]
commit 5f6faa05 (2016-10-17): Upstream service name change

Upstream has changed the name of the systemd service from `displaylink` to `dlm`.  The tests explained here work fine if you substitute dlm.service for displaylink.service.  You can see the new file name in the add_systemd_service() function in 1.2.58/displaylink-driver-1.2.58/displaylink-installer.sh starting on line 112.

rcw@antec:~/Projects/displaylink-debian/1.2.58/displaylink-driver-1.2.58$ systemctl status displaylink.service
Unit displaylink.service could not be found.
rcw@antec:~/Projects/displaylink-debian/1.2.58/displaylink-driver-1.2.58$ systemctl stat [... cut]