AdnanHodzic/displaylink-debian #24: displaylink kali 2016.2

panosd, 2016-09-30:
```
--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: I
Checking dependencies...
unzip is installed
linux-headers-4.6.0-kali1-amd64 is installed
dkms is installed
lsb-release is installed

Platform requirements satisfied, proceeding ...

Downloading DisplayLink Ubuntu driver:
--2016-09-30 22:46:20--  http://www.displaylink.com/downloads/file?id=701
Resolving www.displaylink.com (www.displaylink.com)... 52.48.31.214, 52.210.165.42, 54.229.186.74, ...
Connecting to www.displaylink.com (www.displaylink.com)|52.48.31.214|:80... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f701_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1475265431&Signature=V5ZrHGcm3EUYFe3G7kze9Y8Hi1Y%3D [following]
--2016-09-30 22:46:20--  http://assets.displaylink.com/live/downloads/software/f701_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1475265431&Signature=V5ZrHGcm3EUYFe3G7kze9Y8Hi1Y%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.48.108
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.48.108|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12757717 (12M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_1.2.58.zip’

DisplayLink_Ubuntu_1.2. 100%[==============================>]  12.17M   944KB/s    in 14s     

2016-09-30 22:46:34 (907 KB/s) - ‘DisplayLink_Ubuntu_1.2.58.zip’ saved [12757717/12757717]

Prepring for install ...

Archive:  DisplayLink_Ubuntu_1.2.58.zip
  inflating: 1.2.58/displaylink-driver-1.2.58.run  
  inflating: 1.2.58/LICENSE          
   creating: 1.2.58/__MACOSX/
  inflating: 1.2.58/__MACOSX/._LICENSE  
Creating directory displaylink-driver-1.2.58
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.2.58  100%  

Installing ... 

DisplayLink Linux Software 1.2.58 install script called: install
Distribution discovered: Kali GNU/Linux Rolling
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 4.6.0-kali1-amd64.
This is a fatal error, cannot install DisplayLink Linux Software.

Performing clean-up

Removing redundant: "DisplayLink_Ubuntu_1.2.58.zip" file
Removing redundant: "1.2.58" directory

Install complete, please reboot to apply the changes
```

any help because **linux-headers-4.6.0-kali1-amd64 is installed

deckrider, 2016-10-08:
I am on Debian 8.6 and had to make the following patch to get past this problem (I don't know if this is the right thing to do or not):

```
diff --git a/displaylink-debian.sh b/displaylink-debian.sh
index 1798572..34b5a75 100755
--- a/displaylink-debian.sh
+++ b/displaylink-debian.sh
@@ -196,6 +196,7 @@ sed -i "s/detect_distro/detect_distro/g" $driver_dir/displaylink-driver-${versio
 sed -i "s/detect_distro()/detect_distro()/g" $driver_dir/displaylink-driver-${version}/displaylink-installer.sh 
 sed -i "s/check_requirements/check_requirements/g" $driver_dir/displaylink-driver-${version}/displaylink-installer.sh
 sed -i "s/check_requirements()/check_requirements()/g" $driver_dir/displaylink-driver-${version}/displaylink-installer.sh
+sed -i "s/Kconfig/.config/g" $driver_dir/displaylink-driver-${version}/displaylink-installer.sh
 
 # install
 echo -e "\nInstalling ... \n"
```

AdnanHodzic, 2016-10-20:
I just pushed bunch of updates, one of them is ["Fixed Debian/Kali Linux install bug (failure to identify kernel headers)"](https://github.com/AdnanHodzic/displaylink-debian/commit/e1225f208e30e2aa1ace75969be6c81e479e14df)

Could you please pull the latest version and let me know if you're still facing same issue?

panosd, 2016-10-25:
--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.6.0-kali1-amd64 is installed
dkms is installed
lsb-release is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Downloading DisplayLink Ubuntu driver:

--2016-10-20 23:44:26--  http://www.displaylink.com/downloads/file?id=708
Resolving www.displaylink.com (www.displaylink.com)... 52.209.162.172, 52.48.31.214, 52.208.159.97, ...
Connecting to www.displaylink.com (www.displaylink.com)|52.209.162.172|:80... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f708_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1476996879&Signature=3Kwqaz33RR4nEnchoOU3V4VKxmk%3D [following]
--2016-10-20 23:44:26--  http://assets.displaylink.com/live/downloads/software/f708_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1476996879&Signature=3Kwqaz33RR4nEnchoOU3V4VKxmk%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 54.231.134.123
Connecting to assets.displaylink.com (assets.displaylink.com)|54.231.134.123|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12744086 (12M) [application/octet-stream]
Saving to: 'DisplayLink_Ubuntu_1.2.65.zip'

DisplayLink_Ubuntu_1.2.65. 100%[======================================>]  12.15M   939KB/s    in 14s

2016-10-20 23:44:40 (910 KB/s) - 'DisplayLink_Ubuntu_1.2.65.zip' saved [12744086/12744086]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_1.2.65.zip
  inflating: 1.2.65/displaylink-driver-1.2.65.run
  inflating: 1.2.65/LICENSE
Creating directory displaylink-driver-1.2.65
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.2.65  100%

-------------------------------------------------------------------

Installing driver version: 1.2.65

DisplayLink Linux Software 1.2.65 install script called: install
Distribution discovered: Kali GNU/Linux Rolling
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 4.6.0-kali1-amd64.
This is a fatal error, cannot install DisplayLink Linux Software.

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_1.2.65.zip" file
Removing redundant: "1.2.65" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------

it shows me the same error

________________________________
Από: Adnan Hodzic <notifications@github.com>
Στάλθηκε: Πέμπτη, 20 Οκτωβρίου 2016 11:40 μμ
Προς: AdnanHodzic/displaylink-debian
Κοιν.: panosd; Author
Θέμα: Re: [AdnanHodzic/displaylink-debian] displaylink kali 2016.2 (#24)

I just pushed bunch of updates, one of them is "Fixed Debian/Kali Linux install bug (failure to identify kernel headers)"<https://github.com/AdnanHodzic/displaylink-debian/commit/e1225f208e30e2aa1ace75969be6c81e479e14df>

Could you please pull the latest version and let me know if you're still facing same issue?

-

AdnanHodzic, 2016-10-26:
I just made another small change, which should fix the Debian headers issue/install.

Could you please pull the latest version and run the Re-install? 

Let me know if it works.

remoteprocess, 2016-11-19:
Also having the same issue with 

`Installing driver version: 1.2.65

DisplayLink Linux Software 1.2.65 install script called: install
Distribution discovered: Kali GNU/Linux Rolling
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 4.8.0-kali1-amd64.
This is a fatal error, cannot install DisplayLink Linux Software.
`
Any ideas or a fix?  Thanks!

AdnanHodzic, 2016-11-27:
I apologize for such delayed reply. It seems like this problem was part of "dlm.service" being enable to start (dead process). I just pushed changes which fix the problem with "enabling dlm.service". Now this is done as part of installation procedure. 

Please pull the latest change and run install/reinstall again and re-open the issue if you're still having same issues.

closed 2016-11-27 by AdnanHodzic