AdnanHodzic/displaylink-debian #27: Crashing when starting X

davirtavares, 2016-10-17:
Hi I used your script to install DisplayLink drivers, I had to make a little change to it:

```
diff --git a/displaylink-debian.sh b/displaylink-debian.sh
index 6cbb6ad..5aceb03 100755
--- a/displaylink-debian.sh
+++ b/displaylink-debian.sh
@@ -196,6 +196,7 @@ sed -i "s/detect_distro/detect_distro/g" $driver_dir/displaylink-driver-${versio
 sed -i "s/detect_distro()/detect_distro()/g" $driver_dir/displaylink-driver-${version}/displaylink-installer.sh 
 sed -i "s/check_requirements/check_requirements/g" $driver_dir/displaylink-driver-${version}/displaylink-installer.sh
 sed -i "s/check_requirements()/check_requirements()/g" $driver_dir/displaylink-driver-${version}/displaylink-installer.sh
+sed -i "/Linux headers for running kernel/ s?^?//?" $driver_dir/displaylink-driver-${version}/displaylink-installer.sh
 
 # install
 echo -e "\nInstalling ... \n"
```

This change fixes a problem on the newest DisplayLink driver, where it fails saying I don't have the kernel headers.

Anyway, with that change done, the script installs everything with no errors or warnings, then I restarted the system and tried to start X:

`xf86EnableIOPorts: failed to set IOPL for I/O (Operation not permitted)`

For being able to start X I have to unload the evdi module (rmmod evdi), but then I can't get my display working anyway.

Another way for being able to start my X server is to start to whole system with the DIsplayLink device UNPLUGGED, this way I can startx with any problem, however as soon as I plug the device, the xorg crashes with the same error.

I use Debian (testing/unstable) with kernel 4.7.

davirtavares, 2016-10-17:
Please let me know if I can do anything to help or any other info you need.

davirtavares, 2016-10-18:
Here is my dmesg output:

[log.txt](https://github.com/AdnanHodzic/displaylink-debian/files/536426/log.txt)

AdnanHodzic, 2016-10-20:
I just made bunch of updates, one of them was ["Fixed Debian/Kali Linux install bug (failure to identify kernel headers)"](https://github.com/AdnanHodzic/displaylink-debian/commit/e1225f208e30e2aa1ace75969be6c81e479e14df)

Latest version of driver (1.2.65) is also included, so you shouldn't be facing same problem anymore. I just verified this on Sid (4.7 kernel).

Could you please do the following:

1. Git pull the latest version
2. Run Uninstall
3. Reboot
4. Run Install again

This should fix your problem, in case you're still facing same issues please let me know/re-open the issue.

closed 2016-10-20 by AdnanHodzic

davirtavares, 2016-10-20:
Hi @AdnanHodzic, I posted the error also to the DisplayLink Linux support forum and they confirmed it was a bug specific do some monitor models:

http://displaylink.org/forum/showthread.php?t=64868

So the problem is not directly related to your tool, the kernel bug otherwise is, and it seems to be fixed now.

Thanks!

AdnanHodzic, 2016-10-20:
@davirtavares great, thanks for the update!