AdnanHodzic/displaylink-debian #28: linux-headers is wrongly declared as missing and blocks installation

pprw, 2016-10-19:
Hello,

I am unable to install display-link on my debian unstable. 

I have a fatal error message complaining about linux-headers but the linux headers for my kernel are installed !

How can I fix that?

error:
```
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 4.7.0-1-amd64.
This is a fatal error, cannot install DisplayLink Linux Software.
```
```
# apt-get install linux-headers-$(uname -r) dkms lsb-release
[...]
dkms is already the newest version (2.2.1.0+git20160527-1).
linux-headers-4.7.0-1-amd64 is already the newest version (4.7.6-1).
lsb-release is already the newest version (9.20160629).
```

Full log:
```
# ./bin/displaylink-debian-master/displaylink-debian.sh 

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: r
Checking dependencies...
unzip is installed
linux-headers-4.7.0-1-amd64 is installed
dkms is installed
lsb-release is installed

Platform requirements satisfied, proceeding ...

Uninstalling ...

./bin/displaylink-debian-master/displaylink-debian.sh: ligne 210: displaylink-installer : commande introuvable

Performing clean-up

Checking dependencies...
unzip is installed
linux-headers-4.7.0-1-amd64 is installed
dkms is installed
lsb-release is installed

Platform requirements satisfied, proceeding ...

Downloading DisplayLink Ubuntu driver:
--2016-10-19 11:10:46--  http://www.displaylink.com/downloads/file?id=708
Résolution de www.displaylink.com (www.displaylink.com)… 52.48.31.214, 52.208.159.97, 54.229.186.74, ...
Connexion à www.displaylink.com (www.displaylink.com)|52.48.31.214|:80… connecté.
requête HTTP transmise, en attente de la réponse… 303 See Other
Emplacement : http://assets.displaylink.com/live/downloads/software/f708_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1476868852&Signature=FzOwHbp0edcL%2B7T5RQ12L90gouU%3D [suivant]
--2016-10-19 11:10:47--  http://assets.displaylink.com/live/downloads/software/f708_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1476868852&Signature=FzOwHbp0edcL%2B7T5RQ12L90gouU%3D
Résolution de assets.displaylink.com (assets.displaylink.com)… 52.218.64.76
Connexion à assets.displaylink.com (assets.displaylink.com)|52.218.64.76|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 12744086 (12M) [application/octet-stream]
Sauvegarde en : « DisplayLink_Ubuntu_1.2.65.zip »

DisplayLink_Ubuntu_1.2.65.z 100%[===========================================>]  12,15M  2,59MB/s    in 4,9s    

2016-10-19 11:10:57 (2,48 MB/s) — « DisplayLink_Ubuntu_1.2.65.zip » sauvegardé [12744086/12744086]

Prepring for install ...

Archive:  DisplayLink_Ubuntu_1.2.65.zip
  inflating: 1.2.65/displaylink-driver-1.2.65.run  
  inflating: 1.2.65/LICENSE          
Creating directory displaylink-driver-1.2.65
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.2.65  100%  

Installing ... 

DisplayLink Linux Software 1.2.65 install script called: install
Distribution discovered: Debian GNU/Linux stable-updates (sid)
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 4.7.0-1-amd64.
This is a fatal error, cannot install DisplayLink Linux Software.

Performing clean-up

Removing redundant: "DisplayLink_Ubuntu_1.2.65.zip" file
Removing redundant: "1.2.65" directory

Re-install complete, please reboot to apply the changes

---------------------------------------------------------------------

```

yepla, 2016-10-19:
cd /lib/modules/your_kernel_version/build
ln -s Makefile Kconfig
dont forget to install linux-source for your kernel
than restart the installer
all will must be install
but i have a problem after : https://github.com/AdnanHodzic/displaylink-debian/issues/30

pprw, 2016-10-19:
Thank you. It worked !

And I don't have any problem after that to use my external monitor. Of course I have to launch manually xrandr every time I disconnect/reboot (like hotplug doesn't work on debian)  it but it works.

closed 2016-10-19 by pprw