AdnanHodzic/displaylink-debian #31: displaylink installed but xrandr --listproviders list nothing

yepla, 2016-10-19:
Hello
My driver is now installed...
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static)
   Active: active (running) since Wed 2016-10-19 16:13:32 CEST; 13s ago
  Process: 3234 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 3237 (DisplayLinkMana)
   CGroup: /system.slice/dlm.service
           └─3237 /opt/displaylink/DisplayLinkManager
but xrandr --listproviders give me:
Providers: number : 0
I m on xfce4 someone can help me ?

dragGH102, 2016-11-22:
@yepla I had the same issue. I noticed you closed the issue shortly afterwards opening it.
Did you manage to fix it?

AdnanHodzic, 2016-11-27:
I apologize for such delayed reply. I just pushed changes which fix the problem with "enabling dlm.service". Now this is done as part of installation procedure. 

Please pull the latest change and run install/reinstall again.

closed 2016-11-27 by AdnanHodzic