AdnanHodzic/displaylink-debian #35: gnome-control-center display not find my adapter

yepla, 2016-10-21:
$ cd displaylink-debian/
$ git pull
Already up-to-date.

###########################################################################

# ./displaylink-debian.sh 

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: u

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.7.0-1-amd64 is installed
dkms is installed
lsb-release is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

DisplayLink Linux Software 1.2.65 install script called: uninstall
Distribution discovered: Debian GNU/Linux 8.6 (jessie)
Uninstalling
Removing EVDI from kernel tree, DKMS, and removing sources.
Stopping DLM systemd service
Removing suspend-resume hooks
Removing udev rule
Removing Core folder

Uninstallation steps complete.

-------------------------------------------------------------------

Performing clean-up

-------------------------------------------------------------------

Uninstall complete

-------------------------------------------------------------------

###########################################################################

I have reboot then install again :

# ./displaylink-debian.sh 

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.7.0-1-amd64 is installed
dkms is installed
lsb-release is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Downloading DisplayLink Ubuntu driver:

--2016-10-21 14:11:24--  http://www.displaylink.com/downloads/file?id=708
Resolving www.displaylink.com (www.displaylink.com)... 54.229.186.74, 52.209.162.172, 52.48.31.214, ...
Connecting to www.displaylink.com (www.displaylink.com)|54.229.186.74|:80... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f708_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1477052491&Signature=SzbDPkR%2FavjwBMH7kR2iJ%2FpUoJo%3D [following]
--2016-10-21 14:11:24--  http://assets.displaylink.com/live/downloads/software/f708_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1477052491&Signature=SzbDPkR%2FavjwBMH7kR2iJ%2FpUoJo%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.17.4
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.17.4|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12744086 (12M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_1.2.65.zip’

DisplayLink_Ubuntu_1.2.65.zi 100%[==============================================>]  12.15M  6.01MB/s   in 2.0s   

2016-10-21 14:11:27 (6.01 MB/s) - ‘DisplayLink_Ubuntu_1.2.65.zip’ saved [12744086/12744086]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_1.2.65.zip
  inflating: 1.2.65/displaylink-driver-1.2.65.run  
  inflating: 1.2.65/LICENSE          
Creating directory displaylink-driver-1.2.65
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.2.65  100%  

-------------------------------------------------------------------

Installing driver version: 1.2.65

DisplayLink Linux Software 1.2.65 install script called: install
Distribution discovered: Debian GNU/Linux 8.6 (jessie)
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
Installing EVDI kernel module to kernel tree
EVDI kernel module built successfully
Installing x64-ubuntu-1604/DisplayLinkManager
Installing libraries
Installing firmware packages
Installing license file
Adding udev rule for DisplayLink DL-3xxx/5xxx devices

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_1.2.65.zip" file
Removing redundant: "1.2.65" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------

Then i reboot...

# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static; vendor preset: enabled)
   Active: inactive (dead)

# systemctl start dlm.service
# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static; vendor preset: enabled)
   Active: active (running) since Fri 2016-10-21 14:15:53 CEST; 1s ago
  Process: 4537 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 4542 (DisplayLinkMana)
   CGroup: /system.slice/dlm.service
           └─4542 /opt/displaylink/DisplayLinkManager

Oct 21 14:15:53 desktop systemd[1]: Starting DisplayLink Manager Service...
Oct 21 14:15:53 desktop systemd[1]: Started DisplayLink Manager Service.

# systemctl enable dlm.service
The unit files have no installation config (WantedBy, RequiredBy, Also, Alias
settings in the [Install] section, and DefaultInstance for template units).
This means they are not meant to be enabled using systemctl.
Possible reasons for having this kind of units are:
1) A unit may be statically enabled by being symlinked from another unit's
   .wants/ or .requires/ directory.
2) A unit's purpose may be to act as a helper for some other unit which has
   a requirement dependency on it.
3) A unit may be started when needed via activation (socket, path, timer,
   D-Bus, udev, scripted systemctl call, ...).
4) In case of template units, the unit is meant to be enabled with some
   instance name specified.

###########################################################################

Then i plug my adapter.

$ xrandr --listproviders 
Providers: number : 1
Provider 0: id: 0x1b6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

Then start gnome-control-center display

But it not find my adapter and my screen ! :(

yepla, 2016-10-23:
Hello,

I have reinstall a fresh debian jessie
I found a bug in your install script
For debian jessie and kernel 3.16.0-4-amd64 so the default kernel,
your script not copy the good file in /opt/displaylink
For debian jessie and kernel 3.16 it must copy:
DisplayLinkManager  libevdi.so  libusb-1.0.so.0.1.0
from x64-ubuntu-1404 and not x64-ubuntu-1604
or there are an error when starting DisplayLinkManager

At least now i m still experienced an issue for my adapter

$ xrandr --listproviders 
Providers: number : 0

yepla@desktop:~$ sudo systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static)
   Active: inactive (dead)

yepla@desktop:~$ sudo systemctl start dlm.service

yepla@desktop:~$ sudo systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static)
   Active: active (running) since Sun 2016-10-23 18:09:00 CEST; 1s ago
  Process: 2554 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 2559 (DisplayLinkMana)
   CGroup: /system.slice/dlm.service
           └─2559 /opt/displaylink/DisplayLinkManager

Oct 23 18:09:00 desktop systemd[1]: Started DisplayLink Manager Service.

yepla@desktop:~$ xrandr --listproviders 
Providers: number : 1
Provider 0: id: 0x1af cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

ISSUE: my adapter and his screen aren't recognized 

I have try to made an /etc/X11/xorg.conf.d/52-displaylink file:

Section "Device"
        Identifier "DVI-1"
        driver "fbdev"
        Option "fbdev" "/dev/fb0"
EndSection

Section "Monitor"
        Identifier "DisplayLinkMonitor"
	Modeline "1024x768_61.00"  65.26  1024 1080 1184 1344  768 769 772 796  -HSync +Vsync
EndSection

Section "Screen"
        Identifier "DisplayLinkScreen"
        Device "DVI-1"
        Monitor "DisplayLinkMonitor"
        SubSection "Display"
                Depth 16
                Modes "1024x768_61.00"
        EndSubSection
EndSection

But when i m start the dlm.service plug the adapter and logout i have this error in xorg.log:

[   647.273]
X.Org X Server 1.16.4
Release Date: 2014-12-20
[   647.273] X Protocol Version 11, Revision 0
[   647.273] Build Operating System: Linux 3.16.0-4-amd64 x86_64 Debian
[... 527 lines cut ...]
[   647.305] (EE) 2: /lib/x86_64-linux-gnu/libc.so.6 (0x7fedb8b7b000+0x350e0) [0x7fedb8bb00e0]
[   647.305] (EE) 3: /usr/bin/X (RRSetChanged+0x50) [0x7fedbafa16d0]
[   647.305] (EE) 4: /usr/bin/X (RRScreenSetSizeRange+0x54) [0x7fedbafa5c34]
[   647.305] (EE) 5: /usr/bin/X (xf86RandR12CreateScreenResources+0x2a5) [0x7fedbaf62925]
[   647.305] (EE) 6: /usr/bin/X (0x7fedbae89000+0xcc4f0) [0x7fedbaf554f0]
[   647.305] (EE) 7: /usr/bin/X (0x7fedbae89000+0x5b404) [0x7fedbaee4404]
[   647.305] (EE) 8: /lib/x86_64-linux-gnu/libc.so.6 (__libc_start_main+0xf5) [0x7fedb8b9cb45]
[   647.305] (EE) 9: /usr/bin/X (0x7fedbae89000+0x4590e) [0x7fedbaece90e]
[   647.305] (EE)
[   647.305] (EE) Segmentation fault at address 0xa0
[   647.305] (EE)
Fatal server error:
[   647.305] (EE) Caught signal 11 (Segmentation fault). Server aborting
[   647.305] (EE)
[   647.305] (EE)
Please consult the The X.Org Foundation support
     at http://wiki.x.org
 for help.
[   647.305] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   647.305] (EE)
[   647.308] (EE) Server terminated with error (1). Closing log file.

I really need help !

Thanks.

yepla, 2016-10-23:
xrandr -q
xrandr: Failed to get size of gamma for output default
Screen 0: minimum 640 x 480, current 1280 x 1024, maximum 1280 x 1024
default connected 1280x1024+0+0 0mm x 0mm
   1280x1024      0.00* 
   1024x768       0.00  
   800x600        0.00  
   640x480        0.00

It s all that i have when my adapter is connect :(

AdnanHodzic, 2016-10-26:
I just made another small change, which should fix the Debian headers issue/install.

Could you please pull the latest version and run the Re-install? 

Let me know if it works.

yepla, 2016-10-27:
i m not anymore on debian jessie
but if you have change the cp line to copy DisplayManager file from 1404 it will be good
at least now i m on debian stretch i have install your driver
but xrandr -q not showing my display adapter output
but it s seen with xrandr --listadapters
...
i dont know what to do :(

yepla, 2016-10-27:
it s work on debian stretch !!! coool

yepla, 2016-10-27:
Acutally all it s working on debian stretch !

closed 2016-10-27 by yepla

AdnanHodzic, 2016-10-27:
Great :)