AdnanHodzic/displaylink-debian #37: systemctl status dlm.service failure

markd89, 2016-11-07:
Hi Adnan,

I tried the new version of the script and make it much further.

**There are no errors during the install, but I can't get the service to start after doing the reboot:**

root@ux305:/home/mark# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static)
   Active: inactive (dead)
root@ux305:/home/mark# systemctl start dlm.service
root@ux305:/home/mark# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2016-11-07 06:28:28 PST; 3s ago
  Process: 1912 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 1908 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 **Main PID: 1912 (code=exited, status=1/FAILURE)
Nov 07 06:28:28 ux305 systemd[1]: Unit dlm.service entered failed state.**

**I am using a pretty plain Debian Jessie 8.6.**

Following is a **log of the install:**

root@ux305:/home/mark/Downloads/displaylink20161107# ./displaylink-debian.sh

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-3.16.0-4-amd64 is installed
dkms not found! Install? [y/N] y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  bogofilter bogofilter-bdb bogofilter-common empathy-common gir1.2-gtop-2.0
  gnome-tweak-tool gstreamer1.0-nice hyphen-en-us libfarstream-0.2-2
  libgtkhtml-4.0-0 libgtkhtml-4.0-common libgtkhtml-editor-4.0-0
  libgtkspell3-3-0 libpst4 libreoffice libreoffice-help-en-us
  libtelepathy-farstream3 libytnef0 mythes-en-us sound-theme-freedesktop
  telepathy-gabble telepathy-salut
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  fakeroot libfakeroot linux-headers-amd64
Recommended packages:
  linux-image
The following NEW packages will be installed:
  dkms fakeroot libfakeroot linux-headers-amd64
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 205 kB of archives.
After this operation, 786 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ftp.us.debian.org/debian/ jessie/main dkms all 2.2.0.3-2 [70.9 kB]
Get:2 http://ftp.us.debian.org/debian/ jessie/main libfakeroot amd64 1.20.2-1 [44.7 kB]
Get:3 http://ftp.us.debian.org/debian/ jessie/main fakeroot amd64 1.20.2-1 [84.7 kB]
Get:4 http://ftp.us.debian.org/debian/ jessie/main linux-headers-amd64 amd64 3.16+63 [5,050 B]
Fetched 205 kB in 1s (107 kB/s)                
Selecting previously unselected package dkms.
(Reading database ... 151858 files and directories currently installed.)
Preparing to unpack .../dkms_2.2.0.3-2_all.deb ...
Unpacking dkms (2.2.0.3-2) ...
Selecting previously unselected package libfakeroot:amd64.
Preparing to unpack .../libfakeroot_1.20.2-1_amd64.deb ...
Unpacking libfakeroot:amd64 (1.20.2-1) ...
Selecting previously unselected package fakeroot.
Preparing to unpack .../fakeroot_1.20.2-1_amd64.deb ...
Unpacking fakeroot (1.20.2-1) ...
Selecting previously unselected package linux-headers-amd64.
Preparing to unpack .../linux-headers-amd64_3.16+63_amd64.deb ...
Unpacking linux-headers-amd64 (3.16+63) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up dkms (2.2.0.3-2) ...
Setting up libfakeroot:amd64 (1.20.2-1) ...
Setting up fakeroot (1.20.2-1) ...
update-alternatives: using /usr/bin/fakeroot-sysv to provide /usr/bin/fakeroot (fakeroot) in auto mode
Setting up linux-headers-amd64 (3.16+63) ...
lsb-release is installed
linux-source not found! Install? [y/N] y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  bogofilter bogofilter-bdb bogofilter-common empathy-common gir1.2-gtop-2.0
  gnome-tweak-tool gstreamer1.0-nice hyphen-en-us libfarstream-0.2-2
  libgtkhtml-4.0-0 libgtkhtml-4.0-common libgtkhtml-editor-4.0-0
  libgtkspell3-3-0 libpst4 libreoffice libreoffice-help-en-us
  libtelepathy-farstream3 libytnef0 mythes-en-us sound-theme-freedesktop
  telepathy-gabble telepathy-salut
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  linux-source-3.16
Suggested packages:
  **libncurses-dev ncurses-dev libqt4-dev pkg-config**
The following NEW packages will be installed:
  linux-source linux-source-3.16
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 83.7 MB of archives.
After this operation, 83.7 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ftp.us.debian.org/debian/ jessie/main linux-source all 3.16+63 [5,014 B]
Get:2 http://security.debian.org/ jessie/updates/main linux-source-3.16 all 3.16.36-1+deb8u2 [83.7 MB]
Fetched 83.7 MB in 1min 14s (1,118 kB/s)                                       
Selecting previously unselected package linux-source-3.16.
(Reading database ... 151956 files and directories currently installed.)
Preparing to unpack .../linux-source-3.16_3.16.36-1+deb8u2_all.deb ...
Unpacking linux-source-3.16 (3.16.36-1+deb8u2) ...
Selecting previously unselected package linux-source.
Preparing to unpack .../linux-source_3.16+63_all.deb ...
Unpacking linux-source (3.16+63) ...
Setting up linux-source-3.16 (3.16.36-1+deb8u2) ...
Setting up linux-source (3.16+63) ...

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Downloading DisplayLink Ubuntu driver:

--2016-11-07 06:18:08--  http://www.displaylink.com/downloads/file?id=708
Resolving www.displaylink.com (www.displaylink.com)... 52.48.31.214, 54.229.186.74, 52.208.159.97, ...
Connecting to www.displaylink.com (www.displaylink.com)|52.48.31.214|:80... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f708_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1478528878&Signature=1B3SgyFuCQQsMlKoR751blsyBTk%3D [following]
--2016-11-07 06:18:09--  http://assets.displaylink.com/live/downloads/software/f708_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.2.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1478528878&Signature=1B3SgyFuCQQsMlKoR751blsyBTk%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.16.60
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.16.60|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12744086 (12M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_1.2.65.zip’

DisplayLink_Ubuntu_ 100%[=====================>]  12.15M  1.13MB/s   in 12s    

2016-11-07 06:18:22 (1.03 MB/s) - ‘DisplayLink_Ubuntu_1.2.65.zip’ saved [12744086/12744086]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_1.2.65.zip
  inflating: 1.2.65/displaylink-driver-1.2.65.run  
  inflating: 1.2.65/LICENSE          
Creating directory displaylink-driver-1.2.65
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.2.65  100%  

-------------------------------------------------------------------

Installing driver version: 1.2.65

DisplayLink Linux Software 1.2.65 install script called: install
Distribution discovered: Debian GNU/Linux 8.6 (jessie)
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
Installing EVDI kernel module to kernel tree
EVDI kernel module built successfully
Installing x64-ubuntu-1604/DisplayLinkManager
Installing libraries
Installing firmware packages
Installing license file
Adding udev rule for DisplayLink DL-3xxx/5xxx devices

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_1.2.65.zip" file
Removing redundant: "1.2.65" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------

akersting, 2016-11-07:
I encountered the same problem. Trying to run /opt/displaylink/DisplayLinkManager directly gives:

/opt/displaylink/DisplayLinkManager: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /opt/displaylink/DisplayLinkManager)

This version is indeed not provided by the package libstdc++6 on jessie (see the output of `strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIB`).

A quick fix is to place the file libstdc++.so.6.0.22 from the stretch package (https://packages.debian.org/stretch/libstdc++6) in /opt/displaylink/ and add a soft link: `sudo ln -s /opt/displaylink/libstdc++.so.6.0.22 /opt/displaylink/libstdc++.so.6`.

This problem should have been introduced by the recent upstream update of the driver. An older version of the driver was working for me without this hack.

markd89, 2016-11-08:
Thank you!!

The hack worked and the service now starts. I have other issues but I'll make a new thread.

Thanks again,
Mark

closed 2016-11-08 by markd89

dragGH102, 2016-11-16:
@akersting I  ran in the same issue: where do I find that specific file? I tried to go at the link but couldn't figure out. thank you

saljut7, 2016-11-17:
@dragGH102 thanks for your notification. I just took a look with:

```
# apt-cache policy libstdc++6
libstdc++6:
  Installed: 4.9.2-10
  Candidate: 4.9.2-10
  Version table:
     7-20161116-1 0
          1 http://ftp.de.debian.org/debian/ experimental/main amd64 Packages
     6.2.0-13 0
        750 http://ftp.de.debian.org/debian/ testing/main amd64 Packages
         50 http://ftp.de.debian.org/debian/ unstable/main amd64 Packages
 *** 4.9.2-10 0
        995 http://ftp.de.debian.org/debian/ jessie/main amd64 Packages
        995 http://ftp.de.debian.org/debian/ stable/main amd64 Packages
        100 /var/lib/dpkg/status
```

So as far as I can see you can only install it (as mentioned by @akersting ) from testing repository (stretch). If you don't see the package you have to configure [apt-pinning](http://serverfault.com/questions/22414/how-can-i-run-debian-stable-but-install-some-packages-from-testing ) (which I did that's why I can see the package but I would **not** recommend it: [Repo mix-and-match hit parade](http://forums.debian.net/viewtopic.php?f=20&t=114130&hilit=mixed+repoSo)).

akersting, 2016-11-21:
I just downloaded the .deb-file from https://packages.debian.org/stretch/libstdc++6 (there is a section "Download libstdc++6" there), unpacked it and copied the file libstdc++.so.6.0.22 to /opt/displaylink/ as described above. For amd64 this file will be in folder2whichYouUnpackedTheDEB/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.22.

dragGH102, 2016-11-22:
thanks @akersting it worked!

dragGH102, 2016-11-22:
thanks @akersting it worked!
After doing this could you see the "2nd provider?
Because I can't still see it

markd89, 2016-11-22:
On one of my displaylink drivers, I don't think I see the 2nd provider. This is an external box that goes between my laptop and the the external monitor. I posted my results here:
https://github.com/AdnanHodzic/displaylink-debian/issues/38

I have a second displaylink display. This one is built into a 15" ASUS display. On this one, I see the 2nd provider but have not been able to activate it. I documented here:
https://github.com/AdnanHodzic/displaylink-debian/issues/39

If it's not working for you either, maybe this hack is the cause and we need some other ideas?

Do your xandr --listproviders look like either of the two issues I linked above?

saljut7, 2016-11-23:
@akersting thanks so much. Worked for me - but only partially...

A Lenovo X201 had no problems displaying a screen with the displaylink adapter.
But on a Lenovo T60 after activating via `xrandr --output DVI-1-1 --mode 1280x1024 --pos 1400x0 --rotate normal` something strange happens: _Debian8.6 is "jumping" back to login screen._

Both systems run Debian8.6 (AMD64). Anybody noticed a similar behaviour?

markd89, 2016-11-24:
Yes, I saw something similar. It didn't just go back to the gnome login screen but killed all of the applications running...

dragGH102, 2016-11-24:
@markd89 

About https://github.com/AdnanHodzic/displaylink-debian/issues/38, yes, at least for **listproviders** the issue seems the same. But I see no solution htere.

https://github.com/AdnanHodzic/displaylink-debian/issues/39: instead doesn't help as I don't see the 2nd providers (and therefore can't try to activate it all)

Any clue how to get the 2nd provider to show up?
@saljut7 you are having some issues. but have ever had **1 provider only** issue?

saljut7, 2016-11-24:
@markd89 thank you for your reply. This looks very similar to my case. Could you solve the problem somehow?

@dragGH102 Sorry I have never experienced "1 provider only" issue. :/

markd89, 2016-11-25:
No, mine is still not working (I'm still stuck using one LCD).

I'm not sure if our errors are due to issues with the Debian port or maybe they are inherent to Displaylink's original Ubuntu release. When I have more time, I thought I would go look at their support forum and see if anyone has any solutions. If you do that and get things working, please post back here too...

AdnanHodzic, 2016-11-27:
Hi everyone, 

I apologize for such delayed reply. I just pushed changes which fix the problem with "enabling dlm.service". Now this is done as part of installation procedure.

markd89, 2016-11-27:
I uninstalled then reinstalled with the new script.

1. FYI: There was an error during the script ln: failed to create symbolic link ‘/lib/modules/3.16.0-4-amd64/build/Kconfig’:  because the link already existed. The install seemed to complete OK, but I re-removed the link manually, uninstalled and reinstalled and didn't see the error the second time.

2. After rebooting, checked the service and could not get it to start. Here's what I did:

root@ux305:/home/mark# **systemctl status dlm.service**
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2016-11-27 15:01:32 PST; 330ms ago
  Process: 6199 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 6197 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 6199 (code=exited, status=1/FAILURE)

Nov 27 15:01:32 ux305 systemd[1]: Unit dlm.service entered failed state.
root@ux305:/home/mark# **systemctl start dlm.service**
root@ux305:/home/mark# **systemctl status dlm.service**
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2016-11-27 15:01:53 PST; 3s ago
  Process: 6218 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 6216 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 6218 (code=exited, status=1/FAILURE)

Nov 27 15:01:53 ux305 systemd[1]: Unit dlm.service entered failed state.

markd89, 2016-11-27:
[  213.740710] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[  213.756999] usb 2-1: New USB device found, idVendor=17e9, idProduct=ff03
[  213.757003] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  213.757005] usb 2-1: Product: MB168B+
[  213.757006] usb 2-1: Manufacturer: DisplayLink
[  213.757007] usb 2-1: SerialNumber: E7LMTF092982
[  264.229608] usb 2-1: USB disconnect, device number 4
[  275.858429] usb 2-1: new SuperSpeed USB device number 5 using xhci_hcd
[  275.875154] usb 2-1: New USB device found, idVendor=17e9, idProduct=ff03
[  275.875163] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  275.875167] usb 2-1: Product: MB168B+
[  275.875171] usb 2-1: Manufacturer: DisplayLink
[  275.875174] usb 2-1: SerialNumber: E7LMTF092982

saljut7, 2016-12-01:
@AdnanHodzic thanks alot for your great support and providing the script. I tried it with the updated version again on two Devices running Debian8.6. Just to confirm:

1. Installation
2. Reboot
3. Plugged in DisplayLink adapter
4. Checked, started and again checked service:

```
# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2016-11-29 17:08:32 CET; 799ms ago
  Process: 1518 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 1515 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1518 (code=exited, status=1/FAILURE)

Nov 29 17:08:32 T60-Notebook systemd[1]: Unit dlm.service entered fa....
Hint: Some lines were ellipsized, use -l to show in full.
root@T60-Notebook:~# systemctl start dlm.service
root@T60-Notebook:~# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2016-11-29 17:08:42 CET; 2s ago
  Process: 1626 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 1624 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1626 (code=exited, status=1/FAILURE)

Nov 29 17:08:42 T60-Notebook systemd[1]: Unit dlm.service entered fa....
Hint: Some lines were ellipsized, use -l to show in full.
root@T60-Notebook:~# journalctl -xn
-- Logs begin at Tue 2016-11-29 17:07:19 CET, end at Tue 2016-11-29
Nov 29 17:08:47 T60-Notebook systemd[1]: dlm.service: main process exite
Nov 29 17:08:47 T60-Notebook systemd[1]: Unit dlm.service entered failed
Nov 29 17:08:47 T60-Notebook DisplayLinkManager[1654]: /opt/displaylink/
Nov 29 17:08:53 T60-Notebook systemd[1]: dlm.service holdoff time over, 
Nov 29 17:08:53 T60-Notebook systemd[1]: Stopping DisplayLink Manager Se
-- Subject: Unit dlm.service has begun shutting down
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-d
-- 
-- Unit dlm.service has begun shutting down.
Nov 29 17:08:53 T60-Notebook systemd[1]: Starting DisplayLink Manager Se
-- Subject: Unit dlm.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-d
-- 
-- Unit dlm.service has begun starting up.
Nov 29 17:08:53 T60-Notebook systemd[1]: Started DisplayLink Manager Ser
-- Subject: Unit dlm.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-d
-- 
-- Unit dlm.service has finished starting up.
-- 
-- The start-up result is done.
Nov 29 17:08:53 T60-Notebook DisplayLinkManager[1666]: /opt/displaylink/
Nov 29 17:08:53 T60-Notebook systemd[1]: dlm.service: main process exite
Nov 29 17:08:53 T60-Notebook systemd[1]: Unit dlm.service entered failed
lines 1-28/28 (END)...skipping...
-- Logs begin at Tue 2016-11-29 17:07:19 CET, end at Tue 2016-11-29 17:08:53 CET. --
Nov 29 17:08:47 T60-Notebook systemd[1]: dlm.service: main process exited, code=exited, status=1/FAILURE
Nov 29 17:08:47 T60-Notebook systemd[1]: Unit dlm.service entered failed state.
Nov 29 17:08:47 T60-Notebook DisplayLinkManager[1654]: /opt/displaylink/DisplayLinkManager: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `
Nov 29 17:08:53 T60-Notebook systemd[1]: dlm.service holdoff time over, scheduling restart.
Nov 29 17:08:53 T60-Notebook systemd[1]: Stopping DisplayLink Manager Service...
-- Subject: Unit dlm.service has begun shutting down
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit dlm.service has begun shutting down.
Nov 29 17:08:53 T60-Notebook systemd[1]: Starting DisplayLink Manager Service...
-- Subject: Unit dlm.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit dlm.service has begun starting up.
Nov 29 17:08:53 T60-Notebook systemd[1]: Started DisplayLink Manager Service.
-- Subject: Unit dlm.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit dlm.service has finished starting up.
-- 
-- The start-up result is done.
Nov 29 17:08:53 T60-Notebook DisplayLinkManager[1666]: /opt/displaylink/DisplayLinkManager: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `
Nov 29 17:08:53 T60-Notebook systemd[1]: dlm.service: main process exited, code=exited, status=1/FAILURE
Nov 29 17:08:53 T60-Notebook systemd[1]: Unit dlm.service entered failed state.
```

So still "Unit dlm.service entered failed state.". On this notebook I didn't try the workaround of @akersting before. 

Any ideas?

AdnanHodzic, 2016-12-04:
@saljut7  could you [please do workaround I suggested in #42](https://github.com/AdnanHodzic/displaylink-debian/issues/42#issuecomment-264711774)? This fixes the issue with 3.16 kernel on Jessie.

AdnanHodzic, 2016-12-10:
[in latest commit](https://github.com/AdnanHodzic/displaylink-debian/commit/1744ad3f621e69a2fdcefbd824f9e63c7a536b28) "libstdc++.so.6 hack" has been made part of the installer for Debian.