AdnanHodzic/displaylink-debian #38: --listproviders: interpreting results

markd89, 2016-11-08:
Following, I start the service and query providers. The return says "2 associated providers" but I'm pretty sure there should be another line after Provider 0. I am using Debian Jessie.

I'd appreciate any advice on how to troubleshoot:

root@ux305:/home/mark# systemctl **status dlm.service**
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static)
   Active: inactive (dead)
root@ux305:/home/mark# systemctl **start dlm.service**
root@ux305:/home/mark# systemctl **status dlm.service**
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static)
   Active: active (running) since Tue 2016-11-08 10:23:39 PST; 3s ago
  Process: 3326 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 3332 (DisplayLinkMana)
   CGroup: /system.slice/dlm.service
           └─3332 /opt/displaylink/DisplayLinkManager

Nov 08 10:23:39 ux305 systemd[1]: Started DisplayLink Manager Service.
**root@ux305:/home/mark# xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x44 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:Intel**

markd89, 2016-11-08:
FYI: piece of dmesg results:

[    3.080613] udlfb: DisplayLink WS Tech USB-DVI - serial #154782
[    3.080620] udlfb: vid_17e9&pid_0198&rev_0013 driver's dlfb_data struct at ffff8800d5edf800
[    3.080624] udlfb: console enable=1
[    3.080626] udlfb: fb_defio enable=1
[    3.080627] udlfb: shadow enable=1
[    3.080726] udlfb: vendor descriptor length:29 data:29 5f 01 0027 00 04 04 01 00 03
[    3.080729] udlfb: DL chip limited to 2360000 pixel modes
[    3.080779] udlfb: allocated 4 65024 byte urbs
[    3.080834] usbcore: registered new interface driver udlfb
[    3.082490] usb 1-3.4: new low-speed USB device number 7 using xhci_hcd
[    3.084322] usbcore: registered new interface driver udl
[    3.160483] udlfb: 1920x1080 @ 60 Hz valid mode
[    3.160489] udlfb: 720x400 @ 70 Hz valid mode
[    3.160492] udlfb: 640x480 @ 60 Hz valid mode
[    3.160495] udlfb: 640x480 @ 67 Hz valid mode
[    3.160497] udlfb: 640x480 @ 72 Hz valid mode
[    3.160500] udlfb: 640x480 @ 75 Hz valid mode
[    3.160502] udlfb: 800x600 @ 56 Hz valid mode
[    3.160506] udlfb: 800x600 @ 60 Hz valid mode
[    3.160509] udlfb: 800x600 @ 72 Hz valid mode
[    3.160511] udlfb: 800x600 @ 75 Hz valid mode
[    3.160514] udlfb: 832x624 @ 75 Hz valid mode
[    3.160517] udlfb: 1024x768 @ 60 Hz valid mode
[    3.160519] udlfb: 1024x768 @ 70 Hz valid mode
[    3.160522] udlfb: 1024x768 @ 75 Hz valid mode
[    3.160524] udlfb: 1280x1024 @ 75 Hz valid mode
[    3.160527] udlfb: 1152x864 @ 75 Hz valid mode
[    3.160530] udlfb: 1680x1050 @ 60 Hz valid mode
[    3.160532] udlfb: 1600x1200 @ 60 Hz valid mode
[    3.160535] udlfb: 1600x900 @ 60 Hz valid mode
[    3.160538] udlfb: 1440x900 @ 60 Hz valid mode
[    3.160540] udlfb: 1400x1050 @ 60 Hz valid mode
[    3.160543] udlfb: 1280x1024 @ 60 Hz valid mode
[    3.160546] udlfb: 1280x960 @ 60 Hz valid mode
[    3.160549] udlfb: 1152x864 @ 75 Hz valid mode
[    3.160551] udlfb: Reallocating framebuffer. Addresses will change!
[    3.162651] udlfb: 1920x1080 @ 60 Hz valid mode
[    3.162656] udlfb: set_par mode 1920x1080
[    3.174033] udlfb: DisplayLink USB device /dev/fb1 attached. 1920x1080 resolution. Using 8104K framebuffer memory
[    3.222655] /dev/vmmon[681]: Module vmmon: registered with major=10 minor=165
[    3.222663] /dev/vmmon[681]: Module vmmon: initialized

AdnanHodzic, 2016-11-27:
Hi Mark,

I apologize for such delayed reply. I just pushed changes which fix the problem with "enabling dlm.service". Now this is done as part of installation procedure. 

Please pull the latest change and run install/reinstall again.

closed 2016-11-27 by AdnanHodzic