AdnanHodzic/displaylink-debian #39: --listproviders sees two but cannot activate second screen

markd89, 2016-11-10:
I created a new Issue as I am trying with another, different, displaylink adapter and seeing different, better :-) results.

**The good news is that I am now seeing two providers, but I cannot seem to activate the second display. I think I may be missing a step?**

root@ux305:/home/mark# **xrandr --listproviders**
Providers: number : **2**
Provider 0: id: 0x74 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:Intel
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

**root@ux305:/home/mark# xrandr -q**
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 8192 x 8192
eDP1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 293mm x 165mm
   1920x1080     60.00*+  59.93  
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1280x1024     60.02  
   1440x900      59.89  
   1280x960      60.00  
   1360x768      59.80    59.96  
   1152x864      60.00  
   1024x768      60.00  
   800x600       60.32    56.25  
   640x480       59.94  
HDMI1 disconnected (normal left inverted right x axis y axis)

markd89, 2016-11-27:
So, still no luck.

I am using kernel 3.16.0-4-amd64 on Debian Jessie 8.6. 

I know @AdnanHodzic  is using a newer kernel on his machine. I'm wondering if something got borked in Displaylink's latest release or maybe there's something not happy in Adnan's script because I'm using an "older" kernel. 

Would it be worth trying an older version of the script and editing it to use the older version of Displaylink's stuff? If so, how would I do that? 

Thanks much!
Mark

AdnanHodzic, 2016-11-27:
Hi Mark, 

I just replied in your other issue. It seems like the problem was because dlm.service couldn't be enabled. 

I just pushed changes which fix the problem with "enabling dlm.service". Please pull the latest change and run install/reinstall again and let me know/re-open the issue if you're still having same issue.

Adnan

closed 2016-11-27 by AdnanHodzic