AdnanHodzic/displaylink-debian #40: Failed to start Load Kernel Modules on Debian8.6

saljut7, 2016-11-13:
Hi,

thank you so much for this awesome script!

I tried a [Club3D CSV-2300D](http://www.club-3d.com/index.php/products/reader.en/product/usb-30-to-dvi-graphics-adapter.html) adapter in a shop before buying it and found your script here looking for a way to activate it. Club3D refers to the [DisplayLink Ubuntu drivers](http://www.displaylink.com/downloads/ubuntu.php) so I thought your script should work for this USB-DVI-adapter as well.

In the end the script was just working fine until reboot. Unfortunately I got an error while rebooting:
```
[FAILED] Failed to start Load Kernel Modules.
See `systemctl status systemd-modules-load.service` for details.
[...] A start job is running for udev Wait for complete Device Initialization [...]
```

So I tried to activate dlm.service manually but got:

```
# systemctl start dlm.service
# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2016-11-13 05:01:11 CET; 2s ago
  Process: 2093 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 2091 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 2093 (code=exited, status=1/FAILURE)

Nov 13 05:01:11 Debian8-Notebook systemd[1]: Unit dlm.service entered failed state.
```

Any ideas what I did wrong?

Best regards
sj7

dragGH102, 2016-11-17:
same (or maybe similar) issue here.
I opened another issue just yesterday: https://github.com/AdnanHodzic/displaylink-debian/issues/41

AdnanHodzic, 2016-11-27:
I apologize for such delayed reply. I just pushed changes which fix the problem with "enabling dlm.service". Now this is done as part of installation procedure.

Please pull the latest change and run install/reinstall again.

closed 2016-11-27 by AdnanHodzic