AdnanHodzic/displaylink-debian #41: Debian 8.5 issues (process start, providers)

dragGH102, 2016-11-16:
Note: My system is using Bumblebee (http://bumblebee-project.org/) to correctly support standby but I don't see why it could be the issue since any screen via HDMI port (including with HDMI to VGA adapter) just works

The installation went fine, I restarted the system and followed the post-installation guide.

The issues I'm having are (besides clearly the fact the 2nd external screen is not detected, not even if it's the only screen connected without the 1st one via HDMI):

1)  status dlm.service # **error on DisplayLinkManager process**
```
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; static)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2016-11-16 18:46:27 CET; 4s ago
  Process: 12456 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 12454 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 12456 (code=exited, status=1/FAILURE)
```

I also tried to restart (stop / start) the service, same result.

2) xrandr --listproviders # **only 1 provider found**
```
Providers: number : 1
Provider 0: id: 0x45 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:Intel
```
Same output, regardless the 1st screen is connected via HDMI or not

I didn't try the following instructions since these 2 already give issues.

Thanks in advance.

dragGH102, 2016-11-16:
**Update**: apparently the first issue *may* be solved as explained @ https://github.com/AdnanHodzic/displaylink-debian/issues/37

But I don't understand where to find that file (I replied to that issue but not sure whether it's still monitored or not)

dragGH102, 2016-11-22:
**Update 2**: now the first issue is solved (thanks @akersting).
The 2nd issue (only 1 provider) is still there though

dragGH102, 2016-11-24:
**Update 3**: For some reason now 2 providers show up:
```

Providers: number : 2
Provider 0: id: 0x45 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:Intel
Provider 1: id: 0xff cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

```
However when I try to attach the displaylink adapter to my (1st) external screen I can move the mouse to the other screen.. but the screen remains black.

I tried to play a bit with **arandr** by trying different resolutions and positions (this actually seems to work to change position or resolution but doesn't solve the issue)
I also tried **xrandr --output HDMI1 --auto --right-of eDP1**.

Same result.
I didn't try with 2 2nd external screen but the display doesn't show up even for the 1st one, unless I use directly my computer's HDMI port.

Any clue? Am I doing something wrong here?

AdnanHodzic, 2016-11-27:
Hi,

I apologize for such delayed reply. I just pushed changes which fix the problem with "enabling dlm.service". Now this is done as part of installation procedure. 

Could you please pull the latest change and run install/reinstall again?

Regarding your second issue, this is concerning your display setup and not really problem with displaylink-debian?

If you're reporting it's grayed out, did you try running it with "sudo" to see if it's a permission issue?

Please re-open this issue if you're still having same problem.

closed 2016-11-27 by AdnanHodzic

dragGH102, 2016-11-27:
Dear @AdnanHodzic, 
thanks for your follow-up. 

I noticed you closed all the existing issues with the same message. However for example the issue I (and a couple of other issues) am facing is not about the service itself but xrand list providers. 

Can you please clarify why the issue was closed? 

Thank you

AdnanHodzic, 2016-11-27:
I closed all of them since they all seemed to be related to same problem which was fixed in last commit. 

Furthermore, your issue seems to be of a driver problem itself, and I'd advise you to[ report the problem there.](https://github.com/DisplayLink/evdi/issues)

Please feel free to re-open the issue if you have any updates that are directly concerning displaylink-debian.