AdnanHodzic/displaylink-debian #42: systemctl status dlm.service failure (2016-11-27 build)

markd89, 2016-11-27:
I uninstalled then reinstalled with the new script.

    FYI: There was an error during the script ln: failed to create symbolic link ‘/lib/modules/3.16.0-4-amd64/build/Kconfig’: because the link already existed. The install seemed to complete OK, but I re-removed the link manually, uninstalled and reinstalled and didn't see the error the second time.

    After rebooting, checked the service and could not get it to start. Here's what I did:

root@ux305:/home/mark# **systemctl status dlm.service**
● dlm.service - DisplayLink Manager Service
Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
Active: activating (auto-restart) (Result: exit-code) since Sun 2016-11-27 15:01:32 PST; 330ms ago
Process: 6199 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
Process: 6197 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
Main PID: 6199 (code=exited, status=1/FAILURE)

Nov 27 15:01:32 ux305 systemd[1]: Unit dlm.service entered failed state.
root@ux305:/home/mark# **systemctl start dlm.service**
root@ux305:/home/mark# **systemctl status dlm.service**
● dlm.service - DisplayLink Manager Service
Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
Active: activating (auto-restart) (Result: exit-code) since Sun 2016-11-27 15:01:53 PST; 3s ago
Process: 6218 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
Process: 6216 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
Main PID: 6218 (code=exited, status=1/FAILURE)

Nov 27 15:01:53 ux305 systemd[1]: Unit dlm.service entered failed state.

**dmesg:**
[ 213.740710] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[ 213.756999] usb 2-1: New USB device found, idVendor=17e9, idProduct=ff03
[ 213.757003] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 213.757005] usb 2-1: Product: MB168B+
[ 213.757006] usb 2-1: Manufacturer: DisplayLink
[ 213.757007] usb 2-1: SerialNumber: E7LMTF092982
[ 264.229608] usb 2-1: USB disconnect, device number 4
[ 275.858429] usb 2-1: new SuperSpeed USB device number 5 using xhci_hcd
[ 275.875154] usb 2-1: New USB device found, idVendor=17e9, idProduct=ff03
[ 275.875163] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 275.875167] usb 2-1: Product: MB168B+
[ 275.875171] usb 2-1: Manufacturer: DisplayLink
[ 275.875174] usb 2-1: SerialNumber: E7LMTF092982

redjamesg, 2016-12-02:
This is related to #43

markd89, 2016-12-03:
@AdnanHodzic - would it be possible to have access to an older version of your script which downloads an older version of the Displaylink stuff? Ideally, we'd want to use the latest, but if that's difficult maybe running the older stuff until you have time to dig into the newer stuff more would be a workaround?

I appreciate all of your work on this :-)

Thanks,
Mark

AdnanHodzic, 2016-12-04:
@markd89 @redjamesg with upcoming Stretch freeze and hence new stable release just around the corner. I really don't have time to spend on debugging issues on old kernel. 

So as a workaround, instead of using old code (unsupported by both me and upstream) could you please do following. Install 4.7 kernel from Jessie backports:

**1.** Add to sources.list: 

`deb http://ftp.debian.org/debian jessie-backports main`

**2.** Update sources and install kernel 4.7

`sudo apt-get update && sudo apt-get install -t jessie-backports linux-image-4.7.0-0.bpo.1-amd64`

After this simply run displaylink-debian re-install.

I just did this on on Debian Jessie 8.6 and it worked like a charm.

markd89, 2016-12-08:
Thanks Adnan. I can't upgrade the kernel right now because it will break something else (VMWare Player Unity Mode), which I need for work. It's possible for me to hack on that and probably get it working, then I could upgrade the kernel... or I could hack on one of the older revs of this project and see if I can make it work with the current kernel. Probably I'll do the former as like you said - a newer kernel is coming soon anyway to mainstream support. Thanks again for all your help and creating the project.

AdnanHodzic, 2016-12-08:
@markd89 most welcome!

saljut7, 2016-12-09:
@AdnanHodzic thanks alot I followed your workaround installing `linux-image-4.7.0-0.bpo.1-amd64`. Running now on `Linux T60 4.7.0-0.bpo.1-amd64 #1 SMP Debian 4.7.8-1~bpo8+1 (2016-10-19) x86_64 GNU/Linux`.
Unfortunately it didn't change anything for me. I couldn't activate `dlm.service` until I applied [libstdc++.so.6.0.22 workaround](https://github.com/AdnanHodzic/displaylink-debian/issues/37) in addition. 
When `dlm.service` was finally active I went on with

```
$ xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x77 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 3 associated providers: 0 name:radeon
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```
and

`$ xrandr --setprovideroutputsource 1 0`

Using `arandr` I could activate `DVI-1-1` (my DisplayLink adapter) and could set up my 2 external display. The command line setup in my case is:

```
#!/bin/sh
xrandr --output VGA-0 --mode 1280x1024 --pos 2680x0 --rotate normal --output LVDS --mode 1400x1050 --pos 1280x0 --rotate normal --output DVI-1-1 --mode 1280x1024 --pos 0x0 --rotate normal --output DVI-0 --off
```

But if I execute/apply these settings my system "crashes" back to login screen. Did I miss a certain step with the suggested workarounds?

markd89, 2016-12-10:
fyi: That's the same behavior I had with the older kernel when using the workaround...

AdnanHodzic, 2016-12-10:
@markd89 @saljut7 [in latest commit](https://github.com/AdnanHodzic/displaylink-debian/commit/1744ad3f621e69a2fdcefbd824f9e63c7a536b28) I made "libstdc++.so.6 hack" mentioned in #37 part of the installer.

However since packages in Jessie are now so old, you'll still have to install "libstdc++.so.6" from Stretch repo in order for everything to work. 

@saljut7 following that logic ^, I'd suggest you also upgrade "xserver-xorg" packages to Stretch. I know this helped in one of the issues in past (couldn't find it for reference)

Problem you're facing is that you're using old release with already outdated packages. I have personally never used pure "stable" on my desktop environments. I've always used "Apt-Pinning" to have certain packages from different releases without putting my stability in question. 

I wrote a blog post on this topic which you might find useful: "[Taking control over Debian and its package repositories](http://foolcontrol.org/?p=1708)"

Other option would just to simply upgrade to Stretch.

saljut7, 2016-12-10:
thx @AdnanHodzic 

I just ask because upgrading the kernel was not without problems (forgot that I had VirtualBox installed) do you think that there is a chance that upgrading `xserver-xorg` from `1:7.7+7` to `1:7.7+18` will help?

```
# apt-cache policy xserver-xorg
xserver-xorg:
  Installed: 1:7.7+7
  Candidate: 1:7.7+7
  Version table:
     1:7.7+18 0
        750 http://ftp.de.debian.org/debian/ testing/main amd64 Packages
         50 http://ftp.de.debian.org/debian/ unstable/main amd64 Packages
 *** 1:7.7+7 0
        995 http://ftp.de.debian.org/debian/ jessie/main amd64 Packages
        995 http://ftp.de.debian.org/debian/ stable/main amd64 Packages
        100 /var/lib/dpkg/status
```

AdnanHodzic, 2016-12-10:
xserver-xorg is only a transitional package and doesn't provide the driver itself, I was referring to "xserver-xorg-core" and "xserver-xorg-video-*" (whichever you're using).

saljut7, 2016-12-10:
@AdnanHodzic thanks alot for clarifying. UPgrading with `apt install -t jessie-backports xserver-xorg-video-intel=2:2.99.917+git20161105-1~bpo8+1` was no problem but for `apt install -t testing xserver-xorg-core=2:1.18.4-2` I expect a [repo mix-and-match hit parade](http://forums.debian.net/viewtopic.php?f=20&t=114130&hilit=mixed+repoSo):

```
 # apt install -t testing xserver-xorg-core=2:1.18.4-2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 xserver-xorg-core : Depends: xserver-common (>= 2:1.18.4-2) but 2:1.16.4-1 is to be installed
                     Depends: libdbus-1-3 (>= 1.9.14) but 1.8.20-0+deb8u1 is to be installed
                     Depends: libgcrypt20 (>= 1.7.0) but 1.6.3-2+deb8u2 is to be installed
                     Recommends: libpam-systemd but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

So I think it would be better to upgrade completely to testing. Thanks alot for your article http://foolcontrol.org/?p=1708. I'm already using apt-pinning but I didn't make the best experience with mixing repos. Would you recommend me a special "upgrading to Debian stesing" just for the case you know a good ([How to upgrade Debian stable (Wheezy) to testing (Jessie)?](https://unix.stackexchange.com/questions/90389/how-to-upgrade-debian-stable-wheezy-to-testing-jessie) or [Update Debian 8.3 Jessie "Stable" to "Testing" or "Unstable" for more up-to-date edition](https://www.linkedin.com/pulse/update-debian-83-jessie-stable-testing-unstable-more-edition-bagyono) looks to easy :) ).

AdnanHodzic, 2016-12-10:
It looks easy because it is easy :)

As long as you have correct sources, it's easy as running apt-get dist-upgrade. But if you're doing this for the first time, please follow: https://www.linkedin.com/pulse/update-debian-83-jessie-stable-testing-unstable-more-edition-bagyono

saljut7, 2016-12-11:
@AdnanHodzic ...but is the procedure described in https://www.linkedin.com/pulse/update-debian-83-jessie-stable-testing-unstable-more-edition-bagyono really the same with configured apt-pinning?

I just want to confirm because https://www.debian.org/releases/testing/amd64/release-notes/ch-upgrading.en.html says...

> 4.2.2. Disabling APT pinning
> 
> If you have configured APT to install certain packages from a distribution other than stable (e.g. from testing), you may have to change your APT pinning configuration (stored in /etc/apt/preferences and /etc/apt/preferences.d/) to allow the upgrade of packages to the versions in the new stable release. Further information on APT pinning can be found in apt_preferences(5). 

...which is vague because headling says "disabling" but explanation text just "you may have to _change_  Apt pinning configuration".

AdnanHodzic, 2016-12-13:
Sorry, but now you're asking my help with administering a Debian box, which is definitely out of this projects scope :)

If you're going to upgrade to Stretch, unless you want to use packages from Sid/Experimental, then you don't need pinning anymore and with dist-upgrade it can get you into trouble.

So just do this:

1. disable pinning
`sed -i 's/^\([^#]\)/#\1  /etc/apt/preferences`

2. change every occurrence of jessie with stretch
`sed -e "s/jessie/stretch/g" /etc/apt/sources.list`

k

AdnanHodzic, 2016-12-13:
Sorry, but now you're asking my help with administering a Debian box, which is definitely out of this projects scope. But since I'm in a good mood, I'll help you out :)

If you're going to upgrade to Stretch, unless you want to use packages from Sid/Experimental, then you don't need pinning anymore and with dist-upgrade it can get you into trouble.

So just do this:

1. disable pinning
`sed -i 's/^\([^#]\)/#\1  /etc/apt/preferences`

2. change every occurrence of jessie with stretch
`sed -e "s/jessie/stretch/g" /etc/apt/sources.list`

2.5. make sure you do the same with any other file under /etc/apt/sources.d/

3. update package list and upgrade
sudo apt-get update && sudo apt-get dist-upgrade

That's it, you're running on Stretch.

Happy hacking, and google is your friend :)

saljut7, 2016-12-15:
@AdnanHodzic at first I'm very sorry for bothering you with my last question. In fact it is not part of this issue tracker to help with dist upgrades (but to be honest: Google is _not_ my friend and I try to avoid any kind of (not only) Google surveillance services as much as possible - unfortunately it means sometimes I don't benefit from very good Google search results: I tried to find some upgrading to Debian testing with configured apt-pinning tutorial, of course). So thank you even more for providing helping steps (I promise to help others in same way). 
At first I tried to use my DisplayLink adapter after upgrading on an identical machine running with Debian8 (just wihtout apt-pinning as on my main machine). Uprading to texting was fine but results with debian-displaylink were worse than before: 

Reinstalling (tested on this Notebook before) there was an error report:

```
Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory
sed: can't read /lib/systemd/system/dlm.service: No such file or directory
Failed to enable unit: File dlm.service: No such file or directory
```

After uninstalling, rebooting, installing (this time with an error which seems not to be critical)...

```
Preparing for install

Archive:  DisplayLink_Ubuntu_1.2.65.zip
  inflating: 1.2.65/displaylink-driver-1.2.65.run  
  inflating: 1.2.65/LICENSE          
Creating directory displaylink-driver-1.2.65
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.2.65  100%  
ln: failed to create symbolic link '/lib/modules/4.8.0-1-amd64/build/Kconfig': File exists
```

...and rebooting `dlm.service` was running at startup finally. But then:

```
$ xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x44 cap: 0x0 crtcs: 2 outputs: 3 associated providers: 0 name:ATI Mobility Radeon X1400 @ pci:0000:01:00.0
Provider 1: id: 0x2c3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

$ xrandr --setprovideroutputsource 1 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x44
  Serial number of failed request:  16
  Current serial number in output stream:  17
```

`BadValue (integer parameter out of range for operation)` was not reported in context of displaylink-debian, was it?
[rotation](https://lists.freedesktop.org/archives/xorg/2015-January/057115.html) is not the problem in this case because `arandr`@Testing does not even list the attached display as under Debian Stable.

AdnanHodzic, 2017-02-18:
Marking this issue as obsolete and closing the ticket. 

Please update to latest version of displaylink-debian tool, and please re-open this or create a new issue if the problem persists.

saljut7, 2017-02-19:
Hi @AdnanHodzic 
I've tried again on a fresh installed Debian8.7 and still get "failed state" error:

```
User@T60-Notebook:~$ sudo systemctl start dlm.service
User@T60-Notebook:~$ sudo systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2017-02-19 11:28:54 CET; 4s ago
  Process: 1368 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 1365 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.3.52 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 1368 (code=exited, status=1/FAILURE)

Feb 19 11:28:54 T60-Notebook systemd[1]: Unit dlm.service entered failed state.
User@T60-Notebook:~$ sudo systemctl start dlm.service
User@T60-Notebook:~$ sudo systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2017-02-19 11:29:46 CET; 1s ago
  Process: 1503 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 1501 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.3.52 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 1503 (code=exited, status=1/FAILURE)

User@T60-Notebook:~$ sudo journalctl -xn
-- Logs begin at Sun 2017-02-19 11:27:36 CET, end at Sun 2017-02-19 11:31:37 CET. --
Feb 19 11:31:35 T60-Notebook mtp-probe[1765]: checking bus 5, device 6: "/sys/devices/pci0000:00/0000:00:1d.7/usb5/5-6/5-6.1"
Feb 19 11:31:35 T60-Notebook mtp-probe[1765]: bus: 5, device: 6 was not an MTP device
Feb 19 11:31:36 T60-Notebook sudo[1669]: pam_unix(sudo:session): session closed for user root
Feb 19 11:31:36 T60-Notebook systemd[1]: Starting DisplayLink Manager Service...
-- Subject: Unit dlm.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit dlm.service has begun starting up.
Feb 19 11:31:36 T60-Notebook systemd[1]: Started DisplayLink Manager Service.
-- Subject: Unit dlm.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit dlm.service has finished starting up.
-- 
-- The start-up result is done.
Feb 19 11:31:36 T60-Notebook DisplayLinkManager[1788]: /opt/displaylink/DisplayLinkManager: ./libstdc++.so.6: version `GLIBCXX_3.4.21' not foun
Feb 19 11:31:36 T60-Notebook systemd[1]: dlm.service: main process exited, code=exited, status=1/FAILURE
Feb 19 11:31:36 T60-Notebook systemd[1]: Unit dlm.service entered failed state.
Feb 19 11:31:37 T60-Notebook sudo[1794]: User : TTY=pts/0 ; PWD=/home/User ; USER=root ; COMMAND=/bin/journalctl -xn
Feb 19 11:31:37 T60-Notebook sudo[1794]: pam_unix(sudo:session): session opened for user root by User(uid=0)
```

Well if it is still necessary to upgrade the kernel I would suggest to update the [readme](https://github.com/AdnanHodzic/displaylink-debian/blob/master/README.md) related to...

> Regardless of which kernel version you're using.

...and better say which kernel version is needed.

Anyway thanks alot for this great script!

AdnanHodzic, 2017-02-20:
@saljut7 resolution to your problem has been explained as part of [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues)

saljut7, 2017-02-26:
@AdnanHodzic thanks a lot. I followed the instruction but ended up like [last time ](https://github.com/AdnanHodzic/displaylink-debian/issues/42#issuecomment-266146964) with a crashing desktop environment after enabling the `arand` configuration. Upgrading the kernel /xserver didn't help the last time. Maybe this is a similar problem to [Crashing when starting X](https://github.com/AdnanHodzic/displaylink-debian/issues/27#issue-183531958) (so in the end a problem with my external monitor)?

wordyallen, 2017-05-11:
when i run 

`systemctl status dlm`

I get the error:

`Process: 3689 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.3.52 && modprobe evdi) (code=exited, status=1/FAILURE)`

wordyallen, 2017-05-27:
It was a combo of config for intel cards that @gitowiec suggested as well as uefi settings in bios. but I got it working! Could you put  this info in the post installation guide?

redjamesg, 2017-05-29:
It is already there

saljut7, 2017-06-24:
After release of Debian 9 I dist-upgraded and tried again displaylink-debian:

1. I still get `version `GLIBCXX_3.4.21' not found` error which I could fix via https://github.com/AdnanHodzic/displaylink-debian/blob/905c1563d26c47500474e57f28c4f5ce779635e7/post-install-guide.md#L103 .
2. Applying 3 display setup via `arandr` my desktop environment still crashes into login screen (I just can see a short terminal output `[ OK ] Started Update UTMP about System Runlevel Changes.`).

To confirm: I don't have an intel grafic card:

```
saljut7@Lenovo-T60:~$ inxi -Fx
System:    Host: Lenovo-T60 Kernel: 3.16.0-4-amd64 x86_64 (64 bit gcc: 4.8.4) Desktop: i3 4.8
           Distro: Debian GNU/Linux 9 (stretch)
Machine:   Device: laptop System: LENOVO product: 2007FVG v: ThinkPad T60
           Mobo: LENOVO model: 2007FVG BIOS: LENOVO v: 79ETD8WW (2.18 ) date: 09/05/2007
CPU:       Dual core Intel Core2 T5600 (-MCP-) cache: 2048 KB flags: (lm nx sse sse2 sse3 ssse3 vmx) bmips: 7314 
           clock speeds: max: 1833 MHz 1: 1000 MHz 2: 1833 MHz
Graphics:  Card: Advanced Micro Devices [AMD/ATI] RV515/M54 [Mobility Radeon X1400] bus-ID: 01:00.0
           Display Server: X.Org 1.16.4 drivers: modesetting,ati,radeon (unloaded: fbdev,vesa)
           Resolution: 1400x1050@60.02hz, 1280x1024@60.02hz
           GLX Renderer: Gallium 0.4 on llvmpipe (LLVM 3.5, 128 bits)
           GLX Version: 3.0 Mesa 10.3.2 Direct Rendering: Yes
Audio:     Card Intel NM10/ICH7 Family High Definition Audio Controller driver: snd_hda_intel bus-ID: 00:1b.0
[...]
```

closed 2017-06-26 by AdnanHodzic