AdnanHodzic/displaylink-debian #43: Support for Debian GNU/Linux 8.6 (jessie)

deckrider, 2016-11-28:
I've been unable to get this to work on Debian GNU/Linux 8.6 (jessie).

My latest attempt includes this latest Git commit:

commit f5d451f29d3c7cecbc4b130b29dfaecf436268c5
Author: Adnan Hodzic <adnan@hodzic.org>
Date:   Sun Nov 27 18:50:50 2016 +0100

    added "enable dlm.service" by default

However, after uninstalling/installing, the following illustrates the latest issue:

$ ldd /opt/displaylink/DisplayLinkManager 
/opt/displaylink/DisplayLinkManager: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /opt/displaylink/DisplayLinkManager)

I would really like to stay on (without deviating from) Debian Stable if possible.

Thanks.

AdnanHodzic, 2016-11-28:
What happens if you run: "sudo apt-get install libstdc++6" and the do same thing again?

deckrider, 2016-11-29:
That was already installed and up to date with the latest jessie security:

```
$ dpkg -L "libstdc++6:amd64"
/.
/usr
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.20
/usr/share
/usr/share/doc
/usr/share/gcc-4.9
/usr/share/gcc-4.9/python
/usr/share/gcc-4.9/python/libstdcxx
/usr/share/gcc-4.9/python/libstdcxx/__init__.py
/usr/share/gcc-4.9/python/libstdcxx/v6
/usr/share/gcc-4.9/python/libstdcxx/v6/__init__.py
/usr/share/gcc-4.9/python/libstdcxx/v6/printers.py
/usr/share/gdb
/usr/share/gdb/auto-load
/usr/share/gdb/auto-load/usr
/usr/share/gdb/auto-load/usr/lib
/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu
/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.20-gdb.py
/usr/lib/x86_64-linux-gnu/libstdc++.so.6
/usr/share/doc/libstdc++6

```

AdnanHodzic, 2016-11-29:
Do you happen to have secure boot enabled? If yes, please disable it.

deckrider, 2016-11-29:
I don't think I have secure boot enabled (not unless the Debian installer does this by default).  Is there some command I should run to prove this?

Nevertheless, I think the real issue to the above "ldd" message is that the executable (/opt/displaylink/DisplayLinkManage) is expecting a symbol (or string) it cannot find in the version of the library that is included with jessie.  Perhaps a different executable is needed (I see that displaylink provides two in its upstream package)?

saljut7, 2016-12-01:
I've experienced something similar:

I tried it with the updated version again on two Devices running Debian8.6. Just to confirm:

1. Installation
2. Reboot
3. Plugged in DisplayLink adapter
4. Checked, started and again checked service:

```
# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2016-11-29 17:08:32 CET; 799ms ago
  Process: 1518 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 1515 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1518 (code=exited, status=1/FAILURE)

Nov 29 17:08:32 T60-Notebook systemd[1]: Unit dlm.service entered fa....
Hint: Some lines were ellipsized, use -l to show in full.
root@T60-Notebook:~# systemctl start dlm.service
root@T60-Notebook:~# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2016-11-29 17:08:42 CET; 2s ago
  Process: 1626 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=1/FAILURE)
  Process: 1624 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1626 (code=exited, status=1/FAILURE)

Nov 29 17:08:42 T60-Notebook systemd[1]: Unit dlm.service entered fa....
Hint: Some lines were ellipsized, use -l to show in full.
root@T60-Notebook:~# journalctl -xn
-- Logs begin at Tue 2016-11-29 17:07:19 CET, end at Tue 2016-11-29
Nov 29 17:08:47 T60-Notebook systemd[1]: dlm.service: main process exite
Nov 29 17:08:47 T60-Notebook systemd[1]: Unit dlm.service entered failed
Nov 29 17:08:47 T60-Notebook DisplayLinkManager[1654]: /opt/displaylink/
Nov 29 17:08:53 T60-Notebook systemd[1]: dlm.service holdoff time over, 
Nov 29 17:08:53 T60-Notebook systemd[1]: Stopping DisplayLink Manager Se
-- Subject: Unit dlm.service has begun shutting down
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-d
-- 
-- Unit dlm.service has begun shutting down.
Nov 29 17:08:53 T60-Notebook systemd[1]: Starting DisplayLink Manager Se
-- Subject: Unit dlm.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-d
-- 
-- Unit dlm.service has begun starting up.
Nov 29 17:08:53 T60-Notebook systemd[1]: Started DisplayLink Manager Ser
-- Subject: Unit dlm.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-d
-- 
-- Unit dlm.service has finished starting up.
-- 
-- The start-up result is done.
Nov 29 17:08:53 T60-Notebook DisplayLinkManager[1666]: /opt/displaylink/
Nov 29 17:08:53 T60-Notebook systemd[1]: dlm.service: main process exite
Nov 29 17:08:53 T60-Notebook systemd[1]: Unit dlm.service entered failed
lines 1-28/28 (END)...skipping...
-- Logs begin at Tue 2016-11-29 17:07:19 CET, end at Tue 2016-11-29 17:08:53 CET. --
Nov 29 17:08:47 T60-Notebook systemd[1]: dlm.service: main process exited, code=exited, status=1/FAILURE
Nov 29 17:08:47 T60-Notebook systemd[1]: Unit dlm.service entered failed state.
Nov 29 17:08:47 T60-Notebook DisplayLinkManager[1654]: /opt/displaylink/DisplayLinkManager: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `
Nov 29 17:08:53 T60-Notebook systemd[1]: dlm.service holdoff time over, scheduling restart.
Nov 29 17:08:53 T60-Notebook systemd[1]: Stopping DisplayLink Manager Service...
-- Subject: Unit dlm.service has begun shutting down
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit dlm.service has begun shutting down.
Nov 29 17:08:53 T60-Notebook systemd[1]: Starting DisplayLink Manager Service...
-- Subject: Unit dlm.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit dlm.service has begun starting up.
Nov 29 17:08:53 T60-Notebook systemd[1]: Started DisplayLink Manager Service.
-- Subject: Unit dlm.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit dlm.service has finished starting up.
-- 
-- The start-up result is done.
Nov 29 17:08:53 T60-Notebook DisplayLinkManager[1666]: /opt/displaylink/DisplayLinkManager: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `
Nov 29 17:08:53 T60-Notebook systemd[1]: dlm.service: main process exited, code=exited, status=1/FAILURE
Nov 29 17:08:53 T60-Notebook systemd[1]: Unit dlm.service entered failed state.
```

So still "Unit dlm.service entered failed state.". On this notebook I didn't try the workaround of [akersting](https://github.com/AdnanHodzic/displaylink-debian/issues/37) before and I didn't recognize a secure boot option neither.

Any ideas?

deckrider, 2016-12-02:
I realized you mean UEFI Secure Boot.  This was already disabled on my laptop and I have enabled what I suppose is now called legacy BIOS.

AdnanHodzic, 2016-12-04:
@deckrider could you [please do workaround I suggested in #42](https://github.com/AdnanHodzic/displaylink-debian/issues/42#issuecomment-264711774)? This fixes the issue with 3.16 kernel on Jessie.

redjamesg, 2016-12-06:
I decided to upgrade to stretch instead of just upgrading the kernel.

AdnanHodzic, 2016-12-06:
I think you made a right decision :)

Closing the issue, please re-open it if you have same problem.

closed 2016-12-06 by AdnanHodzic