AdnanHodzic/displaylink-debian #45: Installation Issue Kali 2 (Missing Linux Headers)

flipside2000, 2016-12-13:
So I'm a bit of a noob in these areas however very persistent. I am coming a little stuck here though on installation. The error I am getting:

Installing driver version: 1.2.65

_DisplayLink Linux Software 1.2.65 install script called: install
Distribution discovered: Kali GNU/Linux Rolling
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 4.8.0-kali2-amd64.
This is a fatal error, cannot install DisplayLink Linux Software._

Uname -a output: Linux kali 4.8.0-kali2-amd64 #1 SMP Debian 4.8.11-1kali1 (2016-12-08) x86_64 GNU/Linux
Is this error due to my noob ways, or is there something else going on here...? Any help would be massively appreciated!

AdnanHodzic, 2016-12-14:
During dependency check, if missing, installer will prompt that kernel headers should be installed. 

Can you do `apt-get install linux-headers-$(uname -r)` to manually to get them installed to see what happens?

And can you paste whole output of everything that happens, I think you either skipped it, or one such package doesn't exists in repos (unlikely).

flipside2000, 2016-12-14:
Yup, not a problem...output as below:
kali:~# apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-4.8.0-kali2-amd64 is already the newest version (4.8.11-1kali1).
The following packages were automatically installed and are no longer required:
  glx-alternative-mesa libgldispatch0-nvidia libnvidia-eglcore libnvidia-glcore
  libopengl0-glvnd-nvidia libvulkan1 update-glx
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

flipside2000, 2016-12-19:
Any other ideas here? I don't think i's any step I've missed?

placebo75, 2016-12-21:
Same issue here :(
Any idea?

banzaighoul, 2017-01-05:
Same here

Choose Install (I)
Checking dependencies

unzip is installed
linux-headers-4.8.0-kali2-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

File downloads and extracts ok then....

Installing driver version: 1.2.65

DisplayLink Linux Software 1.2.65 install script called: install
Distribution discovered: Kali GNU/Linux Rolling
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 4.8.0-kali2-amd64.
This is a fatal error, cannot install DisplayLink Linux Software.

lee2098, 2017-01-06:
I'm in the same boat, everything required was installed previously including headers but still hitting the dependency message for them.

AdnanHodzic, 2017-01-15:
There was a bug in code, which disabled steps that needed to be done for Kali Linux. Please update to latest version, which fixes the problem mentioned here. 

If this doesn't resolve your problem, please re-open the issue.

closed 2017-01-15 by AdnanHodzic

dannyvanderzande, 2018-01-07:
Had this same problem, turned out to be not rebooting after an apt upgrade. I had newer kernel headers installed than I was running and that messed stuff up.. Thought I'd share this for others having the same issue :-)