AdnanHodzic/displaylink-debian #46: Black Screens Only Shows Mouse Pointer

dalangston, 2016-12-28:
I'm running Debian Stretch with kernel 4.8.0-2-amd64. I'm using an ASUS HZ-1 Docking Station.  After install dlm service starts up, xrandr sees the displays, but they only show back screens.  I can move the mouse pointer onto the screen and see it's position but no background, or windows are displayed.

systemctl status dlm.service  

    ● dlm.service - DisplayLink Manager Service
       Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
       Active: active (running) since Wed 2016-12-28 11:48:54 EST; 17min ago
      Process: 3251 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
     Main PID: 3266 (DisplayLinkMana)
        Tasks: 29 (limit: 4915)
       CGroup: /system.slice/dlm.service
               └─3266 /opt/displaylink/DisplayLinkManager

Dmesg:  

    Dec 28 11:46:27 myHost kernel: [  719.092820] evdi: [D] evdi_painter_crtc_state_notify:374 (dev=-1) Notifying crtc state: 3
    Dec 28 11:46:27 myHost kernel: [  719.092823] evdi: [W] evdi_painter_send_crtc_state:266 Painter is not connected!
    Dec 28 11:46:27 myHost kernel: [  719.096877] usb 2-1.3.4: USB disconnect, device number 8
    Dec 28 11:46:27 myHost kernel: [  719.144444] evdi: [D] evdi_painter_crtc_state_notify:374 (dev=-1) Notifying crtc state: 3
    Dec 28 11:46:27 myHost kernel: [  719.144448] evdi: [W] evdi_painter_send_crtc_state:266 Painter is not connected!
    Dec 28 11:46:27 myHost kernel: [  719.209232] traps: xfsettingsd[2468] trap divide error ip:55f080a05483 sp:7ffe9de918f0 error:0 in xfsettingsd[55f0809f4000+18000]
    Dec 28 11:46:27 myHost kernel: [  719.262421] evdi: [D] evdi_detect:72 Painter is disconnected
    Dec 28 11:46:27 myHost kernel: [  719.262436] evdi: [D] evdi_detect:72 Painter is disconnected
    Dec 28 11:48:52 myHost kernel: [  864.616902] usb 2-1.3: new high-speed USB device number 9 using ehci-pci
    Dec 28 11:48:52 myHost kernel: [  864.734609] usb 2-1.3: New USB device found, idVendor=05e3, idProduct=0610
    Dec 28 11:48:52 myHost kernel: [  864.734615] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
    Dec 28 11:48:52 myHost kernel: [  864.734618] usb 2-1.3: Product: USB2.0 Hub
    Dec 28 11:48:52 myHost kernel: [  864.734621] usb 2-1.3: Manufacturer: GenesysLogic
    Dec 28 11:48:52 myHost kernel: [  864.736046] hub 2-1.3:1.0: USB hub found
    Dec 28 11:48:52 myHost kernel: [  864.737400] hub 2-1.3:1.0: 4 ports detected
    Dec 28 11:48:53 myHost kernel: [  865.044911] usb 2-1.3.2: new high-speed USB device number 10 using ehci-pci
    Dec 28 11:48:53 myHost kernel: [  865.154815] usb 2-1.3.2: New USB device found, idVendor=17e9, idProduct=4301
    Dec 28 11:48:53 myHost kernel: [  865.154821] usb 2-1.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    Dec 28 11:48:53 myHost kernel: [  865.154824] usb 2-1.3.2: Product: ASUSTEK USB Docking Station
    Dec 28 11:48:53 myHost kernel: [  865.154827] usb 2-1.3.2: Manufacturer: DisplayLink
    Dec 28 11:48:53 myHost kernel: [  865.154830] usb 2-1.3.2: SerialNumber: 000510430830543
    Dec 28 11:48:53 myHost kernel: [  865.161192] usb 2-1.3.2: Warning! Unlikely big volume range (=511), cval->res is probably wrong.
    Dec 28 11:48:53 myHost kernel: [  865.161199] usb 2-1.3.2: [16] FU [ASUSTEK  Audio Playback Volume] ch = 6, val = -8176/0/16
    Dec 28 11:48:53 myHost kernel: [  865.162183] usb 2-1.3.2: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
    Dec 28 11:48:53 myHost kernel: [  865.162188] usb 2-1.3.2: [13] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
    Dec 28 11:48:53 myHost kernel: [  865.185410] cdc_ncm 2-1.3.2:1.5: MAC-Address: 00:24:9b:0a:48:d1
    Dec 28 11:48:53 myHost kernel: [  865.185417] cdc_ncm 2-1.3.2:1.5: setting rx_max = 16384
    Dec 28 11:48:53 myHost kernel: [  865.185599] cdc_ncm 2-1.3.2:1.5: setting tx_max = 16384
    Dec 28 11:48:53 myHost kernel: [  865.186183] cdc_ncm 2-1.3.2:1.5 usb0: register 'cdc_ncm' at usb-0000:00:1d.0-1.3.2, CDC NCM, 00:24:9b:0a:48:d1
    Dec 28 11:48:53 myHost NetworkManager[880]: <info>  [1482943733.2868] manager: (usb0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/4)
    Dec 28 11:48:53 myHost kernel: [  865.268828] usb 2-1.3.3: new low-speed USB device number 11 using ehci-pci
    Dec 28 11:48:53 myHost mtp-probe: checking bus 2, device 10: "/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.3/2-1.3.2"
    Dec 28 11:48:53 myHost mtp-probe: bus: 2, device: 10 was not an MTP device
    Dec 28 11:48:53 myHost kernel: [  865.382008] usb 2-1.3.3: New USB device found, idVendor=06eb, idProduct=0005
    Dec 28 11:48:53 myHost kernel: [  865.382015] usb 2-1.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
    Dec 28 11:48:53 myHost kernel: [  865.382018] usb 2-1.3.3: Product: USB
    Dec 28 11:48:53 myHost kernel: [  865.382021] usb 2-1.3.3: Manufacturer: Rextron
    Dec 28 11:48:53 myHost kernel: [  865.388555] input: Rextron USB as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.3/2-1.3.3/2-1.3.3:1.0/0003:06EB:0005.0004/input/input24
    Dec 28 11:48:53 myHost kernel: [  865.445789] hid-generic 0003:06EB:0005.0004: input,hidraw1: USB HID v1.10 Keyboard [Rextron USB] on usb-0000:00:1d.0-1.3.3/input0
    Dec 28 11:48:53 myHost kernel: [  865.452311] input: Rextron USB as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.3/2-1.3.3/2-1.3.3:1.1/0003:06EB:0005.0005/input/input25
    Dec 28 11:48:53 myHost kernel: [  865.509718] hid-generic 0003:06EB:0005.0005: input,hidraw2: USB HID v1.10 Mouse [Rextron USB] on usb-0000:00:1d.0-1.3.3/input1
    Dec 28 11:48:53 myHost mtp-probe: checking bus 2, device 11: "/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.3/2-1.3.3"
    Dec 28 11:48:53 myHost mtp-probe: bus: 2, device: 11 was not an MTP device
    Dec 28 11:48:53 myHost kernel: [  865.736856] usb 2-1.3.4: new high-speed USB device number 12 using ehci-pci
    Dec 28 11:48:53 myHost kernel: [  865.845226] usb 2-1.3.4: New USB device found, idVendor=0424, idProduct=2514
    Dec 28 11:48:53 myHost kernel: [  865.845233] usb 2-1.3.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
    Dec 28 11:48:53 myHost kernel: [  865.845987] hub 2-1.3.4:1.0: USB hub found
    Dec 28 11:48:53 myHost kernel: [  865.846239] hub 2-1.3.4:1.0: 4 ports detected
    Dec 28 11:48:54 myHost kernel: [  866.400166] cdc_ncm 2-1.3.2:1.5 enx00249b0a48d1: renamed from usb0
    Dec 28 11:48:54 myHost kernel: [  866.432068] IPv6: ADDRCONF(NETDEV_UP): enx00249b0a48d1: link is not ready
    Dec 28 11:48:54 myHost kernel: [  866.432134] IPv6: ADDRCONF(NETDEV_UP): enx00249b0a48d1: link is not ready
    Dec 28 11:48:54 myHost NetworkManager[880]: <info>  [1482943734.5230] device (usb0): interface index 5 renamed iface from 'usb0' to 'enx00249b0a48d1'
    Dec 28 11:48:54 myHost NetworkManager[880]: <info>  [1482943734.5288] devices added (path: /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.3/2-1.3.2/2-1.3.2:1.5/net/enx00249b0a48d1, iface: enx00249b0a48d1)
    Dec 28 11:48:54 myHost NetworkManager[880]: <info>  [1482943734.5288] device added (path: /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.3/2-1.3.2/2-1.3.2:1.5/net/enx00249b0a48d1, iface: enx00249b0a48d1): no ifupdown configuration found.
    Dec 28 11:48:54 myHost NetworkManager[880]: <info>  [1482943734.5293] device (enx00249b0a48d1): state change: unmanaged -> unavailable (reason 'managed') [10 20 2]
[... 220 lines cut ...]
    Dec 28 11:54:41 myHost kernel: [ 1213.432920] evdi: [D] evdi_painter_crtc_state_notify:374 (dev=1) Notifying crtc state: 0
    Dec 28 11:54:41 myHost kernel: [ 1213.432923] evdi: [D] evdi_painter_dpms_notify:361 (dev=1) Notifying dpms mode: 0
    Dec 28 11:54:41 myHost kernel: [ 1213.486381] traps: xfsettingsd[3071] trap divide error ip:562b94ab4483 sp:7ffd3d62bf80 error:0 in xfsettingsd[562b94aa3000+18000]
    Dec 28 11:54:41 myHost kernel: [ 1213.526823] evdi: [D] evdi_detect:69 (dev=2) Painter is connected
    Dec 28 11:54:41 myHost kernel: [ 1213.526827] evdi: [D] evdi_painter_get_edid_copy:192 (dev=2) 00 ff ff
    Dec 28 11:54:41 myHost kernel: [ 1213.526904] evdi: [D] evdi_detect:69 (dev=1) Painter is connected
    Dec 28 11:54:41 myHost kernel: [ 1213.526905] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
    Dec 28 11:54:41 myHost kernel: [ 1213.580149] traps: xfsettingsd[3594] trap divide error ip:55d28728d483 sp:7fff0c1f9ff0 error:0 in xfsettingsd[55d28727c000+18000]
[... cut ...]
    Dec 28 11:54:46 myHost kernel: [ 1218.249857] evdi: [D] evdi_crtc_cursor_set:238 evdi_crtc_cursor_set unlock
    Dec 28 11:54:46 myHost kernel: [ 1218.833765] evdi: [D] evdi_crtc_cursor_set:238 evdi_crtc_cursor_set unlock
    Dec 28 11:54:46 myHost kernel: [ 1218.841822] evdi: [D] evdi_crtc_cursor_set:238 evdi_crtc_cursor_set unlock
    Dec 28 11:54:47 myHost kernel: [ 1219.522797] evdi: [D] evdi_crtc_cursor_set:238 evdi_crtc_cursor_set unlock
    Dec 28 11:54:47 myHost kernel: [ 1219.529711] evdi: [D] evdi_crtc_cursor_set:238 evdi_crtc_cursor_set unlock

glindner, 2017-01-03:
Exactly! I have the same problem. I commented on #47 with my experience which sounds just like this. I believe it will require a fix from either DisplayLink's evdi driver or Gnome...

dalangston, 2017-01-04:
The problem is not limited to Gnome.  I'm currently using XFCE and lightDM.

AdnanHodzic, 2017-01-15:
This issue seems to be related to upstream, and I'd suggest you [report this problem there](https://github.com/DisplayLink/evdi)

AdnanHodzic, 2017-02-18:
Marking this issue as obsolete and closing the ticket. 

Please update to latest version of displaylink-debian tool, and please re-open this or create a new issue if the problem persists.

closed 2017-02-18 by AdnanHodzic