AdnanHodzic/displaylink-debian #47: USB screen turns on and mirror correctly but display is frozen

aleph-zer0, 2016-12-29:
Hi, 

Thanks a lot for the script, helped a lot! I am almost done making it work but probably have some config issues so any help would be much appreciated.

I have successfully installed dlm.service and can run it correctly on debian Jessie stretch with kernel 3.16.0-4. The USB monitor I am trying to set-up is a AOC E1759FWU and computer is an Intel NUC i7 with Iris 6100 graphic.

xrandr sees both monitors within providers and I can access config for both through xfce  display GUI.
Upon running:
`xrandr --setprovideroutputsource 1 0`
the usb monitor succesfully turn on and mirror main screen but freeze and does not refresh afterward (like displaying a print screen of when it was connected).

Tried so far: 
- connecting on USB 2 or 3 with or without second power USB
- update/upgrade and reboot
- rerun display Link script / uninstall and reinstall successfully
- updated intel graphic module 
- restart dlm.service (update print screen but still no refresh)
 
sample output below

`x@cube  ~ $ uname -a`
`Linux cube 3.16.0-4-amd64 #1 SMP Debian 3.16.36-1+deb8u2 (2016-10-19) x86_64 GNU/Linux`

 `x@cube ~ $ xrandr --listproviders`
`Providers: number : 2`
`Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting`
`Provider 1: id: 0x125 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting`

 `x@cube ~ $ sudo systemctl status  dlm.service `
 `● dlm.service - DisplayLink Manager Service ` 
 `   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled) `
  `  Active: active (running) since Thu 2016-12-29 23:16:46 GMT; 5min ago `
 `  Process: 3710 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS) `
  `Main PID: 3714 (DisplayLinkMana) `
  `  CGroup: /system.slice/dlm.service `
  `          └─3714 /opt/displaylink/DisplayLinkManager `
 
 `Dec 29 23:16:46 cube systemd[1]: Starting DisplayLink Manager Service... `
 `Dec 29 23:16:46 cube systemd[1]: Started DisplayLink Manager Service. `

glindner, 2017-01-03:
Similar boat. Mouse moves on the external screen, but like you said, the screen acts like a screenshot and doesn't update.
arandr allows me to define the screens side-by-side (not mirrored) and the external monitor is black, but mouse movement displays on it. I drag a window from the laptop screen and still nothing but a mouse icon displays. Very strange.
I think there is a break between evdi and gdm with the 4.8.0 kernel upgrade. I've been hoping Kali picks up a fix, but no luck for 2-3 weeks now. One screen is tough...
Tried going back to 4.7.0 kernel with little luck. Other problems.
Oh well...
I don't think it has anything to do with AdnanHodzic's script. Probably going to be a change to evdi (painter?) or gdm.

aleph-zer0, 2017-01-04:
Was able to replicate glindner's issue by using side by side display. I show the mouse pointer and desktop + shapes of some windows  but nothing more and does not refresh. Will try to move the issue upstream to evdi project to see if someone has some insight on this.

closed 2017-01-04 by aleph-zer0