AdnanHodzic/displaylink-debian #48: dml.service won't start Ubuntu 16.04

yesennes, 2016-12-31:
I am on a fresh install of Ubuntu 16.04 booted with bios. dlm.service refuses to start.

`systemctl status dlm.service` gives me:

> ● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2016-12-31 13:57:00 EST; 1s ago
  Process: 3567 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)
Dec 31 13:57:00 yesennes-Lenovo-G50-70 systemd[1]: Failed to start DisplayLink Manager Service.
Dec 31 13:57:00 yesennes-Lenovo-G50-70 systemd[1]: dlm.service: Unit entered failed state.
Dec 31 13:57:00 yesennes-Lenovo-G50-70 systemd[1]: dlm.service: Failed with result 'exit-code'.

If there is any more information that would be useful, let me know.

glindner, 2017-01-03:
my guess is evdi drive didn't install properly. Have you tried uninstall, install, and reboot?

you may be able to manually remove and reinstall evdi using the following commands:
sudo dkms remove evdi/1.2.65 --all
sudo dkms add evdi/1.2.65
sudo dkms build evdi/1.2.65
sudo dkms install evdi/1.2.65
reboot

report any error messages that pop, but this should work.

AdnanHodzic, 2017-01-04:
@yesennes just yesterday I witnessed install on fresh install of Ubuntu 16.04 with 4.8 kernel from 16.10. Everything worked out just fine and dlm.service starts fine.

Have you tried uninstall/install/reboot as @glindner suggested?

AdnanHodzic, 2017-02-18:
You need to have UEFI disabled, please [see this issue for more info](https://github.com/AdnanHodzic/displaylink-debian/issues/15#issuecomment-280824050).

closed 2017-02-18 by AdnanHodzic

tc7, 2018-10-23:
This workaround worked for me on Ubuntu 18.04 and DisplayLink driver v4.4

[adnan hodzic fix](https://github.com/AdnanHodzic/displaylink-debian/issues/36)

To fix, add an install section to **/lib/systemd/system/dlm.service**

**[Install]
WantedBy=multi-user.target**