AdnanHodzic/displaylink-debian #51: ln: failed to create symbolic link ‘/opt/displaylink/libstdc++.so.6’: No such file or directory

unit309, 2017-01-11:
This is caused by
```
ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 /opt/displaylink/libstdc++.so.6
```
when /opt/displaylink is not exist.

You should probably add
```
mkdir -p /opt/displaylink
```
before.

AdnanHodzic, 2017-01-11:
Could you please pull the latest changes and try with (re)install? 

In my last commits, I thought it was a good idea to move that symlink creation from "post install" step to configuration part that was done exclusively for Debian and Kali.

But it turns out that at this stage "/opt/displaylink" dir wasn't even created. I just reverted these changes, so you shouldn't have this problem again.

AdnanHodzic, 2017-01-15:
Pushed additional changes related to libstdc++ symlink, this problem no longer exists.

closed 2017-01-15 by AdnanHodzic