AdnanHodzic/displaylink-debian #52: Glibc GLIBCXX_3.4.21 missing on Jessie

Vitexus, 2017-01-26:
# /opt/displaylink/DisplayLinkManager
/opt/displaylink/DisplayLinkManager: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /opt/displaylink/DisplayLinkManager)

http://displaylink.org/forum/showthread.php?t=64844 - download older driver 

http://www.displaylink.com/downloads/ubuntu
1.2.58 19 Sep 2016

and use displaylink-driver-1.2.58/x64-ubuntu-1404 from it

This solution work for me

markd89, 2017-02-15:
Are you using a newer kernel or the 3.16 that comes with Jessie?

Thanks,
Mark

Vitexus, 2017-02-17:
Yes: 3.16.0-4-amd64

AdnanHodzic, 2017-02-18:
I would suggest you update to newer kernel version, as well as xserver-xorg related packages. You can follow my [Apt Pinning](http://foolcontrol.org/?p=1708) on how to install packages from other releases while your base system remains on Stable.

I just pushed an update, in which displaylink-debian will install latest driver version (1.3.52). So you can also try with that and see if that fixes your problem.

AdnanHodzic, 2017-02-19:
@Vitexus I've just added [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues) section to Post-install guide.

In which there's a more elaborate explanation on how to fix your problem. If that doesn't address your problem, please feel free to reopen this issue or submit a new one.

closed 2017-02-19 by AdnanHodzic