AdnanHodzic/displaylink-debian #58: It doesn't work. Can't get resolution. 16.04.

finetype, 2017-02-18:
[image]
[image]
[image]
[image]

finetype, 2017-02-18:
Note, a new release from displaylink came out literally yesterday, it seems. It's available [here](http://www.displaylink.com/downloads/ubuntu).

after going through the process of trying to install the proper displaylink driver above originally, when the dlm.service just wouldn't start, and then going back through amazon reviews, hearing this package isn't just for 14.04 and does work for 14.04 (I originally had dismissed using this repo because it says 14.04 at the top of the README, leading me to assume it was far out of date), but no errors.... I uninstalled it, tried this one.

After going through post-install guide, finding a link to the issue for UEFI, struggling to figure out what I was supposed to do about UEFI (just boot and turn it off in the bios--not bad at all, just difficult to find someone talking plainly about it), reinstalling, getting errors above, uninstalling...

...I **FINALLY** got the screen to work after now reinstalling the original, _new_ driver from displaylink that I linked above.

Unfortunately, as arandr shows me, using it in portrait is not possible, which is sorely disappointing, as I bought this to code with. :/

Not sure what I'll do. Anyways, it's working. Hopefully this helps someone.

AdnanHodzic, 2017-02-18:
@finetype yes, I've [just explained in other issue](https://github.com/AdnanHodzic/displaylink-debian/issues/15#issuecomment-280824050) why you must have UEFI disabled in order for DisplayLink to work.

I also just pushed an update, in which displaylink-debian will install latest driver version (1.3.52).

Portrait mode problems seems to be upstream related problem, so you can [try reporting it there](https://github.com/DisplayLink/evdi).

closed 2017-02-18 by AdnanHodzic