AdnanHodzic/displaylink-debian #59: can not install

justmag, 2017-02-18:
hi ,
Thank you for support.
I can not install he driver displaylink 1.2.65 07 Oct 2016
I am a novice in Linux and  run upgraded linux Debian to 9,  kernel 3.16.0-4-amd64
 lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.0 (n/a)
Release:	9.0
Codename:	n/a

Initially, your script did not find linux-source installed.
It seems I have chosen INSTALL and it downloaded and installed something (I did not notice).
After rebooting I found no driver installed.

Now displaylink-debian.sh outputs: 

Checking dependencies

unzip is installed
linux-headers-3.16.0-4-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed

---------------------------------------------------------------

Unsuported platform: Debian n/a
.....

Can it be because of "Codename:	n/a"?

What can I do next?

Thanks, 
Best wishes,
Max

AdnanHodzic, 2017-02-18:
If you're running on 3.16 kernel then you're running Debian 8 (Jessie) and not 9 (Stretch).

Since libstdc package is already too old in Debian 8 (Jessie), please download one [from Stretch](https://packages.debian.org/stretch/libstdc++6) and install it.

After you've done that, please pull the latest changes from displaylink-debian, because just today I push the changes which install driver version 1.3.52. Run tool and select reinstall option.

This same problem was reported in [issue 37](https://github.com/AdnanHodzic/displaylink-debian/issues/37) awhile back.

AdnanHodzic, 2017-02-19:
@justmag  I've just added [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues) section to Post-install guide.

In which there's a more elaborate explanation on how to fix your problem. If that doesn't address your problem, please feel free to reopen this issue or submit a new one.

justmag, 2017-02-20:
hi Adnan,
I installed libstdc and tried it with driver 1.3.52 and new displaylink-debian.
It feels like the installer does not run at all.
There is only message 
"Unsuported platform: Debian n/a
This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/
"
so, after reboot:
dlm.service: Unit dlm.service not found.
and I can not find any traces of displaylink

Best, Max

AdnanHodzic, 2017-02-20:
That seems like a different issue then, and it really fails to install.

Can you give me output of following commands:

```
dpkg -s lsb-release
```
and then:
```
lsb_release -a
```

justmag, 2017-02-20:
dpkg -s lsb-release

Package: lsb-release
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 97
Maintainer: Debian LSB Team <debian-lsb@lists.debian.org>
Architecture: all
Multi-Arch: foreign
Source: lsb
Version: 4.1+Debian13+nmu1
Depends: python (>= 2.7), python (<< 2.8)
Recommends: apt
Suggests: lsb
------------------------------------------------------------------------
lsb_release -a

No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.0 (n/a)
Release:	9.0
Codename:	n/a

AdnanHodzic, 2017-02-20:
Turns out this is a [bug in Debian](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=845651#15). 

To resolve this bug, please edit/update `/etc/debian_version` file and make sure its contents are: `9.0`

After which `lsb_release -a` should return `stretch` for codename. Once this happens, please run displaylink-debian tool again.

justmag, 2017-02-20:
it appeared, that /etc/debian_version really contains 9.0, but 
lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.0 (n/a)
Release:	9.0
Codename:	n/a

by the way, 
@ more /etc/os-release
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian

and 
@hostnamectl | grep "Debian"
Operating System: Debian GNU/Linux 9 (stretch)

Should we add "stretch"  somewhere manually?
Thanks, Max

AdnanHodzic, 2017-02-20:
No, I don't like the idea of having codename hardcoded. 

Either, since this is known bug and and other people in Jessie > Stretch transition might end up having same issue. I just pushed changes in code which will accept your codename as such.

Please pull the latest changes and run displaylink-debian again.

justmag, 2017-02-23:
hi Adnan,
now the  driver is up and running, thanks

yet I do not see the 2nd provider. 
The screen on the 2nd monitor is green.

best regards,
Max

AdnanHodzic, 2017-02-23:
Sorry, that's something I can't help you with, as it doesn't concern this tool.

I suggest you use newer kernel and newer xserver-xorg packages. I think something's wrong with your install, as default kernel in Stretch is 4.9 and you're stuck with 3.6 from Jessie.

closed 2017-02-23 by AdnanHodzic

justmag, 2017-02-24:
ok, thanks