AdnanHodzic/displaylink-debian #60: Loaded: not-found (Reason: No such file or directory)

gitowiec, 2017-02-24:
I have this issue on Ubuntu 16.04.2 with Dell 7559 laptop and D3100 Displaylink adapter.
After I run `systemctl status displaylink.service`
```
● displaylink.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)
```
I have latest `1.3.52 16 Feb 2017` DisplayLink drivers installed
Also I have Additional Drivers set to Nvidia version 378.13

AdnanHodzic, 2017-02-24:
@gitowiec what happens if you run "uninstall", reboot. "install", and reboot again.

It seems like displaylink starts, but then stops for some reason, could you include output of following logs when you do that `/var/log/displaylink/DisplayLinkManager.log`

gitowiec, 2017-02-24:
Here is content of /var/log/displaylink/DisplayLinkManager.log after I uninstalled, rebooted and again installed version 1.3.52 and rebooted again. This log file looks a little suspicious for me \o/
```
UVLY+3xs9LIslFFJL9zz9cAZyoUu8ZgDOKws+du3O+0OJ5FTMFk1//ke52r9BzegdXlxqJs0xBmErvAB/Sfl/F8oNb5gSLii7jvXbjDQ/b7u6aH+Q6vdHV+WRL1ayD7x4Lb/lRcrmBIxg22NUJ5L9K81Ezt3vtvKBsE6xpr+jtpnroja/+xXGut0PslK8NCaNbD2B9enIgD5cmbpdBVnQJ3eK/YAPMmwmqgnpl8NZ+eiThGMtJWg5eoWUE6Aqe1Q+yzuAWW3g+XJb8XMLl6apMyDGLV9CcvFP6nT26RRv2UxL0Y4uR22p9LFJ1u7ICLRydaZ8Tm+hF0xs8/4KAZKjC8zkIcRpmoOYTVqmHPrWD8wHJo27iVBGFrWx6e39+QTbgxt2Uenmw9l05vQpdM/DaUSfQe4Tmbwvyw2xSmH4qzye8b+In5qfu2J8RC9gqaf
F6ny/x7M/S/F1fu0oS2K49VgEgDJmdNbTHn6H7T89VMqQPp3VMCZ0J2n0VM0xDcgT3lxpuoykAZMtUEwRyY297JDQIPcICI68Tmj4ZpPizLi9CX9cyUmPzfABR0htOif3atb5XpYQ8bj95WE38BMXQ2jHzfFxs9uLgNDZ1n9oZwyPG7jD8lcyoUcF6tbXIrGg/4bf+DbfHIx60KR9b3ze8Gh+VGQ0LtVSS37RyfRAj8=
nL4TC2qWBc4VYcjAc441hQYbJ4s+ifoFStbFjq1c1JE0ub8SWuRRojpVxlJk7REZ4SD1ZdNkoxfNtPZKie+YJaVylBPCRsMSbA7v7W5/NKU0GlhkmNuWS4iSmry4fVzeXcJpS63cfeRcWqFtnY720oNB13S+jS0wt81YN0uqL4B7df8+qOsR4EAQnwWDUXf9aoaPUPtKN6LSm9pAi81nYQ==
nL4TC2qWBc4VYcjAc441hQYbJ4s+ifoFStbFjq1c1JE0ub8SWuRRojpVxlJk7REZ4SD1ZdNkoxfNtPZKie+YJZKbBuWXpU9WHGCfj72NKqCpYrVU/zPBsJn8c/NCFM+FyRbggEjCdgaY087FEQ1RMNhXw/5LDf8dD/e/lIkMTY5E6VFepHpTpJ48pCICyHYlYJ9BL0IwSwdU7M4QFxVNLQ==
hfugTV6GMJAWvc5soKOJtrc4dQJcls47wm5S54D8SLyCd7jOzvEEU8nRRDaNoZrxwbCt0wY6o7hNM+PrrRRAVb6DinAshamv4Ge1DVjqM03n2meKhwmcQV0dyFTUIF9QctxPVun14YfKjL8ZzlfBDuzvkmywk5svBsT9TEcN8jXu34Wc30ELKTX09c9bBdvXEOOo56ZNBwlxCryWyCyDrd7X1soCNyTtIP9vNZfy+N8=
hfugTV6GMJAWvc5soKOJtrc4dQJcls47wm5S54D8SLyCd7jOzvEEU8nRRDaNoZrxwbCt0wY6o7hNM+PrrRRAVb6DinAshamv4Ge1DVjqM03n2meKhwmcQV0dyFTUIF9QctxPVun14YfKjL8ZzlfBDuzvkmywk5svBsT9TEcN8jXu34Wc30ELKTX09c9bBdvXEOOo56ZNBwlxCryWyCyDrXW8w2fsF91TXdROm3jIq4E=
OoJQrlASts2wT0Wed0J/uW7vtXXwu7WZPubXMS6Q1xm9d7HZzXUJIl92z5ac+je2lfv8KUa4S5qskjmYqJLhYvt1OxEsRDMezJ9eOYqfCO9+gdDwqc9R+WSHVVQCvu+poPqrh68xMnAjJ8+cBqvIG9kHBeJCY9grriwtBiNXhMnWKq55Ytb/kspqMapuL8MtNPP3A5g1hx2BXp2sNJqXg05RJUIPKAEn897lrxjNoGBn7joDmYS1OMIAi3Y+BtS2
uR4htyF5yAqak7QyO8fZ51c5l4KGnpcrbZSxVCLm/pMNc/PNRy2CVWID3Q+DMjqVgvSTA1U28qz9CwTSs+/p5XDCGKu8Llm6WQT4jT5P0dmAh5SCnd1fHPgqbRCZtGMXs6iaqH7v6OkptlQXQMqbxx3Mh5jAEsMS+c7EQqn1rPsjeuUa5yRC9w026vr8ENXyclDL9znGeMD2xbeIJUT+iNgkWO7ThtNFcGye6KbKtHY=
OoJQrlASts2wT0Wed0J/ucSTiiQ58l/plz6vaBVcqsGasAxf84Hj0bnrHJwHDO8zbXQGb9bQrrERjmPY4606W0v387TXiPwAv+7eRs6Z3wfI4HOOFaWSedKKxza741V/lHUVj8M48qKahOwbMbgAZUKUZ+NUmNxF1M0O2Umf7Ka8XgDb+rzOYOYw5B3o48nK2S9QLQXdGfgj7hTx9qa3m8ulWbLCvDiPYcsq63L8EH3MeJCxw2cTDa0HDAHgYW0t
nL4TC2qWBc4VYcjAc441hQ+ZNvDdOj7U1gf7J8rhe4x+UOdhrbs3IU/FExH73Hm6MLu/vEmR8SKtGIsUQdOjPTttBT3LoaP7Wj9lwKXNSdewi2V9FeXb9AhCWYBZm7URYSYhCUUYZl2cY18NmOrAhPmDmZar5GB8FUq3KIeRK6Fs3UpgOahX6otKOUm6m52PVCuJayUSwLcMtW5wTCVyLA==
hfugTV6GMJAWvc5soKOJtsVkL6WLOdxbSlfi8lK4RWSBOhp5lIKTSpwYPVdMLkbCOz+fFzLMRPoZvQjv7czgeYDY6XlVeAfU+PvVs+dglv4mIRSe1bt+m8btyc0BmfjxljU1D2gTgy3K+wunXbwzBnCsu1nJFSiuZK5h9C+rgZl6zlj2zFzOWv/pq7hVWlEWDJZho6H47Oii0iC38WMKZN4Y8DCzRiOtZEKUJnCVB5o=
avdrBSBZzzBQgpa4I3gM0aF5j3F/wiw2F/lKiOW3E59wdbfk0y/0Qhtt6Z/LS/5sD1MNIj3lTjhAFReMBBCoSipwiL5B5jySbnmTqoaXvKmmqPDFiLVRmbBYqhaxs6JZJeBNzE5OOmXO9in4kpcWgbE2j/Rh1GPEE+tMjrGP+gvG+KQhvWwZBgsbVa7GBg7C81pHPZHCab4igZCDaEaYzgUoCGn2fkxbvLBW4wBdfLb7EtdIRPjRLodJr60TdqLw59qI0t+RmYXxS2XnEfcDTESlMI2IZxiB8qoMi759bc2NiUL2Yf/G/rLW3rJDEtFG
lsnhG0qlJMqiOZNM6GOTOzMbXmjw73gFzS4GN2N3Qbp7J9v3PjXwYFbwe9PRXT2eZKgxz+YwUtg0DfTnqs1WGuQpSgRS+n+bcvgSr4RIb6TikhblEmu8e30f8muq9PjjZS64cBVhdJXvgktJpq43lKV28GSgiHLhp074Vs1YVfT7jxSrl94HtzrYljcQR/nv6ArXG8RyBryOZsUff8ld68TbIwoyRIQ03D32Ldz1qXg+WMMe1eYL3P3YoirvFM584kz9CAmG2/kIvQmeNYxM9yEXJ7b9qQ1JQrgq4B2ufVfwRluWWoX5qU5j/NHPMJ+0xDGcqTi0GURJzmx/xNXKilO7xtqZ0mX0XWZJFDfh6rf4oz9gHQaRWigJFz2xo+4Xs8Q9OtZM+ttWV/+rZ8+qNmtN4XP0vavoUWWbvrHq4CyhaXwe8jLSe+61smCSOpWD4Adj2MGkS3DtVMEsfgj4ag==
OoJQrlASts2wT0Wed0J/ueonWQyxRVVbpb+CEU5PHXKexR2kPg/AR2AXHmaS8bRp5HdaAA4VnZNVAthhyYnO6ZfudOjedGOA90Jf6B8Kh4XB7uQA4lSD5DafWRRcFXZzckJH+Xohfli6d7dM+AfHy3NAYBFuzV6AivQpaf9aasabrfLbEjbBNL8DV7eV9BMAqnyef3mBkbJfDBTE24e1jyyqiaxMw3Ft714Dt9+po1K3NXjha1u7Su5JKWhMjEAR
nL4TC2qWBc4VYcjAc441hboEMhIifEISOkMIpQwAv/uzEyz8jDMbojXCtDFkT75Q01j7dc+qfXBrYe87FhwtftB9iWwdrOJ2w7p213tw2TxnfA+Rtc+nKTPHvl53xtOy76xcOptjzE3G54UNiDJboMYoeUksKlCWBCrQYrS1XPjbHAXHQk3paHbp5CdOm87vcaOhN+9EHn3BhWStwtooBA==
nL4TC2qWBc4VYcjAc441hboEMhIifEISOkMIpQwAv/uzEyz8jDMbojXCtDFkT75Q01j7dc+qfXBrYe87FhwtfuOHTAgiuuzqdrJkYpQcqGle+0rnYRpmUqxh0M753Ba+IDbBkzDnZsQniwZnSdllW8BrdOK25Xw4UHmhbXiGvmSaa24z4JgWyp9efC5vvb1Ar7awD1C62onlrYIcYQTHxQ==
nL4TC2qWBc4VYcjAc441hfSogY3Sp+Kr0bYB66zoLSpHhNhuPWQUHMrREYtkE8sVR5fKZE4s63x68NyQD8umwXO3QVjejd+sGt2Oxd5glnRzu8OTIHZVARuqb6Tlwkdfq/4Yf407Zmg/vssJlkE0oa8bPpJyTPFpuRDerBDN2GDogDZzEgp1HribCD1JYWDxqPfl92lua9kc49xpwDRwkQ==
qhdM5DcbWZiR02whBfZ5X0OExUrjbJVHGeCwTKfGaY+3mPkXeF9OY00+QrPddlg9FT2tjhLki1aJ/3z3rmdwPLGJ+OnwS+x50BwrGpgZoiGMMsh9vS/T18S/cTASBgQZAWCNdD5HEkQchyHXXgN5jJv1WouZN/EnW8Bwu/DK9D4ENjzBTR+vfAM9Pp74tra6Lp4hGaFp+pwWcQMo4a8U5BGv7E31FA/H+V5J6CYippI=
uR4htyF5yAqak7QyO8fZ516vTOXjNmLLtHRrMa9lGsVm71shFe8bjEw8OTkBVwDEpEhPrytSs1HWe1T+ZRCIPiF4LX4U805FOVOFp27jrSHeFW2LN2KlbFeAVPs8OJBB3qFNbaIsX9JR60rDP+3yHzI/peFDDYRBRpWFgF+LrHmD4YW5MIjPyyJyM5cvqC1XpWH4bX9kWdVYSq+0FS/03LJrwJx6edLHRa5ivsRH+TA=
OoJQrlASts2wT0Wed0J/uVjS5EerFQ7DBIgQi57Ug8WciD773OdRj0woNsuJyilW2EsfnXFd6M3++0o+OaWuo413WpCXFzmx7xWIm12BiTXXvcJgROpSoUnjKgFx5k8njjyvt7nK4KjGy2AS3ztnkG3gBztoYkJsTBn7yZ94tp3nR4jYrkQwjNsblye69CWdjgUFe2jOZOWtN0oHGZWxhhsowpeIygHauna/nCU3wMNPCtaytOr7X2rzL00NJcuy
[... cut ...]
ioDy/JiznFE/F5l46FIuuhX6LhRoWNykuxSs1o5Q1I/r754G2ufigdLAHlrybc0kMHif9xpCDQQBLpjhLTpzbPRFgJEsrAsq5o9RNNCDsaDADmpdRZQhhCTgeVQI+V59fKApeWWgwXhTAA98zAGocxL4Y4z5A+JcC0onaMdm9OqtZxhYTxPOgkicMWOqLFV3UOZ+Ri61rJpinDftDvdV3H3NBUm4lKYXmD83iMSVLUcnZz9LRmAZZPGDrwnBvnRd
2A5ghFPdikxEcVgzOgnlC5JMFj1dNqWbz5IkHNtHYs+zbw54DM5eSLioF+86pCVSgUFXouCxkKPYOP6ZdgwpYFAPTE5n60H1bDKYWJIAPvjD5kUBdD9CpRzzCQDHSTSPfWs1KvYHH3CWAvQtB7PucXvN+UNQzy3/xUJbg6urES8MNO4ZYbtWQCWliskY6M62oIKibQoj3HwFD1Be5kaihr84F5sWgmfsh3DlEdAaNpp6nxZoww132jO7dHHXaAGm3tkbwWMMjS5g7bMv6C00RQ==
```

gitowiec, 2017-02-24:
There is more information:
DisplayLink works if I switch Additional Drivers from `Nvidia binary driver (378.13)` to `X.Org X server - Nouveau`. But then if I shutdown my computer I get this error:
[image]

AdnanHodzic, 2017-02-25:
@gitowiec glad it worked out and you managed to install the driver. 

I'm sorry, but I can't help you with the new problem you're facing. As it doesn't concern this tool and is strictly related to your setup and configuration.

Since it seems like Nvidia related problems, I suggest you submit an [issue with upstream](https://github.com/DisplayLink/evdi)

Good luck

closed 2017-02-25 by AdnanHodzic