AdnanHodzic/displaylink-debian #61: Unsatisfied dependencies. Missing component: DKMS

orggue, 2017-02-27:
Hi there, I have a problem int he process of installing displaylink. Some data:
OS: debian stretch (testing) KDE, amd.
I had to say I managed a few months ago to install it in the same computer in debian stretch gnome. 
and the output:
```bash
Platform requirements satisfied, proceeding ...                                                
                                                                          
-------------------------------------------------------------------

Downloading DisplayLink Ubuntu driver:                             

--2017-02-27 12:30:49--  http://www.displaylink.com/downloads/file?id=744
Resolving www.displaylink.com (www.displaylink.com)... 54.229.186.74, 52.208.159.97, 52.48.31.214, ...
Connecting to www.displaylink.com (www.displaylink.com)|54.229.186.74|:80... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f744_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.3.52.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJU
EA&Expires=1488195653&Signature=L2LX0S%2BCc6AypPDcjfRv03hBr00%3D [following]
--2017-02-27 12:30:50--  http://assets.displaylink.com/live/downloads/software/f744_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.3.52.zip?AWSAccessKeyId=AKI
AJHGQWPVXWHEDJUEA&Expires=1488195653&Signature=L2LX0S%2BCc6AypPDcjfRv03hBr00%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.64.156
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.64.156|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12796532 (12M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_1.3.52.zip’

DisplayLink_Ubuntu_1.3.52.zip              100%[======================================================================================>]  12,20M  7,48MB/s    in 1,6s

2017-02-27 12:30:52 (7,48 MB/s) - ‘DisplayLink_Ubuntu_1.3.52.zip’ saved [12796532/12796532]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_1.3.52.zip
  inflating: 1.3.52/displaylink-driver-1.3.52.run
Creating directory displaylink-driver-1.3.52
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.3.52  100%

-------------------------------------------------------------------

Installing driver version: 1.3.52

DisplayLink Linux Software 1.3.52 install script called: install
Distribution discovered: Debian GNU/Linux 9.0 (stretch)
Unsatisfied dependencies. Missing component: DKMS.
This is a fatal error, cannot install DisplayLink Linux Software.

-------------------------------------------------------------------

Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory
sed: can't read /lib/systemd/system/dlm.service: No such file or directory
Failed to enable unit: File dlm.service: No such file or directory

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_1.3.52.zip" file
Removing redundant: "1.3.52" directory

-------------------------------------------------------------------

Re-install complete, please reboot to apply the changes

-------------------------------------------------------------------
```

AdnanHodzic, 2017-06-26:
You must have DKMS installed as it's one of the dependencies, and during the install process you will be prompted to install it if it's missing.

I just ran displaylink-debian on clean install of Debian Stretch (DKMS was also missing) and I couldn't reproduce the problem:

```
ahodzic@stretch:~/displaylink-debian$ sudo ./displaylink-debian.sh 

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.9.0-3-amd64 not found! Install? [Y/n] 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  linux-compiler-gcc-6-x86 linux-headers-4.9.0-3-common linux-kbuild-4.9
The following NEW packages will be installed:
  linux-compiler-gcc-6-x86 linux-headers-4.9.0-3-amd64 linux-headers-4.9.0-3-common
  linux-kbuild-4.9
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 9049 kB of archives.
After this operation, 51.4 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://security.debian.org stretch/updates/main amd64 linux-compiler-gcc-6-x86 amd64 4.9.30-2+deb9u1 [462 kB]
Get:2 http://security.debian.org stretch/updates/main amd64 linux-headers-4.9.0-3-common all 4.9.30-2+deb9u1 [7471 kB]
Get:3 http://security.debian.org stretch/updates/main amd64 linux-kbuild-4.9 amd64 4.9.30-2+deb9u1 [668 kB]
Get:4 http://security.debian.org stretch/updates/main amd64 linux-headers-4.9.0-3-amd64 amd64 4.9.30-2+deb9u1 [448 kB]
Fetched 9049 kB in 1s (5613 kB/s)                
Selecting previously unselected package linux-compiler-gcc-6-x86.
(Reading database ... 160583 files and directories currently installed.)
Preparing to unpack .../linux-compiler-gcc-6-x86_4.9.30-2+deb9u1_amd64.deb ...
Unpacking linux-compiler-gcc-6-x86 (4.9.30-2+deb9u1) ...
Selecting previously unselected package linux-headers-4.9.0-3-common.
Preparing to unpack .../linux-headers-4.9.0-3-common_4.9.30-2+deb9u1_all.deb ...
Unpacking linux-headers-4.9.0-3-common (4.9.30-2+deb9u1) ...
Selecting previously unselected package linux-kbuild-4.9.
Preparing to unpack .../linux-kbuild-4.9_4.9.30-2+deb9u1_amd64.deb ...
Unpacking linux-kbuild-4.9 (4.9.30-2+deb9u1) ...
Selecting previously unselected package linux-headers-4.9.0-3-amd64.
Preparing to unpack .../linux-headers-4.9.0-3-amd64_4.9.30-2+deb9u1_amd64.deb ...
Unpacking linux-headers-4.9.0-3-amd64 (4.9.30-2+deb9u1) ...
Setting up linux-headers-4.9.0-3-common (4.9.30-2+deb9u1) ...
Setting up linux-compiler-gcc-6-x86 (4.9.30-2+deb9u1) ...
Setting up linux-kbuild-4.9 (4.9.30-2+deb9u1) ...
Setting up linux-headers-4.9.0-3-amd64 (4.9.30-2+deb9u1) ...
dkms not found! Install? [Y/n] 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
[... 119 lines cut ...]

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------
```

After reboot, driver was working just fine, i.e:

```
ahodzic@stretch:~$ systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-06-26 17:23:38 CEST; 1min 32s ago
  Process: 537 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.3.54 && mod
 Main PID: 569 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─569 /opt/displaylink/DisplayLinkManager
```

Please run [U]ninstall, reboot and then run [I]install, make sure that DKMS is installed during installation process as you can see above ^. 

If problem still persists, please feel free to re-open this issue.

Ajorona, 2018-12-29:
When you download the zip file and run the executable from DisplayLink there is no prompt to install DKMS, which is not part of the ubuntu distro. You still get the error: 

Unsatisfied dependencies. Missing component: DKMS.
This is a fatal error, cannot install DisplayLink Linux Software.

To get around this first install the DKMS framework from a terminal

You can install the DKMS framework from a Terminal using the command sudo apt-get install dkms, as mentioned on the how to install page from DisplayLink

Then running the executable will install the DisplayLink software

AdnanHodzic, 2018-12-30:
@Ajorona `dkms` package is [installed as one of the dependencies](https://github.com/AdnanHodzic/displaylink-debian/blob/master/displaylink-debian.sh#L67) when starting install process with displaylink-debian.

But if this didn't happen for @jorgechato then this issue should definitively be re-opened. 

If this is still the issue feel free to re-open the issue and provide me with output of: `sudo ./displaylink-debian.sh --debug`

petkostas, 2019-02-21:
I am experiencing the same problem, DKMS is installed:

```
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.9.0-8-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------
```
Yet the installation is failing:

```
-------------------------------------------------------------------

Installing driver version: 5.1

DisplayLink Linux Software 5.1.26 install script called: install
Distribution discovered: Debian GNU/Linux 9.7 (stretch)
Unsatisfied dependencies. Missing component: DKMS.
This is a fatal error, cannot install DisplayLink Linux Software.

-------------------------------------------------------------------
```

AdnanHodzic, 2019-02-21:
@petkostas `dkms` package is installed as part of the dependencies as you can see here:
`dkms is installed` so not sure how you get this problem.

What would happen if you did `sudo apt-get install dkms` or `sudo apt-get install --reinstall dkms` followed by `sudo ./displaylink-debian.sh --install`?

petkostas, 2019-02-21:
You can ignore the previous comment, problem is that dkms is installed under `/usr/sbin` which was not in the path of the root user.

AdnanHodzic, 2019-03-05:
Closing due to inactivity.

closed 2019-03-05 by AdnanHodzic

interstate420productions, 2019-05-16:
Hi.. Sorry for bumping this.. But I'm glad that the answer was discovered.   I am at a loss of how to solve it.. can someone explain to a newbie exactly how to get these drivers installed.. DKMS is installed on my system  but I think as mentioned

> You can ignore the previous comment, problem is that dkms is installed under `/usr/sbin` which was not in the path of the root user.

How do I fix this so that I can get my dock working?  Thanks In Advance

GoodGuyGregory, 2020-05-09:
Just FYI this also process works great for Ubuntu Thanks @Ajorona :rocket:

AysadKozanoglu, 2020-07-28:
before running ./displaylink-debian.sh 

on terminal:
```
PATH=$PATH":/usr/sbin""
```
solved the issue.