AdnanHodzic/displaylink-debian #62: ERROR* EDID has major version 0, instead of 1 on Kalilix

kumudkhk, 2017-03-05:
root@KumudXPS:~# uname -a
Linux KumudXPS 4.9.0-kali2-amd64 #1 SMP Debian 4.9.13-1kali1 (2017-03-01) x86_64 GNU/Linux

I tried installation on Kali Linux 4.9. It would not install on 4.6. I had to upgrade the kernel. On 4.9 the installation completes. I can see all my 4 monitors, however when the computer reboots I can not login anymore. After entering username and password to login, the screen would show below errors and immediately kick back to login screen. The only way I was able to login was boot the computer in recovering mode, run the displaylink.sh and uninstall the application. 

Mar  4 23:05:27 KumudXPS kernel: [  223.095098] [drm:drm_edid_block_valid [drm]] *ERROR* EDID has major version 0, instead of 1
Mar  4 23:05:27 KumudXPS kernel: [  223.166630] [drm:drm_edid_block_valid [drm]] *ERROR* EDID has major version 0, instead of 1

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which could resolve your problem.

Could you please run re-install with latest changes and report if this same problem exists?

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic