AdnanHodzic/displaylink-debian #63: Not work. Cannot find dlm.service

renat2985, 2017-03-24:
How can I install dlm.service?

```
renat@HPeOS [22:19:31] [~/git/displaylink-debian] [master]
-> % sudo ./displaylink-debian.sh 

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-3.19.0-82-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Downloading DisplayLink Ubuntu driver:

--2017-03-24 22:19:36--  http://www.displaylink.com/downloads/file?id=744
Распознаётся www.displaylink.com (www.displaylink.com)… 52.209.162.172, 54.229.186.74, 52.48.31.214, ...
Подключение к www.displaylink.com (www.displaylink.com)|52.209.162.172|:80... соединение установлено.
HTTP-запрос отправлен. Ожидание ответа... 303 See Other
Адрес: http://assets.displaylink.com/live/downloads/software/f744_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.3.52.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1490387400&Signature=tKydzWBvmFM29j71eIeEyAwZMTg%3D [переход]
--2017-03-24 22:19:36--  http://assets.displaylink.com/live/downloads/software/f744_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.3.52.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1490387400&Signature=tKydzWBvmFM29j71eIeEyAwZMTg%3D
Распознаётся assets.displaylink.com (assets.displaylink.com)… 52.218.64.4
Подключение к assets.displaylink.com (assets.displaylink.com)|52.218.64.4|:80... соединение установлено.
HTTP-запрос отправлен. Ожидание ответа... 200 OK
Длина: 12796532 (12M) [application/octet-stream]
Сохранение в: «DisplayLink_Ubuntu_1.3.52.zip»

100%[===============================================================================>] 12 796 532  3,61MB/s   за 3,4s   

2017-03-24 22:19:39 (3,61 MB/s) - «DisplayLink_Ubuntu_1.3.52.zip» сохранён [12796532/12796532]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_1.3.52.zip
  inflating: 1.3.52/displaylink-driver-1.3.52.run  
Creating directory displaylink-driver-1.3.52
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.3.52  100%  

-------------------------------------------------------------------

Installing driver version: 1.3.52

DisplayLink Linux Software 1.3.52 install script called: install
Distribution discovered: elementary OS Freya
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
Installing EVDI kernel module to kernel tree
EVDI kernel module built successfully
Installing x64-ubuntu-1604/DisplayLinkManager
Installing libraries
Installing firmware packages
Installing license file
Adding udev rule for DisplayLink DL-3xxx/5xxx devices

-------------------------------------------------------------------

Performing post install steps

sed: невозможно прочитать /lib/systemd/system/dlm.service: Нет такого файла или каталога
sudo: systemctl: command not found

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_1.3.52.zip" file
Removing redundant: "1.3.52" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------

renat@HPeOS [22:19:55] [~/git/displaylink-debian] [master]
```

NocturnalFred, 2017-05-10:
@renat2985
 
I recently was able to make this work.  There are many things to be aware of.

One, are you dual booting?  Was your device a windows device originally?  If so you need to disable secure boot because secure boot will force drivers to need to be signed before you can install them.  You can do this in your BIOS.

Once you disable secure boot, reinstall, and then restart your computer you should find that the dlm service is running.  (sudo systemctl status dlm.service)

Then you need to be sure you have the proper video driver installed.  SOFTWARE CENTER > Additional Drivers > pick the right driver.   (If the driver isn't listed you will need to install the driver by hand from your graphics card providers homepage)

After that follow the post-installation instructions for using xrandr to set your output devices to be discovered so that you can enable them in the "DISPLAY" system setting of ubuntu.

mkamioner, 2017-06-22:
Same problem. Bump

William-Gearhart, 2017-06-23:
I have a similar problem:

`william@william:~$ systemctl status dlm.service`

`● dlm.service`
`Loaded: not-found (Reason: No such file or directory)`
`Active: inactive (dead)`

AdnanHodzic, 2017-06-26:
@renat2985 there was an exception made in case distribution was Freya to rely on Upstart. However since then there have been many changes in code, which makes this tool soley rely that distro is runing systemd.

With that said, would it be possible for you to upgrade to Loki? Loki is using systemd, and this problem shouldn't exists. Please make sure to run [R]einstall, or [U]uninstall followed by [I]nstsall and your problem will be fixed. 

@mkamioner @William-Gearhart could you please tell me which distro are using, and preferably paste the whole output which will help me debug it.

Before that, please make sure you run [U]install first, and then run [I]install again.

William-Gearhart, 2017-06-28:
I'm using Debian. I reinstalled the drivers, and here is the output:

`root@william:/home/william# /home/william/Downloads/displaylink-debian.sh

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: u

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-3.16.0-4-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

/home/william/Downloads/displaylink-debian.sh: line 257: displaylink-installer: command not found

-------------------------------------------------------------------

Performing clean-up

-------------------------------------------------------------------

Uninstall complete

-------------------------------------------------------------------

root@william:/home/william# /home/william/Downloads/displaylink-debian.sh

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-3.16.0-4-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Downloading DisplayLink Ubuntu driver:

--2017-06-28 13:09:10--  http://www.displaylink.com/downloads/file?id=993
Resolving www.displaylink.com (www.displaylink.com)... 54.229.186.74, 52.48.31.214, 52.208.159.97, ...
Connecting to www.displaylink.com (www.displaylink.com)|54.229.186.74|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 23308 (23K) [text/html]
Saving to: ‘DisplayLink_Ubuntu_1.3.54.zip’

DisplayLink_Ubuntu_ 100%[===================>]  22.76K  --.-KB/s    in 0.1s    

2017-06-28 13:09:11 (164 KB/s) - ‘DisplayLink_Ubuntu_1.3.54.zip’ saved [23308/23308]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_1.3.54.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of DisplayLink_Ubuntu_1.3.54.zip or
        DisplayLink_Ubuntu_1.3.54.zip.zip, and cannot find DisplayLink_Ubuntu_1.3.54.zip.ZIP, period.
chmod: cannot access '1.3.54/displaylink-driver-1.3.54.run': No such file or directory
/home/william/Downloads/displaylink-debian.sh: line 214: ./1.3.54/displaylink-driver-1.3.54.run: No such file or directory
mv: cannot stat 'displaylink-driver-1.3.54/': No such file or directory
sed: can't read 1.3.54/displaylink-driver-1.3.54/displaylink-installer.sh: No such file or directory
sed: can't read 1.3.54/displaylink-driver-1.3.54/displaylink-installer.sh: No such file or directory

-------------------------------------------------------------------

Installing driver version: 1.3.54

/home/william/Downloads/displaylink-debian.sh: line 232: cd: 1.3.54/displaylink-driver-1.3.54: No such file or directory

-------------------------------------------------------------------

Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory
sed: can't read /lib/systemd/system/dlm.service: No such file or directory
Failed to enable unit: File dlm.service: No such file or directory

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_1.3.54.zip" file

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------

(REBOOTED)

william@william:~$ systemctl status dlm.service 
Unit dlm.service could not be found.
`

adangel, 2017-06-28:
This

> Archive: DisplayLink_Ubuntu_1.3.54.zip
End-of-central-directory signature not found. Either this file is not
a zipfile, or it constitutes one disk of a multi-part archive. In the
latter case the central directory and zipfile comment will be found on
the last disk(s) of this archive.

Looks now like the problems in #76 , #77, #78 - which should be fixed now.

Download the latest version of [displaylink-debian.sh](https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh) and try the installation again.

AdnanHodzic, 2017-07-01:
This problem is definitely fixed in last few commits, please pull the latest version of displaylink-debian and run Reinstall. I verified this is true for both Jessie and Stretch. 

If problem persists, please feel free to re-open the issue.

closed 2017-07-01 by AdnanHodzic

SdnSantos, 2019-07-09:
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: LinuxMint
Release: tessa
Kernel: 4.15.0-54-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Core
VGA: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x89 cap: 0x9, Source Output, Sink Offload crtcs: 2 outputs: 4 associated providers: 2 name:modesetting
Provider 1: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------