AdnanHodzic/displaylink-debian #65: Syntax error

pa102, 2017-04-02:
Get the following error when running in Ubuntu 16.04;

./displaylink-debian.sh: 33: ./displaylink-debian.sh: Syntax error: "(" unexpected

Thanks for any help and thanks for working to make DL less painful in Linux! :)

AdnanHodzic, 2017-04-02:
I just verified, and install worked flawlessly for me on Ubuntu 16.04.

Please reset code changes and pull the latest sources `git reset --hard && git pull` and run `./displaylink-debian.sh` again

pa102, 2017-04-04:
Cloned a fresh copy and still get the same issue. Could it be related to something else, like the version of bash or something similar?

AdnanHodzic, 2017-04-05:
Definitely, which she'll are you using/version?

pa102, 2017-04-05:
bash 4.3-14ubuntu1.1 according to dpkg

AdnanHodzic, 2017-06-26:
What does following return:

```
ls -l /bin/sh
ls -l /bin/bash
```

pa102, 2017-06-27:
lrwxrwxrwx 1 root root 4 Mar 31 14:37 /bin/sh -> dash
-rwxr-xr-x 1 root root 1037528 May 16 13:49 /bin/bash

pa102, 2017-06-29:
I tried this again today now that DisplayLink have updated the drivers and it worked perfectly.

Thanks for your help.

closed 2017-06-29 by pa102