AdnanHodzic/displaylink-debian #67: Unable to locate package linux-headers-3.18.0

peytr, 2017-04-29:
Cannot install Displaylink Driver.

E: Unable to locate package linux-headers-3.18.0-13980-g7357b88
E: Could not find any package by glob linux-headers-3.18.0-13980-g7357b88
E: Could not find any package by regex linux-headers-3.18.0-13980-g7357b88
linux-headers-3.18.0-13980-g7357b88 installation failed.

Asus Chromebook Flip C302a
Crouton Ubuntu 16.04

Have tried
sudo apt-get install linux-headers-generic
and
sudo apt-get install linux-headers-generic

Have also tried to install the standard Displaylink driver with the /lib/modules/4.6.0-1-amd64/build/Kconfig line commented out.

Any assistance would be greatly appreciated.

AdnanHodzic, 2017-06-26:
You have that problem because when you try to run: 
```
sudo apt-get install linux-headers-generic
```

it can't install headers, because that kernel is not generic kernel. That would work if your kernel image was installed using:

```
sudo apt-get install linux-image-generic
```

If you want to install headers for your custom kernel please run:

```
sudo apt-get install linux-headers-$(uname -r)
```

If the install runs fine, then please [U]ninstall/reboot/[I]nstall. Otherwise install "generic image" as mentioned above.

I'm closing the issue, if the problem persists please feel free to re-open it.

t1v0, 2017-07-20:
I am getting this issue with kali installed on a chromebook (using crouton).

```
sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-3.18.0-14861-g70c266fc4519
E: Couldn't find any package by glob 'linux-headers-3.18.0-14861-g70c266fc4519'
E: Couldn't find any package by regex 'linux-headers-3.18.0-14861-g70c266fc4519'
```

Trying to install the generic kernal image as suggested yields this:

```sudo apt-get install linux-image-generic
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package linux-image-generic is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
```

Any suggestions?

AdnanHodzic, 2017-08-06:
Could you tell me what's the output of
`uname -r`

and can you confirm that in your `/etc/apt/sources.list` you have besides `main` you also `contrib non-free` enabled?, i.e:

`deb http://http.kali.org/kali kali-rolling main contrib non-free`

It seems like on Kali Linux reason why `linux-headers-$(uname -r)` doesn't work is because there's a "kali" string inside of it. So kernel headers package is refer to i.e:

`linux-headers-4.8.0-kali1-amd64`

Could you try manually installing headers with "kali" string added to it?

AdnanHodzic, 2017-12-25:
Closed due to inactivity.

closed 2017-12-25 by AdnanHodzic

rahoolnikam, 2018-02-23:
I have output of uname -r as 4.13.0-kali1-686-pae. However I am unable to update headers as well as the distro. The source list is modified to deb http://http.kali.org/kali kali-rolling main contrib non-free.

Kindly suggest any workaround. Unable to install any packages or getting any updates.

redjamesg, 2018-02-23:
What error do you see when you run the displaylink-debian.sh ?

rahoolnikam, 2018-02-23:
Error for displaylink-debian.sh 
bash : displaylink-debian.sh command not found

out for lsb_release are  Kali, kali-rolling and Kali kali-rolling respectively.

redjamesg, 2018-02-23:
And you are executing it as you should?
 **./displaylink-debian.sh** 
And the execution bit is set on the file ? 
 **chmod +x displaylink-debian.sh**

What if you issue:
> /bin/bash displaylink-debian.sh

rahoolnikam, 2018-02-23:
No such file or directory

for all the executions.

redjamesg, 2018-02-23:
Are you sure you even have the file correctly downloaded?
If you fetched it as a zip dowbload, you need to unzip that and find the script in directory **displaylink-debian-master**

rahoolnikam, 2018-02-23:
Issue resolved. It was due to outdated archive-keyring package.

Thanks for your help buddy.