AdnanHodzic/displaylink-debian #68: Doesn't work with Debian Stretch on Dell 7720

gitowiec, 2017-05-13:
I have Dell laptop with Optimus as mentioned in topic, I installed Debian 8 + KDE and made immediate upgrade to Stretch right after that.
I installed DisplayLink driver with Your script (http://foolcontrol.org/?p=1777).
I removed anything nvidia (`apt purge nvidia*`) and blocked nouveau like it is written here (http://www.allaboutlinux.eu/remove-nouveau-and-install-nvidia-driver-in-debian-8/) - Because I need this laptop as working station and not for gaming.
I would like to run additional (3rd) monitor with Dell D3100 USB station.
Yestarday I managed it with Ubuntu 16.04 but with plenty of stability issues I couldn't resolve on my own (turning on monitors randomly, changing positions relative to each other, very slow refresh rate of DisplayLink connected monitor), so I decided to go stable with Debian.
Here is `dmesg -w` right after inserting D3100 USB plug into USB 3.0 port of my Inspiron 17 (7720) laptop:

```
[   75.850820] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[   75.992965] usb 3-2: New USB device found, idVendor=2109, idProduct=2812
[   75.992968] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   75.992969] usb 3-2: Product: USB2.0 Hub             
[   75.992970] usb 3-2: Manufacturer: VIA Labs, Inc.         
[   75.994044] hub 3-2:1.0: USB hub found
[   75.994231] hub 3-2:1.0: 4 ports detected
[   76.110976] usb 4-2: new SuperSpeed USB device number 2 using xhci_hcd
[   76.368153] usb 4-2: New USB device found, idVendor=2109, idProduct=0812
[   76.368155] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   76.368156] usb 4-2: Product: USB3.0 Hub             
[   76.368157] usb 4-2: Manufacturer: VIA Labs, Inc.         
[   76.370028] hub 4-2:1.0: USB hub found
[   76.370275] hub 4-2:1.0: 4 ports detected
[   76.458774] usb 3-2.2: new high-speed USB device number 4 using xhci_hcd
[   76.572933] usb 3-2.2: New USB device found, idVendor=2109, idProduct=2812
[   76.572934] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   76.572935] usb 3-2.2: Product: USB2.0 Hub             
[   76.572936] usb 3-2.2: Manufacturer: VIA Labs, Inc.         
[   76.574025] hub 3-2.2:1.0: USB hub found
[   76.574230] hub 3-2.2:1.0: 4 ports detected
[   76.759020] usb 4-2.2: new SuperSpeed USB device number 3 using xhci_hcd
[   77.016125] usb 4-2.2: New USB device found, idVendor=2109, idProduct=0812
[   77.016127] usb 4-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   77.016128] usb 4-2.2: Product: USB3.0 Hub             
[   77.016129] usb 4-2.2: Manufacturer: VIA Labs, Inc.         
[   77.018644] hub 4-2.2:1.0: USB hub found
[   77.018869] hub 4-2.2:1.0: 4 ports detected
[   77.766769] usb 4-2.1: new SuperSpeed USB device number 4 using xhci_hcd
[   77.787318] usb 4-2.1: New USB device found, idVendor=17e9, idProduct=436e
[   77.787320] usb 4-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   77.787321] usb 4-2.1: Product: Dell USB3.0 Dock
[   77.787321] usb 4-2.1: Manufacturer: DisplayLink
[   77.787322] usb 4-2.1: SerialNumber: 1504290519B
[   80.922511] usb 4-2.3: new SuperSpeed USB device number 5 using xhci_hcd
[   80.943036] usb 4-2.3: New USB device found, idVendor=17e9, idProduct=436e
[   80.943037] usb 4-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   80.943038] usb 4-2.3: Product: Dell USB3.0 Dock
[   80.943039] usb 4-2.3: Manufacturer: DisplayLink
[   80.943040] usb 4-2.3: SerialNumber: 1504290519
[   80.978917] cdc_ncm 4-2.3:1.5: MAC-Address: 9c:eb:e8:20:dc:cc
[   80.978920] cdc_ncm 4-2.3:1.5: setting rx_max = 16384
[   80.979041] cdc_ncm 4-2.3:1.5: setting tx_max = 16384
[   80.979198] cdc_ncm 4-2.3:1.5 usb0: register 'cdc_ncm' at usb-0000:00:14.0-2.3, CDC NCM, 9c:eb:e8:20:dc:cc
[   80.979688] usbcore: registered new interface driver cdc_ncm
[   80.981315] usbcore: registered new interface driver cdc_wdm
[   80.986078] usb 4-2.3: Warning! Unlikely big volume range (=511), cval->res is probably wrong.
[   80.986079] usb 4-2.3: [15] FU [Dell USB Audio Playback Volume] ch = 6, val = -8176/0/16
[   80.986550] usb 4-2.3: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
[   80.986552] usb 4-2.3: [12] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
[   80.986746] usbcore: registered new interface driver cdc_mbim
[   80.986748] usbcore: registered new interface driver snd-usb-audio
[   80.988480] cdc_ncm 4-2.3:1.5 enx9cebe820dccc: renamed from usb0
[   81.030910] IPv6: ADDRCONF(NETDEV_UP): enx9cebe820dccc: link is not ready
[   81.030973] IPv6: ADDRCONF(NETDEV_UP): enx9cebe820dccc: link is not ready
[   81.838214] evdi: [D] add_store:195 Increasing device count to 1
[   81.838475] evdi: [D] evdi_crtc_init:312 drm_crtc_init: 0
[   81.838730] evdi: [W] evdi_painter_crtc_state_notify:363 Painter does not exist!
[   81.838733] evdi: [D] evdi_detect:78 Painter is disconnected
[   81.838735] evdi evdi.0: No connectors reported connected with modes
[   81.838737] [drm] Cannot find any crtc or sizes - going 1024x768
[   81.840160] evdi evdi.0: fb1: evdidrmfb frame buffer device
[   81.840161] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   81.840162] [drm] No driver support for vblank timestamp query.
[   81.840162] [drm] evdi: evdi_stats_init
[   81.840166] [drm] Initialized evdi 1.3.43 20170117 on minor 1
[   82.201593] evdi: [D] evdi_detect:78 Painter is disconnected
[   82.201628] evdi: [D] evdi_crtc_cursor_set:238 evdi_crtc_cursor_set unlock
[   82.201649] evdi: [D] evdi_detect:78 Painter is disconnected
[   82.201980] evdi: [D] evdi_painter_crtc_state_notify:360 (dev=0) Notifying crtc state: 3
[   82.201981] evdi: [W] evdi_painter_send_crtc_state:266 Painter is not connected!
[   82.203960] evdi: [D] evdi_painter_crtc_state_notify:360 (dev=0) Notifying crtc state: 3
[   82.203961] evdi: [W] evdi_painter_send_crtc_state:266 Painter is not connected!
[   82.276763] evdi: [W] evdi_painter_disconnect:462 (dev=0) An unknown connection to ffff9a6deb59a200 tries to close us
[   82.276764] evdi: [W] evdi_painter_disconnect:463  - ignoring
[   82.285111] evdi: [W] evdi_painter_disconnect:462 (dev=0) An unknown connection to ffff9a6deb59a200 tries to close us
[   82.285112] evdi: [W] evdi_painter_disconnect:463  - ignoring
[   82.289988] evdi: [D] evdi_painter_connect:443 (dev=1) Connected with ffff9a6deb59a200
[   82.289989] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[   82.290011] evdi: [D] evdi_painter_crtc_state_notify:360 (dev=1) Notifying crtc state: 3
[   82.290872] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[   82.290875] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[   82.843151] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[   82.843154] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
```
Here is screenshot of `systemctl dlm` NOTICE that I have additional small monitor plugged into laptops external HDMI, 3rd monitor I wish it could worked using DisplayLink:

[image]

Please help turning my 3rd screen on.
Best regards

gitowiec, 2017-05-13:
I just read this comment https://github.com/AdnanHodzic/displaylink-debian/issues/42#issuecomment-266146964 and I did:
```
$ xrandr --setprovideroutputsource 1 0
$ xrandr --setprovideroutputsource 2 0
$ xrandr --setprovideroutputsource 3 0
```
Then I installed `arandr` and it detected my third monitor (2560x1440) - it was surprise for me.
Then I rightclicked it and made Active.
And it started up, but with strange behaviour - 
1. I see only mouse pointer
2. Whole screen is black
3. Screenshot made shows properly the content on 2560x1440 monitor - I am stumbled!
I don't know what to do else. Please help.

[image]

adangel, 2017-05-13:
I've seen a similar problem (only mouse pointer showing, but otherwise black screen). Try disable the "modesetting" driver for your internal graphics card - this driver doesn't seem to work well as a outputsource...

As I have a intel graphics card, for me it was enough to add the file `/etc/X11/xorg.conf` with the following content:

~~~
Section "Device"
	Identifier "Intel"
	Driver "intel"
#	Option "AccelMethod" "uxa"
	Option "VSync" "false"
EndSection
~~~

As you are using debian, you should be able to create this file via `sudo cp -i /usr/share/doc/xserver-xorg-video-intel/xorg.conf /etc/X11/xorg.conf`

I've added additionally the option `"VSync" "false"`, as this reduces constant error logging about VSync failures, when only the external monitor is active.

The file makes sure, that X11 is using the intel driver and not the default modesetting.

After creating this file, reboot.

The command `xrandr --listproviders` should now look like this:

~~~
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 1 name:Intel
Provider 1: id: 0x153 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
~~~

Note the name being "Intel" now for provider 0.
Provider 1 is the D3100 and is working fine for me with modesetting driver.

gitowiec, 2017-05-13:
Thank You very much @adangel for Your help.
I did `/etc/X11/xorg.conf` file with content You provided and made reboot.
After login I checked providers and its:
```
Providers: number : 2
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:Intel
Provider 1: id: 0x126 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```
Then I issued this command `sudo xrandr --setprovideroutputsource 1 0` and Xorg crashed.
I attached full Xorg log, but here is some last lines with exception.
```
[   433.131] (EE) 
[   433.131] (EE) Backtrace:
[   433.131] (EE) 0: /usr/lib/xorg/Xorg (xorg_backtrace+0x4a) [0x56426479446a]
[   433.131] (EE) 1: /usr/lib/xorg/Xorg (0x5642645db000+0x1bd219) [0x564264798219]
[   433.132] (EE) 2: /lib/x86_64-linux-gnu/libpthread.so.0 (0x7f49c5732000+0x110c0) [0x7f49c57430c0]
[   433.132] (EE) 3: /usr/lib/xorg/modules/drivers/intel_drv.so (0x7f49c132f000+0x13e87c) [0x7f49c146d87c]
[   433.132] (EE) 4: /usr/lib/xorg/modules/drivers/intel_drv.so (0x7f49c132f000+0x13eb5f) [0x7f49c146db5f]
[   433.132] (EE) 5: /usr/lib/xorg/modules/drivers/intel_drv.so (0x7f49c132f000+0x139ae8) [0x7f49c1468ae8]
[   433.132] (EE) 6: /usr/lib/xorg/Xorg (0x5642645db000+0xd3e7e) [0x5642646aee7e]
[   433.132] (EE) 7: /usr/lib/xorg/Xorg (ProcRRSetScreenSize+0xe8) [0x564264700ab8]
[   433.132] (EE) 8: /usr/lib/xorg/Xorg (0x5642645db000+0x545e5) [0x56426462f5e5]
[   433.132] (EE) 9: /usr/lib/xorg/Xorg (0x5642645db000+0x58568) [0x564264633568]
[   433.132] (EE) 10: /lib/x86_64-linux-gnu/libc.so.6 (__libc_start_main+0xf1) [0x7f49c53b42b1]
[   433.132] (EE) 11: /usr/lib/xorg/Xorg (_start+0x2a) [0x56426461d21a]
[   433.132] (EE) 
[   433.132] (EE) Segmentation fault at address 0x0
[   433.132] (EE) 
Fatal server error:
[   433.132] (EE) Caught signal 11 (Segmentation fault). Server aborting
[   433.132] (EE) 
[   433.132] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   433.132] (EE) Please also check the log file at "/home/marek/.local/share/xorg/Xorg.0.log" for additional information.
[   433.132] (EE) 
[   433.132] (II) AIGLX: Suspending AIGLX clients for VT switch
[   433.197] (EE) Server terminated with error (1). Closing log file.
```
Please take a look at file or this excerpt above - what can I do more?
I'm going to google content of this backtrace to find solution.

https://gist.github.com/gitowiec/7cf0f809be372b59edcd946ac46fc8d3

gitowiec, 2017-05-13:
ok, I am so lame :frowning:
I skipped hash sign in configuration!!!
```
Section "Device"
	Identifier "Intel"
	Driver "intel"
#	Option "AccelMethod" "uxa"
	Option "VSync" "false"
EndSection
```
My monitor works! Thank YOU @adangel !!! :heart::pray:

wordyallen, 2017-05-20:
I only have 1 provider and its Intel. However, I still get a black screen. Here is my output
```sh
 xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 0 name:Intel

```

adangel, 2017-05-20:
It sometimes helps, if I replug the Dell 3100 and/or restart the display link manager service (`sudo systemctl restart dlm.service`). It also takes a while, until the monitors are detected.

redjamesg, 2017-05-20:
What I found on the Dell XPS 13 9360 is that I had to run 'xrandr --listproviders' twice, so in my activation script I let it execute that command until "Providers: numbers" matches the number of monitors I am using (currently three)

vngrd1, 2017-05-26:
this took me more then it should have - Thank @adangel !

benpturner, 2017-09-03:
Hi all, i'm having the same issues as above. I've tried adding an xorg.conf file but then when I run xrandr I only get one output so I cannot activate this. When I remove the xorg.conf file it works but the screen is either blank or doesnt allow me to click or move windows, the mouse cursor moves though.

benpturner, 2017-09-03:
[   14.575944] evdi: [W] evdi_painter_disconnect:471 (dev=1) An unknown connection to ffff8a6ac4ad3c00 tries to close us
[   14.575947] evdi: [W] evdi_painter_disconnect:472  - ignoring
[   14.691221] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[   14.691223] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[   14.691333] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[   14.691334] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[   14.748037] evdi: [D] evdi_painter_crtc_state_notify:359 (dev=1) Notifying crtc state: 3
[   14.818813] evdi: [D] evdi_painter_crtc_state_notify:359 (dev=1) Notifying crtc state: 3
[   40.989228] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[   40.989231] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[   43.007828] evdi: [D] evdi_painter_mode_changed_notify:383 (dev=1) Notifying mode changed: 1920x1080@60; bpp 32; 
[   43.007830] evdi: [D] evdi_painter_mode_changed_notify:384 pixel format 875713112
[   43.007834] evdi: [D] evdi_painter_crtc_state_notify:359 (dev=1) Notifying crtc state: 0
[   43.157183] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[   43.157185] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[  292.176178] evdi: [D] evdi_painter_disconnect:489 (dev=1) Disconnected from ffff8a6ac4b23e00
[  292.176182] evdi: [D] evdi_detect:78 Painter is disconnected
[  292.176216] evdi: [W] evdi_painter_disconnect:471 (dev=-1) An unknown connection to ffff8a6ac4b23e00 tries to close us
[  292.176217] evdi: [W] evdi_painter_disconnect:472  - ignoring
[  292.199645] evdi: [D] evdi_detect:78 Painter is disconnected
[  292.205202] evdi: [D] evdi_painter_crtc_state_notify:359 (dev=-1) Notifying crtc state: 3
[  292.205204] evdi: [W] evdi_painter_send_crtc_state:265 Painter is not connected!
[  292.373984] evdi: [D] evdi_detect:78 Painter is disconnected
[  293.912471] evdi: [W] evdi_painter_disconnect:471 (dev=-1) An unknown connection to ffff8a6aa7e41c00 tries to close us
[  293.912473] evdi: [W] evdi_painter_disconnect:472  - ignoring
[  293.918619] evdi: [D] evdi_painter_connect:450 (dev=1) Connected with ffff8a6aa7e41c00
[  293.918622] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[  293.918903] evdi: [D] evdi_painter_crtc_state_notify:359 (dev=1) Notifying crtc state: 3
[  293.939612] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[  293.939615] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[  294.009158] evdi: [D] evdi_painter_mode_changed_notify:383 (dev=1) Notifying mode changed: 1920x1080@60; bpp 32; 
[  294.009160] evdi: [D] evdi_painter_mode_changed_notify:384 pixel format 875713112
[  294.009164] evdi: [D] evdi_painter_crtc_state_notify:359 (dev=1) Notifying crtc state: 0
[  294.009167] evdi: [D] evdi_painter_dpms_notify:346 (dev=1) Notifying dpms mode: 0
[  294.181466] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
[  294.181470] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff

benpturner, 2017-09-03:
When I add an xorg.conf and run xrandr i get this

╰─>$ xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x48 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 0 name:Intel

benpturner, 2017-09-03:
it now shows 4 outputs not 3

benpturner, 2017-09-03:
but cant seem to use them, any ideas

bartziengs, 2017-10-16:
Thanks a lot for everybody who has contributed to resolving this issue! I was experiencing the same with Elementary OS on a hp notebook with shared intel and nvidia graphics. After an entire day of struggling adangel's trick finally worked out for me. Thank you!!

brianmay, 2018-03-11:
I followed the steps in https://github.com/AdnanHodzic/displaylink-debian/issues/68#issuecomment-301232674, to try to solve what appears to be an identical problem. The screen initially starts drawing fine, but then it freezes except for the mouse pointer. Unfortunately, as soon as I change the driver to Intel, ``xrandr --listproviders`` no longer lists the DisplayLink providers. Which ummm.... somewhat defeats the point.

sirosen, 2018-03-21:
@brianmay, I experienced the same issue on a ThinkPad X230 (integrated intel graphics) and Debian 9.
However, this `/etc/X11/xorg.conf` worked for me, and hopefully works for you too:
```
Section "Device"
  Identifier "DisplayLink"
  Driver     "modesetting"
  Option     "PageFlip" "false"
EndSection
```

According to [this source](https://support.displaylink.com/knowledgebase/articles/1181623-displaylink-ubuntu-driver-after-recent-x-upgrades) the issue only happens on X versions 1.18.3 and newer.

As always, reboot or just restart X to load the new config.

AdnanHodzic, 2018-03-24:
I just made change which will automatically disable PageFlip for modesetting depending on which version of Xorg server is detected. 

Currently changes have been pushed as part of separate branch, which if proved to be working as expected I'll merge with master.

`git pull && git checkout modesetting && sudo ./displaylink-debian.sh`

Please test these changes and let me know if everything works as expected. Thanks!

brianmay, 2018-03-25:
The good news is that the change, when applied manually, works.

The bad news is that the modesetting branch doesn't make this change automagically. At least on Debian/stretch.

I have no idea when I didn't get a ``/usr/share/X11/xorg.conf.d/20-displaylink.conf`` file, if I run the logic from the git commit by hand, it looks fine to me.

redjamesg, 2018-03-25:
It should have been 
cat >! 
from the initial line, to rewrite the 20-something Xorg file.

AdnanHodzic, 2018-03-25:
@brianmay could you please try again with changes made in latest commit?

@redjamesg problem was that I put modesetting file removal as part of `clean-up` where it should've been part of `uninstall` function. This way subsequently after file was written it would be removed :)

AdnanHodzic, 2018-03-26:
File will be owned by root and will be created with 0644 permission, just as rest of the files under `.../xorg.conf.d`/ dir.  

So in this case it's seems this is all good. I'll do bit more testing and then merge these changes with `master`.

brianmay, 2018-03-26:
Not on my system. Like I said, not sure it actually matters:

```
$ ls -l /usr/share/X11/xorg.conf.d/20-displaylink.conf
-rw------- 1 root root 117 Mar 26 17:12 /usr/share/X11/xorg.conf.d/20-displaylink.conf
```

AdnanHodzic, 2018-03-26:
In this case, and just in case. I just pushed changes which will set explicit permissions needed for this file, please pull and try again.

brianmay, 2018-03-26:
LGTM. Thanks.

AdnanHodzic, 2018-03-26:
Most welcome, merging changes with `master`. 

Also closing the issue, in meantime please free to re-open the issue if same problem persists.

timthelion, 2018-04-07:
This is now closed, but on Debian Stretch I had to follow https://github.com/AdnanHodzic/displaylink-debian/issues/68#issuecomment-301232674 manually.

AdnanHodzic, 2018-04-09:
@timthelion please take a backup of `/etc/X11/xorg.conf` before doing this. 

But could you please tell me what happens for you if you run `sudo cp /usr/share/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf` and restart the computer/X?

timthelion, 2018-04-09:
@AdnanHodzic what I did that fixed it was add the intel part, not the display-link page flip dissabling part. BUt now I'm confused, because, even with the Xorg.conf file containing the intel line, I still get modesetting for my main provider:

````
timothy@yoga:~[] > xrandr --listproviders
Providers: number : 3
Provider 0: id: 0xa1 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 2 name:modesetting
Provider 1: id: 0x77 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
````

My laptop is a thinkpad Yoga 11 with intel HD graphics:

````
timothy@yoga:~[] > lspci
00:00.0 Host bridge: Intel Corporation Broadwell-U Host Bridge -OPI (rev 09)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 5500 (rev 09)
00:03.0 Audio device: Intel Corporation Broadwell-U Audio Controller (rev 09)
00:14.0 USB controller: Intel Corporation Wildcat Point-LP USB xHCI Controller (rev 03)
00:16.0 Communication controller: Intel Corporation Wildcat Point-LP MEI Controller #1 (rev 03)
00:1b.0 Audio device: Intel Corporation Wildcat Point-LP High Definition Audio Controller (rev 03)
00:1c.0 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #1 (rev e3)
00:1c.2 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #3 (rev e3)
00:1c.3 PCI bridge: Intel Corporation Wildcat Point-LP PCI Express Root Port #4 (rev e3)
00:1d.0 USB controller: Intel Corporation Wildcat Point-LP USB EHCI Controller (rev 03)
00:1f.0 ISA bridge: Intel Corporation Wildcat Point-LP LPC Controller (rev 03)
00:1f.2 SATA controller: Intel Corporation Wildcat Point-LP SATA Controller [AHCI Mode] (rev 03)
00:1f.3 SMBus: Intel Corporation Wildcat Point-LP SMBus Controller (rev 03)
00:1f.6 Signal processing controller: Intel Corporation Wildcat Point-LP Thermal Management Controller (rev 03)
04:00.0 Network controller: Intel Corporation Wireless 7265 (rev 59)
05:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader (rev 01)
timothy@yoga:~[] > lsusb
Bus 001 Device 003: ID 0483:91d1 STMicroelectronics Sensor Hub
Bus 001 Device 002: ID 8087:8001 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 007: ID 17e9:4302 DisplayLink 
Bus 003 Device 006: ID 2109:0811 VIA Labs, Inc. Hub
Bus 003 Device 005: ID 2109:0811 VIA Labs, Inc. Hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 009: ID 056a:00ec Wacom Co., Ltd 
Bus 002 Device 007: ID 1bcf:2c79 Sunplus Innovation Technology Inc. 
Bus 002 Device 010: ID 06cb:7244 Synaptics, Inc. 
Bus 002 Device 005: ID 8087:0a2a Intel Corp. 
Bus 002 Device 003: ID 534c:0001 SatoshiLabs Bitcoin Wallet [TREZOR]
Bus 002 Device 014: ID 058f:6387 Alcor Micro Corp. Flash Drive
Bus 002 Device 013: ID 2109:3431 VIA Labs, Inc. Hub
Bus 002 Device 012: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 002 Device 011: ID 2109:3431 VIA Labs, Inc. Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
timothy@yoga:~[] > 
````
I'm using the thinkpad USB3.0 dock, which has two DVI ports. Before today I only had one monitor plugged in, and it worked after adding the Xorg file with the intel driver block. I got a second external monitor today, and when I plugged it in, I had the same problem as with the first, frozen display. But after various restarting of the computer and a crash of xorg, now BOTH monitors work. Obviously there is some kind of black magic going on that is unrelated to the xorg files.

timthelion, 2018-04-18:
I think you can close this again. My problem is different. I am using a Thinkpad USB3.0 Dock, which has two DVI ports. This device works for display, but sometimes causes the following problems (workarounds included):

 - No  new USB devices can be connected while dock is plugged in due to bandwidth error. [Probably related to this bug](https://bugzilla.redhat.com/show_bug.cgi?id=1411604).
 - One or both screens show a frozen image. Solution: Unplug the dock, put computer to sleep, wake computer up, replug dock.
 - Xrandr does not detect screens. Solution: Run `xrandr --setprovideroutputsource 1 0 ; xrandr --setprovideroutputsource 2 0 ; xrandr`, unplug and replug device, run same command, put computer to sleep and wake it up, run same command again, run same command again while standing on your head, go take a shower then run same command again (perhaps it depends on the humidity of the air?), repeat until it works.
 - Xorg crashes. Solution: Restart xorg and try again.
 - Screens turn on and off and flicker. Solution: Unplug dock and replug it.

Overall, despite the problems I'm very satisfied with being able to have 4 screens available. Would buy again at ebay prices. But I certainly felt like I was loosing hope for a while there...

brianmay, 2018-04-18:
I have a ThinkPad USB 3.0 Ultra Dock and running Debian/stable, not seen anything like the problems described above. The problems I have seen:

* Plugged dock in while playing Netflix in Chrome. Computer crashed. Not sure if this was due to plugging in while watching Netflix, or if this was a once off coincidence.

* Disconnecting Dock and reconnecting it sometimes results in a situation where the screen contents don't get routed correctly to the screens. I have to disconnect the displays (I use ARandR), disconnect the Dock, run `xrandr --setprovideroutputsource 1 0 ; xrandr --setprovideroutputsource 2 0`, then reconnect the Dock and run the xrandr commands again, and reconnect the displays using ARandR, after which point it works fine. Not sure if all of these steps are important.

markd89, 2018-04-21:
I have this working using the xorg.conf as above. What is strange is that my listproviders still shows modsetting...

```
Providers: number : 2
Provider 0: id: 0x73 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 1 name:modesetting
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

kopax, 2018-09-13:
Hi, thanks again for maintaining this driver for everyone.
I was using it but then docker wanted me to upgrade the kernel and the screen was stored somewhere.

I've seen that you have fixed it in the latest version. I have tried to install it on Debian Stretch.

```bash
$ uname -a 
Linux dev-01 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux
```

I was having a frozen screen but the display is showing ever black, ever the background with some block (cropped).

This is my output with `xrandr --listproviders`:

```bash
Providers: number : 2
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 1 name:modesetting
Provider 1: id: 0x154 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

I have tried all the fixes with `/etc/X11/xorg.conf`: 

```bash
$ sudo cp -i /usr/share/doc/xserver-xorg-video-intel/xorg.conf /etc/X11/xorg.conf
# edit the files..
$ cat /etc/X11/xorg.conf
Section "Device"
	Identifier "Intel"
#	Driver "intel"
	Driver "modesetting"
#	Option "AccelMethod" "uxa"
#	Option "VSync" "false"
#	Option "PageFlip" "false"
EndSection
```
I have tried to enable/disable the commented lines without any effect, the screen is still freezing.

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic

toddsnider-epc, 2018-12-28:
First, thanks for providing this script.  I was able to get it working after manually modifying /etc/X11/xorg.conf.d/20-displaylink.conf to contain
`Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection`
instead of 
`Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection`
I'm using Debian testing on a Dell Inspiron 15 7568 with 4.18 kernel.
`todds@todd-debian:~$ lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 520 (rev 07)
todds@todd-debian:~$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915
todds@todd-debian:~$ lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5
[HD
`
Hope this is helpful

AdnanHodzic, 2018-12-29:
@toddsnider-epc sure has, thanks to this snippet as part of last commit I've added proper Xorg config for anyone with Intel Corporation HD Graphics 520 cards.

adangel, 2019-01-05:
@AdnanHodzic I've got it working now again - here are my settings. Apparently I'm using now modesetting for both the intel card and displaylink. PageFlip needs to be false. If I use the intel/modestting combination, I get rendering errors.

```
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.18.0-3-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4
EVDI service status: up and running
EVDI service version: 4.4.24

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 540
VGA: Intel Corporation Iris Graphics 540 (rev 0a)

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

#Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 1 name:modesetting
Provider 1: id: 0x1df cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

Looks like "540" needs to be added for "xorg_modesetting" as well.
Thanks for the great script!

AdnanHodzic, 2019-01-05:
@adangel great, thanks for the update, I've just pushed changes which will add `xorg_modesetting` for yours as well as other users with `540`. 

Problem with 4.19 kernel was brought up in: https://github.com/AdnanHodzic/displaylink-debian/issues/191 so I'll make sure to reference it to your comment here until I figure out if I'll also manually patch it or wait until they fixed the problem on upstream.

jonathancardoso, 2019-04-03:
Hello, guys.
I'm working hard to get this working on a Parrot OS.

I'm made a lot of workarounds to get install the displaylink-driver.service (dosen't work if I try dlm), I'm double check if evdi it's running (it's).

I have intel graphics card, i'm getting 2 providers but both on modsetting if I set nomodeset all goes blank.

Please, someone help me, it's have been more than 4 days I working on it.

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

Unit dlm.service could not be found.
--------------- Linux system info ----------------

Distro: Parrot
Release: n/a
Kernel: 4.19.0-parrot1-20t-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1
EVDI service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: [HD
Integrated
VGA: Intel Corporation HD Graphics 520 (rev 07)
X11 version: 1.20.3-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x7b cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

damnhalfling, 2019-05-06:
Hi everybody. 

I'm on debian, and screen are freeze after apply de configs on arandr!!!

here follow my configuration

please help me! 

--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-9-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 5916 (rev 02)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
  Identifier "DisplayLink"
  Driver     "modesetting"
  Option     "PageFlip" "false"
EndSection

#Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x72 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting