AdnanHodzic/displaylink-debian #70: High CPU usage

phlegx, 2017-05-16:
Hi!

First of all thanks for this nice script it really works like a charm! 

I first tried to install the driver from the displaylink homepage, but when the external monitors are plugged in, I experience a really high CPU load. I now then tried this installation method but the load stays the same and of course windows I put onto the monitor are laggingg for example when i just write here in the browser.

Does anyone know what the reason for this behaviour might be? How can I debug this?

My model is a ```Lenovo x1 Carbon 2017```

thanks!
[image]

AdnanHodzic, 2017-05-16:
Glad to hear that :)

I reported this same issue with [driver upstream](https://github.com/DisplayLink/evdi/issues/65) awhile back. 

Basically it seems like there's a problem with process not going to sleep when it should and/or proper system utilization when it is connected.

phlegx, 2017-05-16:
@AdnanHodzic thanks a lot for the fast reply! Ok so one has to wait for a new version I guess, right?

Do you still personally experience this high CPU usage on your machine @AdnanHodzic?

AdnanHodzic, 2017-05-16:
No I don't, I also think it might be a combination of distro/kernel/xorg version which might be causing this issue.

If you want to do some debugging, make sure you check out: [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues)

closed 2017-06-26 by AdnanHodzic