AdnanHodzic/displaylink-debian #72: dlm.service could not be found

drcoll, 2017-05-30:
I previously was using Debian Jessie, and just upgraded to Stretch to try and get this working.
After upgrading I went to re-install as below...

Notice the `ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory`
in the post-install... which assumingly is the cause for this?:

```
~ $ systemctl status dlm.service
Unit dlm.service could not be found.
~ $ systemctl start dlm.service
Failed to start dlm.service: Unit dlm.service not found.
```

What are the steps in resolving this?

```sh
$ sudo ./displaylink-debian.sh 

--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: r

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.9.0-3-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

DisplayLink Linux Software 1.3.52 install script called: uninstall
Distribution discovered: Debian GNU/Linux 9.0 (stretch)
Uninstalling
Removing EVDI from kernel tree, DKMS, and removing sources.
Stopping DLM systemd service
Removed /etc/systemd/system/multi-user.target.wants/dlm.service.
Removing suspend-resume hooks
Removing udev rule
Removing Core folder

Uninstallation steps complete.
Please note that the evdi kernel module is still in the memory.
A reboot is required to fully complete the uninstallation process.
rm: cannot remove '/lib/modules/4.9.0-3-amd64/build/Kconfig': No such file or directory
Removing evdi module
rmmod: ERROR: Module evdi is in use

-------------------------------------------------------------------

Performing clean-up

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.9.0-3-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Downloading DisplayLink Ubuntu driver:

--2017-05-30 12:18:08--  http://www.displaylink.com//downloads/file?id=744
Resolving www.displaylink.com (www.displaylink.com)... 54.229.186.74, 52.208.159.97, 52.209.162.172, ...
Connecting to www.displaylink.com (www.displaylink.com)|54.229.186.74|:80... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f744_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.3.52.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1496165230&Signature=Xih2L6TRQQ4X4%2BCW43%2F7a9%2BP4LQ%3D [following]
--2017-05-30 12:18:08--  http://assets.displaylink.com/live/downloads/software/f744_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.3.52.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1496165230&Signature=Xih2L6TRQQ4X4%2BCW43%2F7a9%2BP4LQ%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.65.36
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.65.36|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12796532 (12M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_1.3.52.zip’

DisplayLink_Ubuntu_1.3.52.zip           100%[============================================================================>]  12.20M  2.75MB/s    in 5.1s    

2017-05-30 12:18:14 (2.38 MB/s) - ‘DisplayLink_Ubuntu_1.3.52.zip’ saved [12796532/12796532]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_1.3.52.zip
  inflating: 1.3.52/displaylink-driver-1.3.52.run  
Creating directory displaylink-driver-1.3.52
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.3.52  100%  

-------------------------------------------------------------------

Installing driver version: 1.3.52

DisplayLink Linux Software 1.3.52 install script called: install
Distribution discovered: Debian GNU/Linux 9.0 (stretch)
WARNING: Version 1.3.43 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.

-------------------------------------------------------------------

Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory
sed: can't read /lib/systemd/system/dlm.service: No such file or directory
Failed to enable unit: File dlm.service: No such file or directory

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_1.3.52.zip" file
Removing redundant: "1.3.52" directory

-------------------------------------------------------------------

Re-install complete, please reboot to apply the changes

-------------------------------------------------------------------

```

drcoll, 2017-05-30:
Added `http://ftp.de.debian.org/debian stretch main ` to my `sources.list`
and ran a `sudo apt-get update` and `sudo apt-get upgrade`. Then ran `./displaylink-debian.sh` again and rebooted, and now it appears to be working. However, 2nd monitor does not appear to be functioning properly. Boxes display (all gray, can't see anything) , and the screen does not appear to be "refreshing". When moving mouse to screen it does not appear, and I am unable to "change" anything on the screen. Appears frozen almost.

dalangston, 2017-05-31:
I had a similar problem.  My fix was to create `/etc/X11/xorg.conf.d/20-intel.conf` with the follwoing: 
```
Section "Device"
Identifier "Intel Graphics"
Driver "Intel"
EndSection 
```

What I gathered from Google searching, is that the X was using the modesetting driver for both the integraged Intel graphics chip in my laptop, and the DisplayLlink adapter.  Explicitly telling X to use the Intel driver made all of my screens work properly.

drcoll, 2017-06-02:
@dalangston Thank you! This has resolved my issue.

AdnanHodzic, 2018-11-12:
@dalangston @drcoll could you please do me a favor, and pull the latest changes and test if this is still the problem?

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

varelajp, 2019-03-09:
> In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card.
>
> This feature may still be bit rough around the [...]

Hi Adnan!
Thank you for your hard work in this great driver.
It worked like a charm with Debian Stretch 9.3.
Recently I did a clean install of Debian Stretch 9.8 and the driver appears not to work enabling dlm.service:
**systemctl start dlm.service**
Failed to start dlm.service: Unit dlm.service not found.

In /usr/lib/x86_64-linux-gnu/ I have libstdc++.so.6 and libstdc++.so.6.0.22.
I made the symbolic link: **ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 /opt/displaylink/libstdc++.so.6**
I overwrote libstdc++.so.6 in /opt/displaylink/
¿Maybe my mistake? ¿Maybe a recent Linux kernel? ¿Maybe a recent libstdc++.so.6?

Now the commands you asked for:

**cat /etc/X11/xorg.conf.d/20-displaylink.conf**
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

**lspci | grep -i vga**
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)

**lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'**
i915

**lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5**
Graphics

**./displaylink-debian.sh --debug**
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Regards...
Juan Varela

AdnanHodzic, 2019-03-09:
Can you try doing same thing I suggested here: https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348

closed 2019-03-09 by AdnanHodzic