AdnanHodzic/displaylink-debian #73: usb usb4-port2: cannot disable (err = -32)

Denizar, 2017-05-31:
Hi Adnan,
 
After running your installer several times on Ubuntu 16.04 I am still unable to activate my monitor.
 
My laptop has an Nvidia graphics card which as far as I've read online may potentially be causing extra issues.
 
The error I am getting on demsg upon plugging in the usb is as below:

`[    4.459016] systemd[1]: systemd 229 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ -LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN)
[    4.459129] systemd[1]: Detected architecture x86-64.
[    4.459312] systemd[1]: Set hostname to <deniz-dev>.
[    4.559213] systemd[1]: Listening on /dev/initctl Compatibility Named Pipe.
[    4.559328] systemd[1]: Created slice System Slice.
[    4.559358] systemd[1]: Listening on fsck to fsckd communication Socket.
[    4.559460] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[    4.559501] systemd[1]: Listening on Journal Socket.
[    4.574963] systemd[1]: Mounting POSIX Message Queue File System...
[    4.575361] systemd[1]: Starting Uncomplicated firewall...
[    4.575762] systemd[1]: Started Read required files in advance.
[    4.576407] systemd[1]: Mounting Debug File System...
[    4.576965] systemd[1]: Starting Create list of required static device nodes for the current kernel...
[    4.577528] systemd[1]: Starting Set console keymap...
[    4.581184] systemd[1]: Starting Load Kernel Modules...
[    4.581251] systemd[1]: Listening on udev Control Socket.
[    4.581319] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[    4.581351] systemd[1]: Listening on udev Kernel Socket.
[    4.581382] systemd[1]: Listening on Journal Socket (/dev/log).
[    4.581806] systemd[1]: Started Braille Device Support.
[    4.581920] systemd[1]: Listening on Syslog Socket.
[    4.581962] systemd[1]: Reached target User and Group Name Lookups.
[    4.582061] systemd[1]: Listening on Journal Audit Socket.
[    4.582736] systemd[1]: Starting Journal Service...
[    4.582792] systemd[1]: Reached target Remote File Systems (Pre).
[    4.582821] systemd[1]: Reached target Remote File Systems.
[    4.583387] systemd[1]: Mounting Huge Pages File System...
[    4.585599] systemd[1]: Reached target Encrypted Volumes.
[    4.585707] systemd[1]: Created slice User and Session Slice.
[    4.585718] systemd[1]: Reached target Slices.
[    4.587128] systemd[1]: Started Create list of required static device nodes for the current kernel.
[    4.591380] systemd[1]: Started Uncomplicated firewall.
[    4.604990] systemd[1]: Mounted POSIX Message Queue File System.
[    4.605071] systemd[1]: Mounted Huge Pages File System.
[    4.605121] systemd[1]: Mounted Debug File System.
[    4.611610] lp: driver loaded but no devices found
[    4.616224] ppdev: user-space parallel port driver
[    4.623462] scsi 4:0:0:0: Direct-Access     Multiple Card  Reader     1.00 PQ: 0 ANSI: 0
[    4.623947] sd 4:0:0:0: Attached scsi generic sg2 type 0
[    4.631273] systemd[1]: Starting Create Static Device Nodes in /dev...
[    4.631827] systemd[1]: Started Load Kernel Modules.
[    4.632782] systemd[1]: Mounting FUSE Control File System...
[    4.633270] systemd[1]: Starting Apply Kernel Variables...
[    4.638026] systemd[1]: Mounted FUSE Control File System.
[    4.638455] systemd[1]: Started Apply Kernel Variables.
[    4.646316] systemd[1]: Started Create Static Device Nodes in /dev.
[    4.648884] systemd[1]: Started Journal Service.
[    4.706385] EXT4-fs (sda1): re-mounted. Opts: errors=remount-ro
[    4.778426] systemd-journald[229]: Received request to flush runtime journal from PID 1
[    4.838660] wmi: Mapper loaded
[    4.909006] ACPI Warning: SystemIO range 0x0000000000000428-0x000000000000042F conflicts with OpRegion 0x0000000000000400-0x000000000000044F (\GPIS) (20150930/utaddress-254)
[    4.909014] ACPI Warning: SystemIO range 0x0000000000000428-0x000000000000042F conflicts with OpRegion 0x0000000000000400-0x000000000000047F (\PMIO) (20150930/utaddress-254)
[    4.909018] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[    4.909022] ACPI Warning: SystemIO range 0x0000000000000540-0x000000000000054F conflicts with OpRegion 0x0000000000000500-0x000000000000057F (\GPIO) (20150930/utaddress-254)
[    4.909025] ACPI Warning: SystemIO range 0x0000000000000540-0x000000000000054F conflicts with OpRegion 0x0000000000000500-0x0000000000000563 (\GP01) (20150930/utaddress-254)
[    4.909029] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[    4.909031] ACPI Warning: SystemIO range 0x0000000000000530-0x000000000000053F conflicts with OpRegion 0x0000000000000500-0x000000000000057F (\GPIO) (20150930/utaddress-254)
[    4.909034] ACPI Warning: SystemIO range 0x0000000000000530-0x000000000000053F conflicts with OpRegion 0x0000000000000500-0x0000000000000563 (\GP01) (20150930/utaddress-254)
[    4.909037] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[    4.909038] ACPI Warning: SystemIO range 0x0000000000000500-0x000000000000052F conflicts with OpRegion 0x0000000000000500-0x000000000000057F (\GPIO) (20150930/utaddress-254)
[    4.909041] ACPI Warning: SystemIO range 0x0000000000000500-0x000000000000052F conflicts with OpRegion 0x0000000000000500-0x0000000000000563 (\GP01) (20150930/utaddress-254)
[    4.909045] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[... 145 lines cut ...]
[  145.187213] usb 4-2: new SuperSpeed USB device number 4 using xhci_hcd
[  145.200746] usb 4-2: New USB device found, idVendor=17e9, idProduct=ff0b
[  145.200766] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  145.200771] usb 4-2: Product: ASUSTEK MB169B+
[  145.200775] usb 4-2: Manufacturer: DisplayLink
[  145.200779] usb 4-2: SerialNumber: h3lmtf090415
[  179.049876] usb 4-2: usbfs: process 3044 (DeviceManager_0) did not claim interface 1 before use
[  179.239905] xhci_hcd 0000:04:00.0: WARN Event TRB for slot 1 ep 0 with no TDs queued?
[  180.044446] usb 4-2: usbfs: process 3044 (DeviceManager_0) did not claim interface 1 before use
[  181.052077] usb 4-2: usbfs: process 3044 (DeviceManager_0) did not claim interface 1 before use
[  182.058159] usb 4-2: usbfs: process 3044 (DeviceManager_0) did not claim interface 1 before use
[  183.064359] usb 4-2: usbfs: process 3044 (DeviceManager_0) did not claim interface 1 before use
`
I highlighted usb usb4-port2: cannot disable (err = -32) as the title of this issue since that was highlighted red on the terminal but that may be misleading.
 
xrandr providers:
`Providers: number : 3
Provider 0: id: 0x2ce cap: 0x1, Source Output crtcs: 2 outputs: 1 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x43 cap: 0x2, Sink Output crtcs: 2 outputs: 2 associated providers: 1 name:modesetting
Provider 2: id: 0x43 cap: 0x2, Sink Output crtcs: 2 outputs: 2 associated providers: 1 name:modesetting
`
usbview: 
[image]

I'd be happy to hear your comments if any. Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which could resolve your problem.

Could you please run re-install with latest changes and report if this same problem exists?

AdnanHodzic, 2018-12-04:
In latest commit, I've added code which will now allow "displaylink-debian" to automatically configure your Xorg config based on your graphics card. 

This feature may still be bit rough around the edges and might need some improvement. So if you have still have problems after pulling latest changes and running "re-install", please re-open this story and add output of following commands:

`cat /etc/X11/xorg.conf.d/20-displaylink.conf`
`lspci | grep -i vga`
`lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'`
`lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5`

closed 2018-12-04 by AdnanHodzic