AdnanHodzic/displaylink-debian #74: Failed at step CHDIR spawning /bin/sh

lacic, 2017-05-31:
Hello Adnan,

I've run the installer on a Laptop with Ubuntu 16.04, dGPU (Nvidia) and linux kernel 4.11

After restart I get the following error and am still not able to activate the monitor. Any ideas?

`Unit dlm.service has begun starting up.`
`dlm.service: Failed at step CHDIR spawning /bin/sh: No such file or directory`
`-- Subject: Process /bin/sh could not be executed`
`-- Defined-By: systemd`
 
`-- The process /bin/sh could not be executed and failed.`
`-- The error number returned by this process is 2.`
`dlm.service: Control process exited, code=exited status=200`
` Failed to start DisplayLink Manager Service.`
`-- Subject: Unit dlm.service has failed`
`-- Defined-By: systemd`

AdnanHodzic, 2017-06-26:
Doesn't seem to be related to the displaylink-debian.

Regardless, what does sh point to? i,e: 

```
ls -l /bin/sh
```

if it doesn't return that it points to dash, i.e:

```
lrwxrwxrwx 1 root root 4 feb  2 21:52 /bin/sh -> dash
```

Run following to fix it:

```
sudo ln -sf dash /bin/sh
```

After that please make sure to run [U]ninstall/reboot/[I]nstall again, if the problem persists please re-open this issue and paste the output of what happens during install.

AdirRahamim, 2018-08-03:
Hello,
I have the same issue in my computer, cant start the service.
/bin/sh points to dash...

AdnanHodzic, 2018-08-06:
Have you tried running the script without `/bin/bash` or `/bin/sh` i.e: `sudo displaylink-debian.sh`

AdirRahamim, 2018-08-11:
Thank you! fixed the problem

closed 2018-08-13 by AdnanHodzic

mdelpozo, 2018-10-12:
Helo,
Same problem

-- Unit dlm.service has begun starting up.
out 12 14:40:48 ub-dell systemd[6564]: dlm.service: Changing to the requested working directory failed: No such file or 
out 12 14:40:48 ub-dell systemd[6564]: dlm.service: Failed at step CHDIR spawning /bin/sh: No such file or directory
-- Subject: Process /bin/sh could not be executed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- The process /bin/sh could not be executed and failed.
-- 
-- The error number returned by this process is 2.
out 12 14:40:48 ub-dell systemd[1]: dlm.service: Control process exited, code=exited status=200
out 12 14:40:48 ub-dell systemd[1]: dlm.service: Failed with result 'exit-code'.
out 12 14:40:48 ub-dell systemd[1]: Failed to start DisplayLink Manager Service.
-- Subject: Unit dlm.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit dlm.service has failed.
-- 
-- The result is RESULT.

Always used sudo displaylink-debian.sh

mdelpozo, 2018-10-12:
> Thank you! fixed the problem

Which kernel version are you using?

arashilmg, 2020-01-06:
similar issue here:
```
-- Unit displaylink-driver.service has begun starting up.
Jan 06 23:06:54 arash-ThinkPad-X1-Extreme systemd[4597]: displaylink-driver.service: Changing to the requested working 
Jan 06 23:06:54 arash-ThinkPad-X1-Extreme systemd[4597]: displaylink-driver.service: Failed at step CHDIR spawning /bin
-- Subject: Process /bin/sh could not be executed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- The process /bin/sh could not be executed and failed.
-- 
-- The error number returned by this process is 2.
```

ubuntu 18.04.3  kernel 5.4.8

allo-, 2020-12-10:
I have the same problem on Debian. /bin/sh is linked to dash.

I think the problem may be the error before executing sh, when changing the working directory fails. Does someone know what the working directory should be?