AdnanHodzic/displaylink-debian #75: Ubuntu 16.04.2 suspension issue

i2000s, 2017-06-11:
I tried to suspend my Ubuntu 16.04.2 laptop by closing the lid, but the curser started jittering around rapidly once the lid was closed. It also seemed that the left button of the mouse was pressed at the same time (I did nothing on that). All of these prevented my computer from sleeping as usual. The latest DisplayLink driver for Ubuntu has been installed using the script. My computer model is Lenovo Thinkpad P50. External monitor: ViewSonic VG2230wn.

I have a Microsoft Wireless mouse plugged into the displaylink ports. The same issue occurs if I turn off the mouse. It has no problem to suspend my computer if the Displaylink is not in use. 
I have used xrandr to set up the external monitor; otherwise, the external monitor doesn't show up.

Any idea?
Thanks!

AdnanHodzic, 2018-03-26:
In couple of latest commits, I've added feature which will disable pageflip for modesetting which could resolve your problem.

Could you please run re-install with latest changes and report if this same problem exists?

AdnanHodzic, 2018-03-26:
@i2000s no worries changes have been tested and merged with master. 

Good luck with your new display :)

closed 2018-03-26 by AdnanHodzic