AdnanHodzic/displaylink-debian #79: X Error of failed request:  BadValue (integer parameter out of range for operation)

saljut7, 2017-06-25:
After release of Debian 9 I dist-upgraded and tried again `displaylink-debian`. Everything works fine until...

```
# xrandr --setprovideroutputsource 1 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x44
  Serial number of failed request:  16
  Current serial number in output stream:  17
```

...any ideas?

This is my hardware setup:

```
saljut7@Lenovo-T60:~$ inxi -Fx
System:    Host: Lenovo-T60 Kernel: 4.9.0-3-amd64 x86_64 (64 bit) Desktop: i3 4.13
           Distro: Debian GNU/Linux 9 (stretch)
Machine:   Device: laptop System: LENOVO product: 2007FVG v: ThinkPad T60
           Mobo: LENOVO model: 2007FVG BIOS: LENOVO v: 79ETD8WW (2.18 ) date: 09/05/2007
CPU:       Dual core Intel Core2 T5600 (-MCP-) cache: 2048 KB 
           clock speeds: max: 1833 MHz 1: 1833 MHz 2: 1333 MHz
Graphics:  Card: Advanced Micro Devices [AMD/ATI] RV515/M54 [Mobility Radeon X1400]
           Display Server: X.Org 1.19.2 drivers: modesetting,ati,radeon (unloaded: fbdev,vesa)
           Resolution: 1400x1050@60.02hz, 1280x1024@60.02hz
           GLX Renderer: Gallium 0.4 on llvmpipe (LLVM 3.9, 128 bits) GLX Version: 3.0 Mesa 13.0.6
Audio:     Card Intel NM10/ICH7 Family High Definition Audio Controller driver: snd_hda_intel
           Sound: Advanced Linux Sound Architecture v: k4.9.0-3-amd64
[...]
```

AdnanHodzic, 2017-06-26:
Seems to be X server issue, regardless what happens if you run [U]ninstall/reboot/[I]nstall?

saljut7, 2017-06-26:
Yes, uninstalled, rebootetd, installed, rebooted, same result.

```
# dmesg
[  414.492082] usb 2-5: new high-speed USB device number 10 using ehci-pci
[  414.641302] usb 2-5: New USB device found, idVendor=17e9, idProduct=4300
[  414.641308] usb 2-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  414.641312] usb 2-5: Product: USB3.0 to DVI Adapter
[  414.641315] usb 2-5: Manufacturer: DisplayLink
[  414.641319] usb 2-5: SerialNumber: 000800160247459

# systemctl status dlm.service
● dlm.service - DisplayLink Manager Service
   Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-06-26 21:24:23 CEST; 4min 27s ago
  Process: 2206 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.3.54 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 2208 (DisplayLinkMana)
    Tasks: 24 (limit: 4915)
   CGroup: /system.slice/dlm.service
           └─2208 /opt/displaylink/DisplayLinkManager

Jun 26 21:24:23 LenovoT60-Debian9 systemd[1]: Starting DisplayLink Manager Service...
Jun 26 21:24:23 LenovoT60-Debian9 systemd[1]: Started DisplayLink Manager Service.

# xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x44 cap: 0x0 crtcs: 2 outputs: 3 associated providers: 0 name:ATI Mobility Radeon X1400 @ pci:0000:01:00.0
Provider 1: id: 0x2b2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

# xrandr --setprovideroutputsource 1 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x44
  Serial number of failed request:  16
  Current serial number in output stream:  17
```

But shouldn't the output of `xrandr --listproviders` alter when I attach/detach DisplayLinkAdapter? I just noticed it doesn't alter even `dmesg` recognizes attaching/detaching.

saljut7, 2017-06-26:
Just tried with a Lenovo x201 and here I get a third display via DisplayLink, but:

Using i3 window manager (which I usually use) I only can see a workspace/desktop on 3rd display but can neither move mouse on this display nor move some windows to the 3rd display.

Using XFCE window manager I can at least access 3rd display with mouse but can not move windows there as well.

Tested hardware:

```
$ inxi -Fi
System:    Host: Lenovo-X201 Kernel: 4.9.0-3-amd64 x86_64 (64 bit)
           Desktop: Xfce 4.12.3 Distro: Debian GNU/Linux 9 (stretch)
Machine:   Device: laptop System: LENOVO product: 3680BR4 v: ThinkPad X201
           Mobo: LENOVO model: 3680BR4
           BIOS: LENOVO v: 6QET70WW (1.40 ) date: 10/11/2012
Battery    BAT0: charge: 0.0 Wh 0.0% condition: 0.7/78.0 Wh (1%)
CPU:       Dual core Intel Core i5 M 540 (-HT-MCP-) cache: 3072 KB 
           clock speeds: max: 2534 MHz 1: 1466 MHz 2: 1199 MHz 3: 1333 MHz
           4: 1199 MHz
Graphics:  Card: Intel Core Processor Integrated Graphics Controller
           Display Server: X.Org 1.19.2 drivers: modesetting (unloaded: fbdev,vesa)
           Resolution: 1280x800@60.16hz, 1280x1024@60.02hz, 1280x1024@60.02hz
           GLX Renderer: Mesa DRI Intel Ironlake Mobile
           GLX Version: 2.1 Mesa 13.0.6
```

saljut7, 2017-07-21:
@AdnanHodzic you can close this issue if you want. My workaround is that I replaced my lenovo T60 with a lenovo X201. I think I will not invest more time to figure out what the problem is with displaylink-debian and a T60.
Thanks a lot for your scripts!

AdnanHodzic, 2017-07-21:
@saljut7 cool, most welcome and thanks for the update

PedroFernandoLima, 2019-01-28:
I'm having this same problem on the ideapad 330s, let's reopen this problem  : )

AdnanHodzic, 2019-01-28:
@PedroFernandoLima I need output of your:
`sudo ./displaylink-debian.sh --debug`

and have you tried making same changes which were mentioned here: https://github.com/AdnanHodzic/displaylink-debian/issues/68#issuecomment-301232674 do they fix this problem for you?

AdnanHodzic, 2019-02-03:
I'm closing this issue due to inactivity.

Feel free re-open the issue If you're still experiencing same problem after pulling latest updates and running reinstall.

closed 2019-02-03 by AdnanHodzic

cristianbass01, 2025-03-19:
Hi, I'm having the same issue:

`$ xrandr --listproviders
Providers: number : 6
Provider 0: id: 0x43 cap: 0x0 crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x119 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xf8 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xd7 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xaa cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0x89 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting
`

and then:

`$ xrandr --setprovideroutputsource 1 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x43
  Serial number of failed request:  20
  Current serial number in output stream:  21
`

The debug output is the following:

`--------------- Linux system info ----------------

Distro: Ubuntu
Release: noble
Kernel: 6.8.0-45-generic

---------------- DisplayLink info ----------------

Driver version: 1.14.8
DisplayLink service status: up and running
EVDI service version: 1.14.8

------------------ Graphics card -----------------

Vendor: i915
Subsystem: UHD
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): NVIDIA Corporation GM108M [GeForce MX130] (rev a2)
X11 version: 21.1.12-1ubuntu1.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x43 cap: 0x0 crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x119 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xf8 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xd7 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xaa cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0x89 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting
`