AdnanHodzic/displaylink-debian #80: recieved unsupported platform error

JR1994, 2017-07-03:
I am running Debain testing and I received this error when trying to run the script. Now I needed to manually download and install the 4.9 linux source package from the debian archive. It was not available in the repo.

    sudo ./displaylink-debian.sh 
    
    --------------------------- displaylink-debian ----------------------------
    
    DisplayLink driver installer for Debian based Linux distributions:
    
    * Debian GNU/Linux
    * Ubuntu
    * Elementary OS
    * Linux Mint
    * Kali Linux
    
    Options:
    
    [I]nstall
    [U]ninstall
    [R]e-install
    [Q]uit
    
    Select a key: [i/u/r/q]: i
    
    -------------------------------------------------------------------
    
    Checking dependencies
    
    unzip is installed
    linux-headers-4.9.0-3-amd64 is installed
    dkms is installed
    lsb-release is installed
    linux-source is installed
    
    ---------------------------------------------------------------
    
    Unsuported platform: Debian buster
    
    This tool is Open Source and feel free to extend it
    GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/
    
    ---------------------------------------------------------------

redjamesg, 2017-07-03:
Try with this patch: https://github.com/AdnanHodzic/displaylink-debian/pull/81

AdnanHodzic, 2017-07-03:
@redjamesg merged the patch with main branch. Thanks!

@JR1994 please pull latest git source and run Reinstall.

JR1994, 2017-07-03:
It installed thanks!

AdnanHodzic, 2017-07-03:
Cool, enjoy!

closed 2017-07-03 by AdnanHodzic