AdnanHodzic/displaylink-debian #82: display will not turn on

JR1994, 2017-07-03:
I installed the driver successfully, plugged in the monitor and rebooted. I got the to the GDM login screen and tried to login. I kept being returned to the login screen. So I unplugged the monitor and rebooted. I was able to login and then plugged in the monitor. I saw the AOC logo then it goes blank. I verified the service is running. and I so the bellow in dmesg.

    sudo systemctl status dlm
    [sudo] password for hhernandez:
    ● dlm.service - DisplayLink Manager Service
       Loaded: loaded (/lib/systemd/system/dlm.service; enabled; vendor preset: enabled)
       Active: active (running) since Mon 2017-07-03 12:45:00 MST; 5min ago
      Process: 1333 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/1.3.54 && modprobe evdi) (code=exited, status=0/SUCCESS)
     Main PID: 1353 (DisplayLinkMana)
        Tasks: 23 (limit: 4915)
       CGroup: /system.slice/dlm.service
               └─1353 /opt/displaylink/DisplayLinkManager

    sudo dmesg | grep -e drm -e evdi
    [   25.595095] [drm] Initialized
    [   25.680394] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
    [   33.066718] evdi: [I] Initialising logging on level 5
    [  346.956564] evdi: [D] add_store:235 Increasing device count to 1
    [  346.956849] evdi: [D] evdi_crtc_init:328 drm_crtc_init: 0
    [  346.956939] evdi: [W] evdi_painter_crtc_state_notify:362 Painter does not exist!
    [  346.956941] evdi: [D] evdi_detect:78 Painter is disconnected
    [  346.956944] evdi evdi.0: No connectors reported connected with modes
    [  346.956948] [drm] Cannot find any crtc or sizes - going 1024x768
    [  346.959009] evdi evdi.0: fb0: evdidrmfb frame buffer device
    [  346.980039] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
    [  346.980040] [drm] No driver support for vblank timestamp query.
    [  346.980044] [drm] evdi: evdi_stats_init
    [  346.980053] [drm] Initialized evdi 1.4.1 20170417 on minor 1
    [  347.028500] evdi: [D] evdi_detect:78 Painter is disconnected
    [  347.028593] evdi: [D] evdi_detect:78 Painter is disconnected
    [  347.028926] evdi: [D] evdi_painter_crtc_state_notify:359 (dev=0) Notifying crtc state: 3
    [  347.028927] evdi: [W] evdi_painter_send_crtc_state:265 Painter is not connected!
    [  347.031685] evdi: [D] evdi_painter_crtc_state_notify:359 (dev=0) Notifying crtc state: 3
    [  347.031686] evdi: [W] evdi_painter_send_crtc_state:265 Painter is not connected!
    [  347.996842] evdi: [D] evdi_detect:78 Painter is disconnected
    [  353.775910] evdi: [W] evdi_painter_disconnect:471 (dev=0) An unknown connection to ffff92c9a7dcba00 tries to close us
    [  353.775911] evdi: [W] evdi_painter_disconnect:472  - ignoring
    [  360.506814] evdi: [W] evdi_painter_disconnect:471 (dev=0) An unknown connection to ffff92c9dc5af200 tries to close us
    [  360.506815] evdi: [W] evdi_painter_disconnect:472  - ignoring
    [  360.518147] evdi: [D] evdi_painter_connect:450 (dev=1) Connected with ffff92c9dc5af200
    [  360.518149] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
    [  360.518182] evdi: [D] evdi_painter_crtc_state_notify:359 (dev=1) Notifying crtc state: 3
    [  360.519029] evdi: [D] evdi_detect:75 (dev=1) Painter is connected
    [  360.519032] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff

redjamesg, 2017-07-03:
Have you looked at the post install guide?
https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md

JR1994, 2017-07-03:
> Have you looked at the post install guide?
> https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-
> install-guide.md

Installing arandr allowed me to turn the display on, but I am experiencing
a lot of screen tearing

redjamesg, 2017-07-04:
Take a look at https://github.com/AdnanHodzic/displaylink-debian/issues/68

JR1994, 2017-07-10:
Yup got it to work! Sorry for the late response

closed 2017-07-10 by AdnanHodzic