AdnanHodzic/displaylink-debian #83: Missing /opt/displaylink folder

KamHugo, 2017-07-11:
Hi,

I'm on Debian Stretch, and when I'm running your script, it seems that /opt/displaylink is missing.
Here is the full trace.

```
--------------------------- displaylink-debian ----------------------------

DisplayLink driver installer for Debian based Linux distributions:

* Debian GNU/Linux
* Ubuntu
* Elementary OS
* Linux Mint
* Kali Linux

Options:

[I]nstall
[U]ninstall
[R]e-install
[Q]uit

Select a key: [i/u/r/q]: I

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-3.16.0-4-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement
available at http://www.displaylink.com/downloads/file?id=993
and accept here: [Y]es or [N]o: Y

Downloading DisplayLink Ubuntu driver:

--2017-07-11 15:09:51--  http://www.displaylink.com/downloads/file?id=993
Résolution de www.displaylink.com (www.displaylink.com)… 52.209.162.172, 54.229.186.74, 52.48.31.214, ...
Connexion à www.displaylink.com (www.displaylink.com)|52.209.162.172|:80… connecté.
requête HTTP transmise, en attente de la réponse… 303 See Other
Emplacement : http://assets.displaylink.com/live/downloads/software/f993_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.3.54.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1499779074&Signature=IAtBN0py1t21PKbO%2BPjTAeO7Euk%3D [suivant]
--2017-07-11 15:09:51--  http://assets.displaylink.com/live/downloads/software/f993_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%201.3.54.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1499779074&Signature=IAtBN0py1t21PKbO%2BPjTAeO7Euk%3D
Résolution de assets.displaylink.com (assets.displaylink.com)… 52.218.48.12
Connexion à assets.displaylink.com (assets.displaylink.com)|52.218.48.12|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 12808721 (12M) [application/octet-stream]
Sauvegarde en : « DisplayLink_Ubuntu_1.3.54.zip »

DisplayLink_Ubuntu_1.3.54.zip 100%[=================================================>]  12,21M  4,46MB/s    in 2,7s    

2017-07-11 15:09:56 (4,46 MB/s) — « DisplayLink_Ubuntu_1.3.54.zip » sauvegardé [12808721/12808721]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_1.3.54.zip
  inflating: 1.3.54/displaylink-driver-1.3.54.run  
  inflating: 1.3.54/LICENSE          
   creating: 1.3.54/__MACOSX/
  inflating: 1.3.54/__MACOSX/._LICENSE  
Creating directory displaylink-driver-1.3.54
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.3.54  100%  

-------------------------------------------------------------------

Installing driver version: 1.3.54

DisplayLink Linux Software 1.3.54 install script called: install
Distribution discovered: Debian GNU/Linux 9.0 (stretch)
WARNING: Version 1.4.1 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.

-------------------------------------------------------------------

Performing post install steps

ln: impossible de créer le lien symbolique '/opt/displaylink/libstdc++.so.6': Aucun fichier ou dossier de ce type
sed: impossible de lire /lib/systemd/system/dlm.service: Aucun fichier ou dossier de ce type
Failed to enable unit: File dlm.service: No such file or directory

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_1.3.54.zip" file
Removing redundant: "1.3.54" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

-------------------------------------------------------------------
```

I tried to mkdir the /opt/displaylink folder myself, the ln is then successful, but the dlm service is still missing.
I guess that the folder should be created by the modified Ubuntu script, but it looks like it doesn't.

Thanks in advance

redjamesg, 2017-07-12:
Your error: 
> WARNING: Version 1.4.1 of EVDI kernel module is already running.
> Please reboot before attempting to re-install DisplayLink Linux Software.
> Installation terminated.
Indicate there's another instance of EVDI running, cuold you pleasetry to run "[U]ninstall" then reboot and run [I]nstall again?

KamHugo, 2017-07-17:
Thanks a lot.

I uninstall, reboot then install again without error.
It still doesn't work thought but I haven't troubleshoot enough to find an error.

Since it's problably due to another issue, I close this one and will open a new one then I find the error.

closed 2017-07-17 by KamHugo